espressif/esp-aliyun #1: 导入就报错，估计是自带的底层库没有放进来！

xuhongv, 2017-11-15:
导入就报错，估计是自带的底层库没有放进来！ esp8266-rtos-sdk文件夹下面为空。

ustccw, 2017-11-15:
你需要将 submodule clone 下来。
```
$git submodule update --init
```

xuhongv, 2017-11-15:
在此工程上加入 stdio.h头文件对一些字符串操作，马上就报错：.text' will not fit in region `iram1_0_seg'，网上找了一个下午没有答案，能否你们提供下解决方法，万分感谢，人都快晕了。。。

ustccw, 2017-11-16:
@xuhongv 你好，在本仓库的user_main.c 头文件中已经包含有 #include <stdio.h>
不清楚你的问题是什么，以及你的配置和环境，比如是在windows 还是 ubuntu , 你的编译器版本，是否了解ESP8266入门指南，[官网资料](http://espressif.com/zh-hans/support/download/documents)。如果你仍然有问题，你可以将编译失败的测试代码提供在 github 上，我会尽快安排测试以便复现，感谢你对ESP的支持。

qq1453373574, 2017-12-07:
@ustccw 你好！  我把代码编译后报这个错误是什么原因？ make: *** No rule to make target 'all'.  Stop.

xcguang, 2017-12-08:
@xuhongv .text' will not fit in region `iram1_0_seg' 你 iram 空间不够了，是不是 iram 放的函数比较多？

ustccw, 2017-12-08:
@qq1453373574 你好，可否提供一下你的详细操作步骤呢？是不是编译器没有导入等问题。

BlueSkyai, 2017-12-16:
@ustccw 是作者吗？能不能加QQ联系？我的Q号是1423592839

BlueSkyai, 2017-12-16:
@ustccw 我按照步骤修改了几个自定义内容，编译通过并且烧录了，但是没有连上阿里云

xcguang, 2017-12-17:
Hi,
若有问题请再次交流，谢谢！

ustccw, 2017-12-18:
@BlueSkyai 你好，请问是否按照 [ESP8266 对接阿里云指南](https://github.com/espressif/esp8266-aliyun) 进行参数设置的？尤其 `6.1.1 SDK 修改` 是否设置正确？没有连接上是卡在哪一步？有问题请在此交流，我们会及时处理，谢谢！

BlueSkyai, 2017-12-18:
@ustccw wifi和阿里云参数都设置过了，我烧录的时候用的windows工具，地址是按照说明，我是4M的8266，我把./gen_misc.sh默认内存大小改为了选项6,也就是4M的，调试软件用接的串口0，除了上电打印ROM Log的乱码信息，其他都没有提示了，连接错误提示信息也没有

ustccw, 2017-12-19:
@BlueSkyai 请详述下自己的步骤，比如乱码问题，是不是波特率正确设置为74880, 编译选项和烧写地址是相互依赖的，你的烧写地址是怎么样的？遇到问题，请提供尽可能多的信息。比如提供你烧写的所有bin 文件，提供问题截图，问题log，以便更快的定位和解决问题，谢谢。

BlueSkyai, 2017-12-19:
@ustccw 你好，从编译烧录到调试的bin和过程显示在压缩文件当中
[debug.zip](https://github.com/espressif/esp8266-aliyun/files/1570697/debug.zip)

ustccw, 2017-12-19:
@BlueSkyai 你的 bin 可以跑，也能显示 log, 只是没有配置好 wifi，请知悉。
至于你的烧录工具和串口工具，我不太了解，请使用 README.md 中推荐的 cygwin。 
另外，你需要学习下 [ESP8266 SDK 入门指南](http://espressif.com/zh-hans/support/download/documents?keys=&field_type_tid%5B%5D=14)，你的 截图2 烧录位置和编译的bin不对应，所以会产生乱码。  

烧录位置如下： 
flash size: 2MB-c1
0x0000: bin/boot_v1.6.bin
0x1000: bin/upgrade/user1.2048.new.5.bin   
0x1fc000: bin/esp_init_data_default.bin 
0x1fe000: bin/blank.bin

[image]

ustccw, 2017-12-19:
@BlueSkyai 选项6是 spi flash size 为 4MB, map 方式为 1024KB+1024KB。不是代表内存大小。
如果配置了选项6,烧写设置如下：
flash size: 4MB-c1
0x0000: bin/boot_v1.6.bin
0x1000: bin/upgrade/user1.4096.new.6.bin   
0x3fc000: bin/esp_init_data_default.bin 
0x3fe000: bin/blank.bin  
建议你阅读下 [ESP8266 SDK 入门指南](http://espressif.com/zh-hans/support/download/documents?keys=&field_type_tid%5B%5D=14)

BlueSkyai, 2017-12-20:
@ustccw  我编译的 6=4096KB(1024KB+1024KB)" ，四个bin对应地址下载，我用的串口调试软件除了乱码没有显示信息，是程序调试信息默认在Linux环境下显示吗？可否简单看看windows工具是不是能正常用?我把工具添加在附件中
[esp8266工具.zip](https://github.com/espressif/esp8266-aliyun/files/1574650/esp8266.zip)

ustccw, 2017-12-20:
@BlueSkyai 请使用我们[官方烧写工具](http://espressif.com/zh-hans/support/download/other-tools)

BlueSkyai, 2017-12-21:
@ustccw 你好，我用官方工具重新烧录了几次，串口始终是乱码，阿里云也显示离线，不知道什么原因，调试信息在附件中
[debug.zip](https://github.com/espressif/esp8266-aliyun/files/1578374/debug.zip)

ustccw, 2017-12-21:
@BlueSkyai 烧写选项配置不正确。这些选项在 ESP8266 SDK 入门指南，第六节均有介绍。
你的截图3中，由于你配置的是 4MB,选项6。所以在选择 FLASH SIZE 时，应该选用 32Mbit-C1, 而不是截图中的 4Mbit,请知悉。

ustccw, 2017-12-21:
@BlueSkyai 你可以将你编译出来的bin文件放在这

BlueSkyai, 2017-12-21:
@ustccw 
[bin.zip](https://github.com/espressif/esp8266-aliyun/files/1578546/bin.zip)

BlueSkyai, 2017-12-21:
@ustccw 已经在线了，但是我用的串口工具74880波特率显示的仍是乱码
[image]

ustccw, 2017-12-21:
@BlueSkyai bin没有问题，log 如下：
[image]
[image]

ustccw, 2017-12-21:
@BlueSkyai 我对你们串口工具一无所知，请使用 secureCRT/XShell 等串口工具

BlueSkyai, 2017-12-21:
@ustccw 我用的是ESP8266 SDK 入门指南的Linux虚拟环境，我写入 miniterm.py /dev/ttyUSB3 74880 的时候出现了图中的问题
[image]

xuhongv, 2017-12-21:
@BlueSkyai 看你设置都没有进去到挂载模式啊亲

xuhongv, 2017-12-21:
@BlueSkyai 

请参考下篇博客：

Esp8266进阶之路11 企业者的福音之8266接入阿里智能，点亮一盏LED灯，期待天猫精灵语音控制的不约而至！(带Demo) - 徐宏的博客。 - CSDN博客 http://blog.csdn.net/xh870189248/article/details/78807018

ustccw, 2017-12-21:
@xuhongv 谢谢！

closed 2017-12-21 by ustccw