espressif/esp-aliyun #2: TLS连接疑点

BlueSkyai, 2017-12-17:
您好！
1.阿里云提供了标准CA证书（2048bit）和阿里云自签发的pubkey证书（1024位），他们是TLS版本的区别吗？
2.在阿里云官网有这么一段描述，设备端通过验证服务器的CA证书来进行连接，而设备发送信息通过MQTT的加密报文，那么在制作CA相关bin文件的时候，是不是只需要通过CA证书制作一个esp_ca_cert.bin证书烧录到8266，而无需制作esp_cert_private_key.bin,也就是阿里云服务器无需验证8266的证书？在代码端的espconn_secure_ca_enable（）在代码什么位置设置呢？
3.目前8266芯片现售的是几乎是16M或者32M，可否支持？烧录过程是否改可选？
4.我通过编译产生的文件，按照说明的位置烧录到8266，但是8266串口没有输出任何信息，波特率采用默认的74880
5.我试着将代码移植到安信可eclipse ide开发工具，因为编辑代码比较直观，乐鑫的工具使用有一定的门槛，是否有可能移植到eclipse开发环境当中？

ustccw, 2017-12-18:
@BlueSkyai 你好。
1,2: 考虑到内存问题，目前本仓库仅仅做 SSL/TLS 连接，而不支持单向认证和双向认证
3. 可否详述一下问题
4. 参考[ESP8266 对接阿里云指南](https://github.com/espressif/esp8266-aliyun)
5. 参考[ESP8266 对接阿里云指南](https://github.com/espressif/esp8266-aliyun)  ,遇到瓶颈问题请详细说明，我们会尽快处理。

BlueSkyai, 2017-12-18:
@ustccw 你好，ssl连接会再以后存在过期问题，那么如何更新证书信息呢？
 此外，编译文件./gen_misc.sh打开后，没有选择烧录条件的程序，一共有5个左右，从烧录的文件类型到内存大小，能提供选项就好了，目前默认是2M的，我的工具有4M的，还有最新的32M的，库函数能不能增加支持，以后8266几乎都是16或32M的

ustccw, 2017-12-19:
@BlueSkyai 
1. 目前暂不支持证书认证过程，走的是无认证的 TLS，所以不必更新证书信息。  
2. 默认编译选项是 1 1 2 0 5。你可以在 gen_misc.sh 文件中修改默认的编译选项，依次设置 boot,app,spi_speed,spi_mode,spi_size_map。
3. 后面的问题请详述一下，2M? spi size 2MB?库函数？

BlueSkyai, 2017-12-19:
@ustccw 目前深圳安信可集成了8266的外围电路，自己的布线工艺能力有限，有的是4层布线，所以直接购买模块，我把产品列表信息传过来，现在都是8M或32M
[image]

ustccw, 2017-12-19:
@BlueSkyai 32Mbit 是支持的，官方默认的 [ESP8266开发板ESP-Launcher](http://espressif.com/zh-hans/products/hardware/development-boards) 就是32Mbit, 请知悉。

BlueSkyai, 2017-12-20:
@ustccw 你好，这里可否添加32Mflash和nodemcu选项？
[image]

ustccw, 2017-12-21:
@BlueSkyai 以上是当前支持的，其余的暂且不支持。

BlueSkyai, 2017-12-21:
@ustccw 你好，用secureCRT依旧显示乱码
[image]
[image]

ustccw, 2017-12-21:
@BlueSkyai 官网和bbs均有乱码情况问题分析和处理。

closed 2017-12-21 by ustccw