espressif/esp-aliyun #4: 库程序编辑

BlueSkyai, 2017-12-22:
@ustccw 你好，我想在库程序的基础上根据自己的需求二次开发，我有几个问题
1.我用什么软件能直观地编辑，现在我只知道用文件夹的方式打开，无法直观子程序之间的关系，之前用eclipse IDE在windows系统下比较直观，Linux系统没有编辑过代码
2.之前切换串口都是取消uart.c中的一句 os_install_putc1注释，现在这样的办法在库程序中还有效吗？串口我需要和其他cpu通信，初看了一下程序通过串口打印信息的方式有很多，不知道有没有关联

BlueSkyai, 2017-12-23:
@ustccw 你好，我从uart.c中串口设置函数uart_setprintport( )修改了uart1为默认输出，但是仍是从uart0输出打印信息，请问，在此库函数中应当如何修改设置？我的理解就这么多了

BlueSkyai, 2017-12-26:
@ustccw 你好，请问库程序切换到串口1输出有没有什么办法？

BlueSkyai, 2017-12-26:
@igrr

ustccw, 2017-12-27:
@BlueSkyai 你好，先调用 system_uart_swap() 再调用 UART_SetPrintPort() 可切换串口，请知悉。

BlueSkyai, 2017-12-27:
@ustccw 编译没通过，undefined reference to `UART_SetPrintPort'，不知为什么
[image]

BlueSkyai, 2017-12-27:
@ustccw 你好，Linux环境也是同样问题
[image]

ustccw, 2017-12-27:
@BlueSkyai 你好，需要链接 libdriver.a , 请修改 esp8266-aliyun/Makefile。 如下:
[image]

BlueSkyai, 2017-12-27:
@ustccw 串口0有一些输出，不知道是不是8266通常上电打印的信息，然后串口1的输出就很少了，我显示的乱码，没有和默认输出一样，定时输出mqtt的消息
[image]
[image]

ustccw, 2017-12-27:
@BlueSkyai Hi, dear，上面说法有误，下面方法亲测有效:
[image]

BlueSkyai, 2017-12-27:
@ustccw thankyou very much！

ustccw, 2017-12-27:
my honor

closed 2017-12-27 by BlueSkyai