espressif/esp-aliyun #7: make: Nothing to be done for `FORCE'.

wangyanzhong, 2018-01-02:
编译阿里云的SDK make时提示这个错误，请问这是怎么回事呢？？？？？？

ustccw, 2018-01-03:
@wangyanzhong 你的编译器环境没有配置好，请根据 [README.md](https://github.com/espressif/esp8266-aliyun) 来配置你的开发环境。

wangyanzhong, 2018-01-03:
采用4.1 Window系统的方法2配置的环境，检查了一遍 xtensa-lx106-elf也是存在的，请问哪里还可能存在问题呢？

ustccw, 2018-01-03:
@wangyanzhong 查看编译器是不是在环境变量中：
```
echo $PATH
```
如果不在的话，需要导出，如果在的话，是不是你编译的路径或者脚本不对呢，你可以提供截图，我们协助分析。

wangyanzhong, 2018-01-03:
重新换了个环境编译了一下，报如下错误：
checking for GNU libc compatible realloc... yes
checking for size_t... yes
checking for working alloca.h... yes
checking for alloca... yes
checking libintl.h usability... yes
checking libintl.h presence... yes
checking for libintl.h... yes
checking whether gettext is declared... yes
checking for library containing bindtextdomain... none required
checking ncurses/ncurses.h usability... no
checking ncurses/ncurses.h presence... no
checking for ncurses/ncurses.h... no
checking ncurses/curses.h usability... no
checking ncurses/curses.h presence... no
checking for ncurses/curses.h... no
checking ncursesw/curses.h usability... no
checking ncursesw/curses.h presence... no
checking for ncursesw/curses.h... no
checking ncurses.h usability... no
checking ncurses.h presence... no
checking for ncurses.h... no
checking curses.h usability... no
checking curses.h presence... no
checking for curses.h... no
configure: error: could not find curses header, required for the kconfig frontends
make[1]: *** [_ct-ng] Error 1
make[1]: Leaving directory `~/esp-open-sdk/crosstool-NG'
make: *** [crosstool-NG/ct-ng] Error 2
到这无法解决了，请问这是什么问题呢

ustccw, 2018-01-03:
@wangyanzhong 请问，可以编译别的 ESP8266_RTOS_SDK/examples 下 demo 吗

ustccw, 2018-01-03:
@wangyanzhong 有问题多提供截图，文件，和自己的环境说明，以及操作流程等说明，我们也好定位，谢谢！

closed 2018-01-08 by ustccw