espressif/esp-aliyun #8: 串口中断响应

BlueSkyai, 2018-01-03:
@ustccw 
你好，我用其他CPU向串口0发送了一些数据，结果8266信号并未闪烁；
我把uart0_rx_intr_handler(void *para)中的uart_tx_one_char(UART0, ......);修改为了uart_tx_one_char(UART1, ......)，串口1通过调试软件也未打印出数据；
我分别用库程序和最新RTOS SDK试过了，现象相同，用以往的SDK通讯过，接受到信号的时候蓝灯闪烁，通过uart1打印了数据，这次不知为什么？

ustccw, 2018-01-03:
@BlueSkyai 你是想串口0接收数据 && 回复数据，串口1打印数据吗？

BlueSkyai, 2018-01-03:
@ustccw 是的，我之前有试过
[image]

ustccw, 2018-01-03:
@BlueSkyai 那你今天运气就比较好了。因为我刚好最近写了这个 demo, 见附件。基本可以工作，不过代码&文档&健壮性都不行，有时间会再整理。 操作说明见: README.md  
[UART0_UART1_demo.zip](https://github.com/espressif/esp8266-aliyun/files/1599597/UART0_UART1_demo.zip)

BlueSkyai, 2018-01-03:
@ustccw 谢谢~

BlueSkyai, 2018-01-03:
@ustccw 
我试了两种情况
1.uart0波特率设置为9600，uart1波特率默认，串口工具74880，结果每次单片机发送给8266，8266显示tout，并不显示发送内容
2.uart0 uart1都设置为9600，串口工具74880，我连续发送几次，在重启的时候显示了tout，同样不显示内容，这是什么原因呢？
[image]
[image]
[image]

ustccw, 2018-01-03:
@BlueSkyai 妈呀，驱动里面调试定位问题啊！ 看看究竟卡在哪一步呢。  
官网上有 [ESP8266 技术参考](http://espressif.com/zh-hans/support/download/documents?keys=&field_type_tid%5B%5D=14) ，里面有介绍ESP8266 的各个接口，包括功能，参数配置，函数说明，应用示例等内容。

BlueSkyai, 2018-01-03:
@ustccw 参数配置是对的，我发送的是01H，02H，03H，04H，8266中断接收到了，显示出了tout tout，应该显示一次啊，数据一共32位，也没超出范围，另外就是01H，02H，03H，04H为什么没有打印出来呢，即使波特率不对也应该是乱码呢？

BlueSkyai, 2018-01-03:
@ustccw 你好，我把uart0_rx_intr_handler(void *para)中的tout修改了，编译下载过后还是显示tout tout，没有显示修改后的t_out，这说明程序没有执行到uart0_rx_intr_handler(void *para)函数中，但是中断确实响应了，会不会是系统编译的问题？

BlueSkyai, 2018-01-05:
@ustccw Thanks a lot，it works

closed 2018-01-05 by BlueSkyai