espressif/esp-aliyun #9: 新建一个task，系统故障

BlueSkyai, 2018-01-10:
@ustccw 
你好，如题，我在usermain.c中create了一个task，然后系统串口输出出现乱码，系统直接无法运行了，我把task函数中的程序也屏蔽掉了，但是同样，不清楚为什么？
可不可帮助用xTaskCreate创建一个任务，让其运行起来，程序为空也可，库程序内容很多很复杂，不太清楚每一部分串口数据是从哪里打印的，尤其是变成了乱码，作为普通使用者更加难分析了

BlueSkyai, 2018-01-10:
[image]

ustccw, 2018-01-11:
@BlueSkyai xTaskCreate 使用参考本repo中 [user_main.c](https://github.com/espressif/esp8266-aliyun/blob/master/user/user_main.c)  
引起乱码原因很多，需要你一点点排查，比如你打印了一个中文标点，但串口工具当前不支持中文输出等。

BlueSkyai, 2018-01-11:
@ustccw 启用了HEAP_CHECK_TASK，结果乱码，这是原程序里的，什么原因呢
[image]

ustccw, 2018-01-11:
@BlueSkyai 你先试试整个 SDK 都不改，会不会乱码。如果会，建议把整个 flash 所有数据重新擦除，重新烧写。

BlueSkyai, 2018-01-11:
@ustccw 没注意烧错程序了，原程序no problem，我再新建试试

closed 2018-01-11 by BlueSkyai