espressif/esp-aliyun #10: 二值信号量xsemaphore，出现疑问

BlueSkyai, 2018-01-16:
@ustccw
你好，我在usermain.c中新建了一个二值信号量xsemaphore，调用了uart_int_new()，然后在uart.c的中断函数中新建了xsemaphoreGiveFromISR（），因为这一点，系统出现了故障，之前用队列的时候在中断这里也出现了问题，不知道什么原因，中断这里应怎么添加类似的程序呢？
[image]
[image]

ustccw, 2018-01-16:
@BlueSkyai 请参考《FREERTOS 实时内核实用指南》使用。
如果你有疑问，请附上完整的测试程序，测试步骤和测试环境，谢谢。

BlueSkyai, 2018-01-16:
@ustccw 谢谢，我一些会儿上传

BlueSkyai, 2018-01-16:
@ustccw 你好，程序很简单，就是首先创建二值向量，然后在驱动uart.c的串口中断中，当有数据从串口0输入就发出这个二值向量信号，我编译通过了，但是下载后出现了乱码，而且我试下来，可以确定乱码原因是由于串口中断这里添加了发出二值向量的程序，从程序上我看不出哪里有错误，附上代
[test.zip](https://github.com/espressif/esp8266-aliyun/files/1634261/test.zip)
码

BlueSkyai, 2018-01-16:
@ustccw  二值向量创建成功了，可以看到打印successful，但是串口中断的简单程序导致了乱码，程序是按照freeRTOS内核说明去写的
[image]

BlueSkyai, 2018-01-16:
@ustccw test文件夹里的内容您可直接替换到库程序里，然后MQTT的部分可以先屏蔽
#if 0
    switch (DEFAULT_TASK_MODE) {
    case MQTT_TASK:
        xTaskCreate(mqtt_proc, "mqtt_proc", 2048, NULL, 5, NULL);
        os_printf("\nMQTT task started...\n");
        break;

    case LOCAL_OTA_TASK:
        xTaskCreate(local_ota_proc, "local_ota_proc", 2048, NULL, 5, NULL);
        os_printf("\nlocal OTA task started...\n");
        break;

    case REMOTE_OTA_TASK:
        xTaskCreate(remote_ota_proc, "ota_proc", 2048, NULL, 5, NULL);
        os_printf("\nremote OTA task started...\n");
        break;

    default:
        os_printf("task mode error happend!\n");
        break;
    }
#endif

ustccw, 2018-01-16:
@BlueSkyai 
不清楚你是不是编译了 driver。也不知道你其他地方改了啥。
我把自己成功的代码上传了，见 [云盘](https://pan.baidu.com/s/1jJnyaOm)

成功log如下：
[image]

BlueSkyai, 2018-01-16:
@ustccw 但是不编译driver，用串口0就接受不到数据，也就是无法进入到uart0 rx intr handler，把接受到的数据打印出来，编译了driver就有乱码了

BlueSkyai, 2018-01-16:
@ustccw 按理编译了driver程序不应该乱啊，那样串口数据就无法处理了

ustccw, 2018-01-16:
@BlueSkyai 你的环境配置可能有些问题，用我云盘分享的代码就可以了。

BlueSkyai, 2018-01-16:
@ustccw 你的log里打印了tout ，是从串口0输入了数据，数据没有打印出来啊

BlueSkyai, 2018-01-16:
@ustccw 你可以试一下编译driver，然后从串口0输入数据，就能有显示了，但是这样如果把freeRTOS的内容加入进去的话会产生问题

ustccw, 2018-01-16:
@BlueSkyai 我提供的代码就是从串口 0 输入数据啊。我连续输入了 三个1啊！不然怎么会触发三次中断处理。

BlueSkyai, 2018-01-16:
@ustccw printf("tout\r\n");
            fifo_len = (READ_PERI_REG(UART_STATUS(UART0)) >> UART_RXFIFO_CNT_S)&UART_RXFIFO_CNT;
            buf_idx = 0;

            while (buf_idx < fifo_len) {
                uart_tx_one_char(UART0, READ_PERI_REG(UART_FIFO(UART0)) & 0xFF);
                buf_idx++;
            }
这段代码的uart_tx_one_char(UART0, READ_PERI_REG(UART_FIFO(UART0)) & 0xFF);就是打印数据了

BlueSkyai, 2018-01-16:
@ustccw 不信你可以编译一下driver看有没有输出呢

ustccw, 2018-01-16:
@BlueSkyai 不明白你的意思。

ustccw, 2018-01-16:
三个1就是输出啊，我的串口工具没有回显。

BlueSkyai, 2018-01-18:
@ustccw 我只不过是偶尔看到了，并不确定原因，请问有进展了吗？

BlueSkyai, 2018-01-18:
@ustccw 附上编译driver之后，其他cpu向8266发送内容，tout字符后面乱码的内容就是数据了
[image]

BlueSkyai, 2018-01-18:
@ustccw 其实就是01H 02H 03H 04H，重复发送重复接受到了
while (buf_idx < fifo_len) {
            uart_tx_one_char(UART  **1**  , READ_PERI_REG(UART_FIFO(UART0)) & 0xFF);
            buf_idx++;
        }

BlueSkyai, 2018-01-18:
@ustccw 几个办法都试过了，貌似是一个库程序的BUG，若无法解决，驱动可能无法用来处理数据

ustccw, 2018-01-26:
@BlueSkyai 本 issue 是解决 crash 问题的，如果觉得驱动有 bug,可通过官网来获取更快的技术支持。[链接](http://espressif.com/zh-hans/company/contact-extra/technical-inquiries-software)

BlueSkyai, 2018-01-26:
@ustccw 感谢你的耐心支持！

closed 2018-01-26 by BlueSkyai