espressif/esp-aliyun #11: 串口中断接受数据存储到8266并打印

BlueSkyai, 2018-01-17:
@ustccw 你好，其实我昨天做了一个简单的测试，串口接受的数据提取到一个变量Rxbuffer中，然后新建一个任务，通过任务的方式查询这个变量，然后把Rxbuffer中保存的数据打印出来，数据要能被内部程序处理才能进一步编程，但效果是这样的，回显并不代表了数据就存储到8266里了
[image]
[image]
[image]
[test.zip](https://github.com/espressif/esp8266-aliyun/files/1637470/test.zip)

BlueSkyai, 2018-01-17:
@ustccw  从上图可以看到回显，程序在最下方，但是receive的并没有任何显示，我进过一段时间观察也是这个效果
[image]

BlueSkyai, 2018-01-17:
@ustccw 上图是usermian.c中新建的task，隔一段时间查询一次，后续我会附上编译driver，然后其他单片机发送给8266的数据显示出来了，然后不编译就没有运行到驱动里，也就打印不出数据，并不是通过电脑的串口工具，这个测试并不直观，感谢！

BlueSkyai, 2018-01-17:
@ustccw 在8266-rtos-sdk论坛里也有人提到类似驱动使用的问题，也许有可能是makefile的问题，烦请解决

BlueSkyai, 2018-01-19:
@ustccw 你好，有在关注这个问题吗？或者还是我哪里需要重新理解

BlueSkyai, 2018-01-19:
@ustccw 你好，是在解决这个问题吗？我有点迫不及待看到结果了，不知道等多久

BlueSkyai, 2018-01-22:
@ustccw 有没有解决办法？如果没有freeRTOS内核或许不会遇到这个问题

ustccw, 2018-01-22:
@BlueSkyai 你好，因上周公司 Annual Gathering，问题会有些延误。我会在这周尾来处理你的 issue，谢谢。

ustccw, 2018-01-26:
@BlueSkyai 请问下面这句是什么意思？
```
os_printf("Receive: s%\n",RxBuffer);
```

BlueSkyai, 2018-01-26:
@ustccw 是想把全局变量RxBufeer的内容打印出来

ustccw, 2018-01-26:
@BlueSkyai 
```
你确定这样能打印出来？
```

BlueSkyai, 2018-01-26:
@ustccw 好像是打印不出来

BlueSkyai, 2018-01-26:
@ustccw 你有办法吗？

ustccw, 2018-01-26:
@BlueSkyai 试试下面这种打印方式：
```
os_printf("Receive: %s\n",RxBuffer);
```

closed 2018-01-26 by BlueSkyai