espressif/esp-aliyun #13: OTA TASK运行失败，但MQTT任务正常

garyyeh, 2018-03-05:
我使用的是安信可的ESP 01-M模块，在https://github.com/espressif/esp8266-aliyun?spm=5176.doc42648.2.9.Oe0lw3 下载了由乐鑫官方整合阿里云物联网套件的程序，然后在乐鑫官网下载了RTOS，放到了esp8266-rtos-sdk 目录下，修改了阿里云套件的三元组，产品KEY、设备名、设备密钥。使用MQTT_TASK任务运行没有什么问题，阿里云也可以接收到数据。设备也可以接收到云发送过来的数据。
然后我就把DEFAULT_TASK_MODE改为REMOTE_OTA_TASK测试一下OTA升级，但这时就出问题了。我把所有的LOG开关打开（LOG在附件里）。发现出现异常
[inf] _ssl_parse_crt(142): crt content:451
[inf] _ssl_client_init(185): ok (0 skipped)
[inf] TLSConnectNetwork(360): Connecting to /iot-auth.cn-shanghai.aliyuncs.com/443...
[inf] TLSConnectNetwork(367): ok
[inf] TLSConnectNetwork(372): . Setting up the SSL/TLS structure...
[inf] TLSConnectNetwork(383): ok
[inf] TLSConnectNetwork(420): Performing the SSL/TLS handshake...
Fatal exception (28):
epc1=0x401000e3
epc2=0x00000000
epc3=0x4025ea17
epcvaddr=0x00000068
depc=0x00000000
rtn_add=0x401003d4

但是MQTT任务和OTA任务走的是一样的流程，不知道为什么MQTT任务能正常运行，而OTA就出错，设备不断重启的样子，不断地走同样的流程，找了阿里云的技术支持，他们说这个错误是芯片这边抛出的异常。于是来到论坛请求支持，希望大神给予帮助,感激不尽！

固件FLASH为1MB。芯片应该是8285

我也在乐鑫官方论坛提交了相关的TOPIC，但依然没得到解决方案。
乐鑫主题地址：https://bbs.espressif.com/viewtopic.php?f=7&t=9138

希望大神能帮助解决一下。谢谢了。

garyyeh, 2018-03-05:
乐鑫主题地址：https://bbs.espressif.com/viewtopic.php?f=7&t=9138
在论坛贴子里，有我上传的日志文件。

ustccw, 2018-03-05:
@garyyeh 你好，感谢你的反馈。
当前仓库不支持 REMOTE_OTA_TASK，但支持 LOCAL_OTA_TASK。
阿里的远程 OTA 会在接下来的工作计划中，请知悉。

garyyeh, 2018-03-06:
@ustccw
感谢您的回复，请问REMOTE_OTA_TASK预计什么时候可以出来？智能硬件缺少远程升级的话，在修复BUG和增加功能上很不方便。

另外LOCAL_OTA_TASK的意思是，我使用APP向硬件传输新固件，然后升级吗？

谢谢！

ustccw, 2018-03-06:
@garyyeh 你好，remote ota 基本功能预计最近一个月内。
local ota 是指通过非阿里云进行固件升级。你可以将要升级的固件放在某个 http server 上，然后修改 ota.h 中 `LOCAL_OTA_SERVER_IP` 和 `LOCAL_OTA_SERVER_PORT`, 即可主动完成固件升级。

garyyeh, 2018-03-06:
@ustccw  谢谢回复，我会尝试使用LOCAL_OTA_TASK作为升级方式，并持续关注GITHUB上的更新。谢谢了。

ustccw, 2018-03-06:
感谢你对 ESP 的关注！

ustccw, 2018-03-18:
@garyyeh Hi, remote ota 基本完成，请知悉。

garyyeh, 2018-03-18:
@ustccw 太好了，我会尽快测试，感谢您的通知。