espressif/esp-aliyun #14: 编译这个SDK的时候一直出错

git4xxx, 2018-03-07:
我的电脑系统是Win7，使用的开发工具是Cygwin+Eclipse, 曾经单独编译过ESP8266-RTOS-SDK项目，没有问题，但是编译这个esp8266-aliyun项目一直提示：
no rule to make target 'all'

这到底是我的的开发环境设置有问题还是别的原因？

ustccw, 2018-03-07:
@git4xxx 你好，你的 submodule 应该没有 clone 下来。根据 README.md 第五步重新 clone 子模块。
```
git submodule update --init --recursive
```

git4xxx, 2018-03-07:
git submodule update --init --recursive 这个指令执行过了的，就是下载rtossdk嘛，还是不行啊

git4xxx, 2018-03-07:
@ustccw git submodule update --init --recursive 这个指令执行过了的，就是下载rtossdk嘛，还是不行啊

ustccw, 2018-03-07:
@git4xxx 
1. 进入 esp8266-rtos-sdk 文件夹，确认文件是否被 clone 下来。
2. 编译器路径是否设置到环境变量里呢？如果没有，设置一下：
```
$ export PATH=/opt/xtensa-lx106-elf/bin/:$PATH
```

git4xxx, 2018-03-07:
@ustccw 
我确认 esp8266-rtos-sdk 文件夹 里边的文件是已经成功clone下来了的；
编译器的路径也设置了，还是解决不了。
现象连接：https://zhidao.baidu.com/question/988448181955327739.html

ustccw, 2018-03-08:
@git4xxx 请问问题解决了吗

git4xxx, 2018-03-08:
@ustccw 还没呢，对Cygwin不熟，折腾了一个下午没搞定。今天有别的工作要做，就暂时不折腾了

ustccw, 2018-03-08:
@git4xxx 好的，有问题再反馈

git4xxx, 2018-03-11:
@ustccw  解决了，需要将脚本文件改成UNIX格式的，下载下来的脚本文件有几行的换行符是 \r\n的

ustccw, 2018-03-12:
@git4xxx 好

closed 2018-03-12 by ustccw