espressif/esp-aliyun #15: MQTT 连接不稳定

git4xxx, 2018-03-11:
代码运行之后，总是在publish几个或者几十个消息之后MQTT连接就出现了如下的错误提示：

event_handle|47 MQTT disconnect.

[inf] utils_network_ssl_disconnect(319): ssl_disconnect
mqtt_client|211 error occur when publish

[inf] utils_network_ssl_write(295): ssl write fail, code=-28928, str=
[inf] HAL_SSL_Destroy(465): handle is NULL
[ALIYUN] MQTT client example end, free heap size:40144

我不知道是不是我这边网络的问题是别的什么，我感觉我这边的网络还可以，因为电脑和ESP8266连接的是同一个Wifi路由，电脑就没掉线过。
ESP8266跟Wifi路由器之间的连接是没有断开，但是MQTT连接老中断，MQTT重连
之后又可以继续发消息了，这是为什么？

ustccw, 2018-03-12:
@git4xxx 在以前是可以连续运行的；最近我们也遇到此情况，原因还在定位，可能是SDK需要更新，我们会尽快修复的，感谢关注。

garyyeh, 2018-03-13:
我在年前因为OTA的问题，也经常测试，但没有发现断网的情况，可以连续运行。有可能是最近才出现的问题。

dstian, 2018-03-18:
请问这个问题是因为内存泄漏吗？阿里帮助文档里面的V2.03是否已经解决这个问题？

git4xxx, 2018-03-18:
@dstian 还没解决，不是内存泄漏

KaneOne, 2018-04-01:
@ustccw 那要自己侦测MQTT连接情况，然后重连吗

git4xxx, 2018-04-04:
@liefyuan 我是在Windows环境编译的，使用Cygwin，木有问题

wanfangbing, 2018-04-08:
@ustccw 这个问题有修复的时间点吗

wanfangbing, 2018-04-09:
@ustccw 这个issue的状态有更新吗？请帮忙回复一下，多谢。

ustccw, 2018-04-09:
@wanfangbing 有计划修复，修复时间待定。如果您那边有订单和项目，请联系官网商务对接，我们会尽快处理，请知悉。

wanfangbing, 2018-04-09:
@ustccw 我们是有项目要使用8266芯片接入到阿里云，现在要做评估，这个问题有没有临时的解决方案以便我们进行评估，如果MQTT连接不稳定，项目不好往下进行。

ustccw, 2018-04-09:
@wanfangbing 您好，我们这一两周会深入分析定位并修复 MQTT 连接不稳定问题。

ustbzm, 2018-05-10:
@ustccw   目前mqtt连接不稳定的问题解决了吗？

mygreatnet, 2018-05-17:
目前做评估，也遇到同样的问题，先是几帧到几十帧都是正常的，然后就没有反应了，过较长时间，出现重连。具体 打印信息如下：

packet-id=10, publish topic msg={"attr_name":"temperature", "attr_value":"0"}

publish success, packet-id=10

********** arrive topic [len:38] start addr:40108c84 **********
/dTMFPnxRNIx/QU9WBiQkA8XgPOqRDIw4/data
---------- arrive topic End ----------

********** arrive payload [len:45] start addr:40108cac **********
{"attr_name":"temperature", 20 "attr_value":"0"}
---------- arrive payload End ----------
[inf] utils_network_ssl_read(254): ssl recv error: code = -30848, err_str = ''
[log_err] iotx_mc_keepalive(2127): network is disconnected!
MQTT disconnect.
[inf] utils_network_ssl_disconnect(319): ssl_disconnect
[log_err] iotx_mc_publish(1763): mqtt client state is error,state = 4
error occur when publish
[log_err] iotx_mc_unsubscribe(1726): mqtt client state is error,state = 4
[log_err] iotx_mc_unsubscribe(1726): mqtt client state is error,state = 4
[inf] utils_network_ssl_write(295): ssl write fail, code=-28928, str=
[inf] HAL_SSL_Destroy(465): handle is NULL
MQTT client example end, free heap size:60824
MQTT client example begin, free heap size:60824
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!step1:MQTT init!!111[inf] _ssl_client_init(175): Loading the CA root certificate ...
cert. version     : 3
serial number     : 04:00:00:00:00:01:15:4B:5A:C3:94
issuer name       : C=BE, O=GlobalSign nv-sa, OU=Root CA, CN=GlobalSign Root CA
subject name      : C=BE, O=GlobalSign nv-sa, OU=Root CA, CN=GlobalSign Root CA
issued  on        : 1998-09-01 12:00:00
expires on        : 2028-01-28 12:00:00
signed using      : RSA with SHA1
RSA key size      : 2048 bits
basic constraints : CA=true
key usage         : Key Cert Sign, CRL Sign
[inf] _ssl_parse_crt(142): crt content:451
[inf] _ssl_client_init(185):  ok (0 skipped)
[inf] TLSConnectNetwork(360): Connecting to /dTMFPnxRNIx.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...
[inf] TLSConnectNetwork(367):  ok
[inf] TLSConnectNetwork(372):   . Setting up the SSL/TLS structure...
[inf] TLSConnectNetwork(383):  ok
[inf] TLSConnectNetwork(420): Performing the SSL/TLS handshake...
[inf] TLSConnectNetwork(429):  ok
[inf] TLSConnectNetwork(433):   . Verifying peer X.509 certificate..
[inf] _real_confirm(85): certificate verification result: 0x00
packet-id=2, publish topic msg={"attr_name":"temperature", "attr_value":"0"}

subscribe success, packet-id=1
publish success, packet-id=2

********** arrive topic [len:38] start addr:40108c84 **********
/dTMFPnxRNIx/QU9WBiQkA8XgPOqRDIw4/data
---------- arrive topic End ----------

********** arrive payload [len:45] start addr:40108cac **********
{"attr_name":"temperature", 20 "attr_value":"0"}
---------- arrive payload End ----------

sofwu, 2018-05-25:
@ustccw目前mqtt连接不稳定修复了吗？？

titawork, 2018-07-28:
这个问题好像没有修复嘛，还是会出这个问题，而且会导致我的线程阻塞，看起来问题还是很严重，最诡异的是，接收是好的。见我最新提的问题

Eavo, 2019-01-09:
您好，SDK 已更新，支持 aliyun v2.30，详情请查阅：#88