espressif/esp-aliyun #18: 我在git clone 这个代码修改了WiFi的两个参数和阿里云产品三元组，然后使用cygwin和ESP8266_Lubuntu编译都出现了make: *** No rule to make target 'clean'. Stop, 该怎么解决呢？

liefyuan, 2018-04-04:
这是ESP_Lunbuntu的：
esp8266@esp8266-VirtualBox:~/Desktop/esp8266-aliyun$ ls
bin  esp8266-rtos-sdk  gen_misc.sh  include  Makefile  mqtt  ota  platforms  README.md  _static  user
esp8266@esp8266-VirtualBox:~/Desktop/esp8266-aliyun$ ./gen_misc.sh 
/home/esp8266/Desktop/esp8266-aliyun/esp8266-rtos-sdk
/home/esp8266/Desktop/esp8266-aliyun/bin
gen_misc.sh version 20150911

SDK_PATH:
/home/esp8266/Desktop/esp8266-aliyun/esp8266-rtos-sdk

BIN_PATH:
/home/esp8266/Desktop/esp8266-aliyun/bin

Please check SDK_PATH & BIN_PATH, enter (Y/y) to continue:

Please follow below steps(1-5) to generate specific bin(s):
STEP 1: use boot_v1.2+ by default
boot mode: new

STEP 2: choose bin generate(0=eagle.flash.bin+eagle.irom0text.bin, 1=user1.bin, 2=user2.bin)
enter (0/1/2, default 0):

STEP 3: choose spi speed(0=20MHz, 1=26.7MHz, 2=40MHz, 3=80MHz)
enter (0/1/2/3, default 2):
spi speed: 40 MHz

STEP 4: choose spi mode(0=QIO, 1=QOUT, 2=DIO, 3=DOUT)
enter (0/1/2/3, default 0):
spi mode: QIO

STEP 5: choose spi size and map
    0= 512KB( 256KB+ 256KB)
    2=1024KB( 512KB+ 512KB)
    3=2048KB( 512KB+ 512KB)
    4=4096KB( 512KB+ 512KB)
    5=2048KB(1024KB+1024KB)
    6=4096KB(1024KB+1024KB)
    7=4096KB(2048KB+2048KB) not support ,just for compatible with nodeMCU board
    8=8192KB(1024KB+1024KB)
    9=16384KB(1024KB+1024KB)
enter (0/2/3/4/5/6/7/8/9, default 0):
spi size: 1024KB
spi ota map:  512KB + 512KB

start...

make: *** No rule to make target `clean'.  Stop.
make: Nothing to be done for `FORCE'.
esp8266@esp8266-VirtualBox:~/Desktop/esp8266-aliyun$ make clean
make: *** No rule to make target `clean'.  Stop.
esp8266@esp8266-VirtualBox:~/Desktop/esp8266-aliyun$ make -f Makefile clean
make: *** No rule to make target `clean'.  Stop.

liefyuan, 2018-04-04:
这是在cygwin上的：
$ ./gen_misc.sh
/home/esp8266-aliyun-master/esp8266-rtos-sdk
/home/esp8266-aliyun-master/bin
gen_misc.sh version 20150911

SDK_PATH:
/home/esp8266-aliyun-master/esp8266-rtos-sdk

BIN_PATH:
/home/esp8266-aliyun-master/bin

Please check SDK_PATH & BIN_PATH, enter (Y/y) to continue:

Please follow below steps(1-5) to generate specific bin(s):
STEP 1: use boot_v1.2+ by default
boot mode: new

STEP 2: choose bin generate(0=eagle.flash.bin+eagle.irom0text.bin, 1=user1.bin, 2=user2.bin)
enter (0/1/2, default 0):

STEP 3: choose spi speed(0=20MHz, 1=26.7MHz, 2=40MHz, 3=80MHz)
enter (0/1/2/3, default 2):
spi speed: 40 MHz

STEP 4: choose spi mode(0=QIO, 1=QOUT, 2=DIO, 3=DOUT)
enter (0/1/2/3, default 0):
spi mode: QIO

STEP 5: choose spi size and map
    0= 512KB( 256KB+ 256KB)
    2=1024KB( 512KB+ 512KB)
    3=2048KB( 512KB+ 512KB)
    4=4096KB( 512KB+ 512KB)
    5=2048KB(1024KB+1024KB)
    6=4096KB(1024KB+1024KB)
    7=4096KB(2048KB+2048KB) not support ,just for compatible with nodeMCU board
    8=8192KB(1024KB+1024KB)
    9=16384KB(1024KB+1024KB)
enter (0/2/3/4/5/6/7/8/9, default 0):
spi size: 1024KB
spi ota map:  512KB + 512KB

start...

make: *** No rule to make target 'clean'。 停止。
make: Nothing to be done for 'FORCE'.

yuanlifu@liefyuan /home/esp8266-aliyun-master
$ make -f Makefile clean
make: *** No rule to make target 'clean'。 停止。

liefyuan, 2018-04-04:
网上找到的方法我都试过：
网上的方法一：$ make -f Makefile clean
make: *** No rule to make target 'clean'. Stop

网上的方法二：
$ make clean
make: *** No rule to make target clean'. Stop.

ustccw, 2018-04-04:
@liefyuan 应该是配置没有配好，可否方便给个 teamviewer 号，远程您的电脑。

wangqinx, 2018-04-04:
刚才我发的帖子找到办法自己关闭了，怕你看不见这里重发一下，解决方法如下：

从乐鑫官网下载基于RTOS的SDK，然后解压到阿里云SDK的/esp8266-rtos-sdk 目录下即可顺利生成固件镜像。

ustccw, 2018-04-04:
@wangqinx @liefyuan 不需要下载官网RTOS的SDK, 本 repo 中 submodule 含有 SDK。
可按照如下操作:
```
$ cd esp8266-aliyun
$ git submodule update --init --recursive
$ ./gen_misc.sh
```

wangqinx, 2018-04-04:
@ustccw  您好，执行
$ git submodule update --init --recursive
的时候，显示 fatal ，是何原因呢

具体： fatal：not a git repository (or any of the parent directories) .git

ustccw, 2018-04-04:
@wangqinx 
```
$ git remote -v
```
看是不是 http 的，这个应该是 git 配置问题。

liefyuan, 2018-04-04:
好了那个问题已经解决了

closed 2018-04-04 by liefyuan

24540477zxc, 2018-08-29:
请问最后是怎么解决的呢