espressif/esp-aliyun #20: 把编译好的固件下载到ESP8266里面去了，已经连上wifi，但是现在一直是did not get a valid time from sntp server please start sntp first ! current time : Thu Jan 01 00:00:00 1970

liefyuan, 2018-04-04:
dhcp client start...
ip:192.168.1.179,mask:255.255.255.0,gw:192.168.1.1
WiFi connected
Initializing SNTP
please start sntp first !
current time : Thu Jan 01 00:00:00 1970

did not get a valid time from sntp server
please start sntp first !
current time : Thu Jan 01 00:00:00 1970

did not get a valid time from sntp server
please start sntp first !
current time : Thu Jan 01 00:00:00 1970

did not get a valid time from sntp server
please start sntp first !
current time : Thu Jan 01 00:00:00 1970

did not get a valid time from sntp server
u Jan 01 00:00:00 1970

did not get a valid time from sntp server
please start sntp first !
current time : Thu Jan 01 00:00:00 1970

did not get a valid time from sntp server
please start sntp first !
current time : Thu Jan 01 00:00:00 1970

liefyuan, 2018-04-04:
@ustccw

liefyuan, 2018-04-04:
@ustccw 求教怎么解决？？

ustccw, 2018-04-04:
@liefyuan 根据您的地址修改代码中 sntp server服务器。或者修改 `#define START_SNTP 0`

liefyuan, 2018-04-04:
去网上找一个sntp server服务器地址吗？

liefyuan, 2018-04-04:
@ustccw 现在我把 #define START_SNTP 0，就没有出现这个问题了，但是在与阿里云消息对接的时候又出现了  》》 失败:No authorization，这又该怎么办呢？

ustccw, 2018-04-09:
@liefyuan 可以去网上找适合自己地区的 sntp server

ustccw, 2018-04-09:
@liefyuan 如果关闭 sntp, 则可能会影响 SSL 握手中校验证书有效期，而导致握手失败。