espressif/esp-aliyun #31: failed ! net_connect returned -0x0052

seblee, 2018-05-26:
MQTT client example begin, free heap size:42064
[inf] _ssl_client_init(175): Loading the CA root certificate ...
cert. version     : 3
serial number     : 04:00:00:00:00:01:15:4B:5A:C3:94
issuer name       : C=BE, O=GlobalSign nv-sa, OU=Root CA, CN=GlobalSign Root CA
subject name      : C=BE, O=GlobalSign nv-sa, OU=Root CA, CN=GlobalSign Root CA
issued  on        : 1998-09-01 12:00:00
expires on        : 2028-01-28 12:00:00
signed using      : RSA with SHA1
RSA key size      : 2048 bits
basic constraints : CA=true
key usage         : Key Cert Sign, CRL Sign
[inf] _ssl_parse_crt(142): crt content:451
[inf] _ssl_client_init(185):  ok (0 skipped)
[inf] TLSConnectNetwork(360): Connecting to a1WxVtI1pGm.iot-as-mqtt.cn-shanghai.aliyuncs.com:1883...
[inf] TLSConnectNetwork(363):  failed ! net_connect returned -0x0052
[inf] HAL_SSL_Destroy(465): handle is NULL
[log_err] iotx_mc_connect(2254): TCP or TLS Connection failed
[inf] utils_network_ssl_write(295): ssl write fail, code=-28928, str=
[inf] HAL_SSL_Destroy(465): handle is NULL
MQTT construct failed
MQTT client example end, free heap size:42064

seblee, 2018-05-26:
merge了最新版本的master后,程序就崩溃了

一直就重启重启
did not get a valid time from sntp server
current time : Sat May 26 16:47:42 2018

MQTT client example begin, free heap size:60640
[inf] iotx_device_info_init(40): device_info created successfully!
[dFatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0<0xa2>@*r<0x85><0xc4>P<0xb1><0xb5><0xb1>K<0xed>t<0xa5>dr<0xd7>.@<0xe9><0xf9>.<0xa9><0xd1><0xaa><0xa1>Ko8a /t<0xb5>$v'<0xe1><0xea><0xc9>*<0xb1>T<0xb1><0xb1><0xb1><0xa9><0xa9><0xc4>P<0x8b>,<0xa9><0xb5><0xf9>*<0xb1><0x85><0xa2><0xfc>nzAl )<0xe9>m~<0xb1>v%!<0xf9>(P<0xb1>+t<0xc7>,<0xb1>@j|<0xbd><0xf5>*<0xb5><0xf9>*<0xb1><0x89><0xf8>nzAl )<0xe9>m~<0xb1>X%!<0xf9>(P<0xb1>+t<0xc7>m<0xb1>@j|<0xbd><0xab>*<0xb5><0xf9>*<0xb1><0x89><0xf8>nzAl )<0xe9>m~<0xb1>t<0x89>8r<0xa7>(<0xb1><0xc7><0xa2>8<0xa1><0xad><0x87><0xe9><0xf9>.<0xbd><0x04>r<0xf9>+<0xb5><0xb1>P<0xa2>@<0xe8>b<0x05><0x95>v<0xa7>(<0xa1><0xa1><0xb5><0xb9><0x89><0xe1><0xa2>@<0xe8>b<0x05><0x89><0xd1>*<0xa1><0xa1><0xa1><0xb5><0x91><0xe5>N<0x08><0xe8>b<0x05><0x8d><0xc9>-Ae<0xe1>*<0xad><0x89><0x04>X<0xb5>[p<0xd1>h<0xb1><0x8b><0xa1>Z<0xb1><0x8b><0xa1>Z i~P<0xb7><0xa8>vA<0xee>v<0x9b>(P<0xb1><0xb1><0x81><0xa2><0xfc>OS SDK ver: 2.0.0(e271380) compiled @ Mar 30 2018 18:54:06
phy ver: 1055_1, pp ver: 10.7

ustccw, 2018-05-26:
-0x0052 代表 DNS解析失败，可能和地区有关，可以向阿里同事咨询其技术细节
```
#define MBEDTLS_ERR_NET_UNKNOWN_HOST                      -0x0052  /**< Failed to get an IP address for the given hostname. */
```

closed 2018-05-27 by seblee

sofwu, 2018-06-10:
@seblee 我能加你qq吗，有些问题请教你？？

erfengwelink, 2018-09-25:
> 
MQTT client example begin, free heap size:42064
[inf] _ssl_client_init(175): Loading the CA root certificate ...
cert. version : 3
serial number : 04:00:00:00:00:01:15:4B:5A:C3:94
issuer name : C=BE, O=GlobalSign nv-sa, OU=Root CA, CN=GlobalSign Root CA
subject name : C=BE, O=GlobalSign nv-sa, OU=Root CA, CN=GlobalSign Root CA
issued on : 1998-09-01 12:00:00
expires on : 2028-01-28 12:00:00
signed using : RSA with SHA1
RSA key size : 2048 bits
basic constraints : CA=true
key usage : Key Cert Sign, CRL Sign
[inf] _ssl_parse_crt(142): crt content:451
[inf] _ssl_client_init(185): ok (0 skipped)
[inf] TLSConnectNetwork(360): Connecting to a1WxVtI1pGm.iot-as-mqtt.cn-shanghai.aliyuncs.com:1883...
[inf] TLSConnectNetwork(363): failed ! net_connect returned -0x0052
[inf] HAL_SSL_Destroy(465): handle is NULL
[log_err] iotx_mc_connect(2254): TCP or TLS Connection failed
[inf] utils_network_ssl_write(295): ssl write fail, code=-28928, str=
[inf] HAL_SSL_Destroy(465): handle is NULL
MQTT construct failed
MQTT client example end, free heap size:42064
> 
@seblee I have meet the same issue with you ? have U already solved it? waiting you reply anytime. thanks!

erfengwelink, 2018-09-25:
@ustccw 您好，刚刚看到您的回复 "代表 DNS解析失败，可能和地区有关" 我现在遇到的情况是在深圳DNS没问题，到山东工厂调试，直接DNS失败，但是现场手机热点却可以连上aliyun MQTT broker，大神，我应该怎么排查，期待您的指导！

ustccw, 2018-09-25:
@erfengwelink 可尝试下面方案调试：
1. 或许和路由器配置有关系，在路由器管理页面查看 DNS 设置是否正确
2. 在该路由器下通过电脑 ping 对应域名，查看是否可以返回正确 IP 地址
3. 可通过在线域名解析查看当前待解析域名对应的 IP 地址 `http://tool.chinaz.com/dns/?type=1&host=&ip=`
4. 可尝试在代码中添加 log 调试，去掉 DNS 解析，直接使用3中的 IP 地址去连接看是否 OK

erfengwelink, 2018-10-10:
@ustccw 
”1.或许和路由器配置有关系，在路由器管理页面查看 DNS 设置是否正确“
>>进入路由器管理页面后，发现该路由是低配版tp-link，只有傻瓜式的几种常见设置，并没有开放DNS设置给用户。

”2.在该路由器下通过电脑 ping 对应域名，查看是否可以返回正确 IP 地址“
>>现场路由器下能成功解析相应的域名。

”3.可通过在线域名解析查看当前待解析域名对应的 IP 地址 http://tool.chinaz.com/dns/?type=1&host=&ip=“
>>待进入现场进行调试
”4.可尝试在代码中添加 log 调试，去掉 DNS 解析，直接使用3中的 IP 地址去连接看是否 OK“
>>待进入现场进行调试

BTW，前两天和阿里工程师沟通得到以下回复：

”工程师 62135 号 :    ‘如果您能明确是“监听打印信息发现是DNS解析阿里云域名失败所致（监听到的打印消息见附件）”，请向山东津宁现场wifi网络运营商投诉解析问题。’  “

十分谢谢您的详细指导！

erfengwelink, 2018-11-28:
@ustccw 
您好， 通过同路由器下PC ping 结果如下：
# ping a1Ugih4wMVP.iot-as-mqtt.cn-shanghai.aliyuncs.com

正在 Ping a1ugih4wmvp.iot-as-mqtt.cn-shanghai.aliyuncs.com [139.196.135.135] 具有 32 字节的数据:
来自 139.196.135.135 的回复: 字节=32 时间=28ms TTL=36
来自 139.196.135.135 的回复: 字节=32 时间=26ms TTL=36
来自 139.196.135.135 的回复: 字节=32 时间=31ms TTL=36
来自 139.196.135.135 的回复: 字节=32 时间=32ms TTL=36

139.196.135.135 的 Ping 统计信息:
    数据包: 已发送 = 4，已接收 = 4，丢失 = 0 (0% 丢失)，
往返行程的估计时间(以毫秒为单位):
    最短 = 26ms，最长 = 32ms，平均 = 29ms

在线域名解析查看当前待解析域名对应的 IP 地址 如下： 浙江杭州与PC ping 结果吻合。
上海[联通]
106.15.83.29 [北京市 CNISP会员(尚未分配)]
106.15.100.2 [北京市 CNISP会员(尚未分配)]
139.196.135.135 [浙江省杭州市 阿里云BGP数据中心]
298
298
298

前面您提到：
“可尝试在代码中添加 log 调试，去掉 DNS 解析，直接使用3中的 IP 地址去连接看是否 OK”
正在尝试，不过需要深入阿里iotSDK实现过程，请问大佬对此过程是否熟悉，希望大佬给指点指点！谢谢！！！

ustccw, 2018-11-28:
@erfengwelink 
可以基于您的 SDK, 先写个简单的 `DNS` 测试 demo, 对比测试 `www.baidu.com` 和 `阿里云`解析是否都 OK.

ustccw, 2018-11-28:
或许和`域名长度`, 或者其他条件限制有关.