espressif/esp-aliyun #32: 最新代码，和阿里云平台的连接变得很慢，比之前的版本都慢很多

Yvan623048750, 2018-05-26:
最新代码，和阿里云平台的连接变得很慢，比之前的版本都慢很多，而且经常失败超时重连，之前的旧版本就不会。

Yvan623048750, 2018-05-26:
[inf] _network_ssl_read(468): ssl recv error: code = -30848, err_str = ''
[dbg] iotx_mc_read_packet(744): mqtt read error, rc=-2

[dbg] iotx_mc_cycle(1239): readPacket error,result = -1

[inf] IOT_MQTT_Yield(2415): error occur 

[err] iotx_mc_keepalive(1871): network is disconnected!

MQTT disconnect.

[inf] _network_ssl_disconnect(531): ssl_disconnect
[dbg] iotx_mc_cycle(1225): state = 4

[inf] IOT_MQTT_Yield(2415): error occur 

[inf] iotx_mc_handle_reconnect(2072): Waiting to reconnect...

[inf] iotx_mc_handle_reconnect(2078): start to reconnect

[dbg] guider_print_dev_guider_info(256): ....................................................

[dbg] guider_print_dev_guider_info(257):           ProductKey : a1AcxcBav6q

[dbg] guider_print_dev_guider_info(258):           DeviceName : yvan

[dbg] guider_print_dev_guider_info(259):             DeviceID : a1AcxcBav6q.yvan

[dbg] guider_print_dev_guider_info(261): ....................................................

[dbg] guider_print_dev_guider_info(262):        PartnerID Buf : ,partner_id=espressif

[dbg] guider_print_dev_guider_info(263):         ModuleID Buf : ,module_id=wroom-02

[dbg] guider_print_dev_guider_info(264):           Guider URL : 

[dbg] guider_print_dev_guider_info(266):       Guider SecMode : 2 (TLS + Direct)

[dbg] guider_print_dev_guider_info(268):     Guider Timestamp : 2524608000000

[dbg] guider_print_dev_guider_info(269): ....................................................

[dbg] guider_print_dev_guider_info(275): ....................................................

[dbg] guider_print_conn_info(233): -----------------------------------------

[dbg] guider_print_conn_info(234):             Host : a1AcxcBav6q.iot-as-mqtt.cn-shanghai.aliyuncs.com

[dbg] guider_print_conn_info(235):             Port : 1883

[dbg] guider_print_conn_info(238):         ClientID : a1AcxcBav6q.yvan|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom-02|

[dbg] guider_print_conn_info(240):       TLS PubKey : 0x40204f5c ('-----BEGIN CERTI ...')

[dbg] guider_print_conn_info(243): -----------------------------------------

[inf] iotx_mc_attempt_reconnect(2049): reconnect params: MQTTVersion=4, clientID=a1AcxcBav6q.yvan|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom-02|, keepAliveInterval=60, username=yvan&a1AcxcBav6q

[inf] _ssl_client_init(189): Loading the CA root certificate ...
[inf] _ssl_parse_crt(156): crt content:2
[inf] _ssl_client_init(197):  ok (0 skipped)
[inf] _TLSConnectNetwork(359): Connecting to /a1AcxcBav6q.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...
[inf] _TLSConnectNetwork(372):  ok
[inf] _TLSConnectNetwork(377):   . Setting up the SSL/TLS structure...
[inf] _TLSConnectNetwork(387):  ok
[inf] _TLSConnectNetwork(422): Performing the SSL/TLS handshake...
[inf] _TLSConnectNetwork(430):  ok
[inf] _TLSConnectNetwork(434):   . Verifying peer X.509 certificate..
[inf] _real_confirm(95): certificate verification result: 0x00
[inf] iotx_mc_connect(2035): mqtt connect success!
[... 54 lines cut ...]

[dbg] guider_print_conn_info(238):         ClientID : a1AcxcBav6q.yvan|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom-02|

[dbg] guider_print_conn_info(240):       TLS PubKey : 0x40204f5c ('-----BEGIN CERTI ...')

[dbg] guider_print_conn_info(243): -----------------------------------------

[inf] iotx_mc_attempt_reconnect(2049): reconnect params: MQTTVersion=4, clientID=a1AcxcBav6q.yvan|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom-02|, keepAliveInterval=60, username=yvan&a1AcxcBav6q

[inf] _ssl_client_init(189): Loading the CA root certificate ...
[inf] _ssl_parse_crt(156): crt content:2
[inf] _ssl_client_init(197):  ok (0 skipped)
[inf] _TLSConnectNetwork(359): Connecting to /a1AcxcBav6q.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...
[inf] _TLSConnectNetwork(372):  ok
[inf] _TLSConnectNetwork(377):   . Setting up the SSL/TLS structure...
[inf] _TLSConnectNetwork(387):  ok
[inf] _TLSConnectNetwork(422): Performing the SSL/TLS handshake...
[inf] _TLSConnectNetwork(430):  ok
[inf] _TLSConnectNetwork(434):   . Verifying peer X.509 certificate..
[inf] _real_confirm(95): certificate verification result: 0x00
[inf] iotx_mc_connect(2035): mqtt connect success!

[inf] iotx_mc_keepalive(1861): network is reconnected!

MQTT reconnect.

ustccw, 2018-05-26:
`ssl recv error: code = -30848`  意思是对端关闭 SSL, 您可否抓包看下，是否是阿里云发送了 RST 包而导致SSL 断开
```
#define MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY                 -0x7880  /**< The peer notified us that the connection is going to be closed. */
```

Yvan623048750, 2018-05-28:
发现一旦出现这个重连不上的情况，就会一直连不上，一直不断重连，但是一重启就立马连上正常了。

24540477zxc, 2018-09-07:
@Yvan623048750 请问这个问题您解决了吗。我也是出现这个情况，每隔三分钟就掉线重连。

closed 2019-01-09 by Eavo