espressif/esp-aliyun #35: 内存申请，程序跑飞

seblee, 2018-05-27:
就运行到下面这里就跑飞了，
```
EXAMPLE_TRACE("Initialize MQTT parameter");
    if (NULL == (msg_buf = (char *)HAL_Malloc(MQTT_MSGLEN))) {
        EXAMPLE_TRACE("not enough memory");
        rc = -1;
        goto do_exit;
    } 

   if (NULL == (msg_readbuf = (char *)HAL_Malloc(MQTT_MSGLEN))) {
        EXAMPLE_TRACE("not enough memory");
        rc = -1;
        goto do_exit;
    }
```
# 下面是打印的log
> 
> rf cal sector: 1019
> tcpip_task_hdl : 40107a00, prio:10,stack:512
> idle_task_hdl : 40107ab0,prio:0, stack:384
> tim_task_hdl : 40107bf8, prio:2,stack:512
> mode : sta(60:01:94:0f:fb:b1)
> add if0
> 
> ******************************************
> demo compile time:May 27 2018 12:48:05
> ******************************************
> scandone
> 
> MQTT task started...
> scandone
> state: 0 -> 2 (b0)
> state: 2 -> 3 (0)
> state: 3 -> 5 (10)
> add 0
> aid 4
> pm open phy_2,type:2 0 0
> cnt 
> 
> connected with wowo-WiFi, channel 6
> dhcp client start...
> ip:192.168.0.117,mask:255.255.255.0,gw:192.168.0.1
> WiFi connected
> Initializing SNTP
> please start sntp first !
> current time : Thu Jan 01 00:00:00 1970
> 
> did not get a valid time from sntp server
> current time : Sun May 27 12:54:27 2018
> 
> MQTT client example begin, free heap size:60640
> Initialize MQTT parameter
> Initialize MQTT parameter done
> [inf] iotx_device_info_init(40): device_info created successfully!
> **Initialize MQTT parameter**Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0<0xa2>@*r<0x85><0xc4>P<0xb1><0xb5><0xb1>K<0xed>t<0xa5>dr<0xd7>.@<0xe9><0xf9>.<0xa9><0xd1><0xaa><0xa1>Ko8a /t<0xb5>$v'<0xe1><0xea><0xc9>*<0xb1>T<0xb1><0xb1><0xb1><0xa9><0xa9><0xc4>P<0x9b><0xac><0xa9><0xb5><0xf9>*<0xb1><0x85><0xa2><0xfc>nzAl )<0xe9>m~<0xb1>t[!<0xf9>(P<0xb1>+t<0xc7>,<0xb1>@j|<0xbd><0xe5>*<0xb5><0xf9>*<0xb1><0x89><0xf8>nzAn )<0xe9>m~<0xb1>Z[!<0xf9>(P<0xb1>+t<0xc7><0xad>r@j|<0xbd><0xf5>*<0xb5><0xf9>*<0xb9><0x89><0xf8>nzAl )<0xe9>m~<0xb1>P[<0xe1>r<0xa7>(<0xb1><0xab><0xa2>8<0xa1><0xad><0x87><0xe9><0xf9>.<0xbd><0x04>r<0xf9>+<0xb5><0xb1>Xh<0xa9><0xb5><0xec><0xbd><0xe1><0xe1>b<0x05><0x95>v<0xa7><0xa8><0xa1><0xc5>"<0xe1><0xa2><0x88>dA<0xe2><0xd1>*<0xb5><0x91><0xe5>N<0xb8>`A#<0xc9>-Ae<0xe1>*<0xad><0x89><0x04>X<0xb5>[p<0xd1>h<0xb1><0x8b><0xa1>Z<0xb1><0x8b><0xa1>Z i~P<0xb7><0xa8>vA<0xee>v<0x9b>(P<0xb1><0xb1><0x81><0xa2><0xfc>OS SDK ver: 2.0.0(e271380) compiled @ Mar 30 2018 18:54:06
> phy ver: 1055_1, pp ver: 10.7
> 
> rf cal sector: 1019
> tcpip_task_hdl : 40107a00, prio:10,stack:512
> idle_task_hdl : 40107ab0,prio:0, stack:384
> tim_task_hdl : 40107bf8, prio:2,stack:512
> mode : sta(60:01:94:0f:fb:b1)

bzxsgz, 2018-05-29:
似乎是用官方的那个虚拟机编译就会这样

用提供的cygwin就可以

虚拟机比这个编译要快非常多，我也想知道为什么用虚拟机不行

seblee, 2018-05-30:
我用官方的Fatal exception 查原因，发现是硬件重启，

官方说是硬件错误或者野指针，现在还没找到解决的办法

ustccw, 2018-06-04:
@seblee 请问有修改源码吗？可否提供测试代码和步骤方便我们复现。

shifa2008, 2018-06-10:
不用修改直接配好这几个参数就可以复现：
模块esp8266-12f   
#define WIFI_SSID       ""       // type:string, your AP/router SSID to config your device networking
#define WIFI_PASSWORD   ""    // type:string, your AP/router password
#define PRODUCT_KEY             ""  // type:string
#define DEVICE_NAME             ""  // type:string
#define DEVICE_SECRET           ""  // type:string
 之前版本 不会， 这个版本  的C-SDK  正常  但是 这个8266的demo 会出现此问题。  不知道 是什么原因导致的。

ustccw, 2018-06-12:
@shifa2008 你好，esp8266-12f 是[安信可的模组](https://www.ai-thinker.com/product/esp8266)，我们无法对其直接维护，可否使用我们[官方模组](https://www.espressif.com/zh-hans/company/contact/buy-a-sample)来测试？

WuFengSheng, 2018-06-28:
******************************************
demo compile time:Jun 28 2018 21:58:25
******************************************
scandone

MQTT task started...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
pm open phy_2,type:2 0 0
cnt 

connected with Xiaomi_E303, channel 4
dhcp client start...
ip:192.168.31.153,mask:255.255.255.0,gw:192.168.31.1
WiFi connected
Initializing SNTP
please start sntp first !
current time : Thu Jan 01 00:00:00 1970

did not get a valid time from sntp server
current time : Fri Jun 29 00:18:07 2018

MQTT client example begin, free heap size:60640
[inf] iotx_device_info_init(40): device_info created successfully!

[dFatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0Fatal exception (0*rjrA(!婼藆嬓vX%+衴阤h)"8醫m莳NI*E,眰1?)DP沎薗{~A??E"HhInA,鬧Mmr,R?1┰vA?祔*eh n?EhI預,Z[Mor,R?5?濯A?祔*U?n?EhI預,P[M忽A,P[M)礠(Pы鷔(!B=d)奠絘€塲?e?x %謵[闑-嵿9徕?r7卾*%p婼Q?{`歋{`K
殈1A韡婒?卙?	X蚈S SDK ver: 2.0.0(e271380) compiled @ Mar 30 2018 18:54:06
phy ver: 1055_1, pp ver: 10.7

closed 2019-01-09 by Eavo