espressif/esp-aliyun #37: 第一次tcp连接失败

MinnowBin, 2018-06-04:
您好。我在使用上一版本基于阿里IOT套件2.0版本中。稳定复现连接MQTT过程中，第一次TCP连接出错。然后重新连接正常。相关打印如下：
MQTT client example begin, free heap size:28536
establish tcp connection with server(host=hd4st5dwsnh.iot-as-mqtt.cn-shanghai.aliyuncs.com port=1883)
fail to establish tcp
[log_err] iotx_mc_connect(2262): TCP or TLS Connection failed
MQTT construct failed
MQTT client example end, free heap size:27480

MQTT client example begin, free heap size:27480
establish tcp connection with server(host=hd4st5dwsnh.iot-as-mqtt.cn-shanghai.aliyuncs.com port=1883)
success to establish tcp, fd=1
目前的基于2.1版本的移植程序中，编译过程遇到 cannot find -liot_sdk 错误。
另外如果8266子模块更新至最新esp-idf风格的版本。头文件相关的makefile应该如何处理。

MinnowBin, 2018-06-05:
我个人理解是 在esp8266-aliyun-demo/components/aliyun/platform/os/espressif/HAL_TCP_esp8266.c文件中HAL_TCP_Establish函数把socket返回0作为结果返回外层函数，外层函数对此作为错误情况返回：iotkit-embedded/src/utils/misc/utils_net.c中connect_tcp。导致最外层重连。这里和您确认下socket返回0是否应该被认为是错误返回。我测试socket=0的连接是可用的。

ustccw, 2018-06-05:
@MinnowBin 当前仓库是给予2.10版本的，连接过程走的是 `esp8266-aliyun-demo/components/aliyun/platform/ssl/mbedtls/HAL_TLS_mbedtls.c` 文件中`_TLSConnectNetwork`函数，不是 HAL_TCP_Establish 函数。

MinnowBin, 2018-06-05:
感谢。我用的是非SSL直连的方式。应该还是要走HAL_TCP_Establish 。谢谢！