espressif/esp-aliyun #38: 最新代码编译失败！！

sofwu, 2018-06-05:
xtensa-lx106-elf-gcc  -L/home/esp8266-aliyun/esp8266-rtos-sdk/lib -L./components/aliyun/iotkit-embedded/output/release/lib -Wl,--gc-sections -nostdlib -T./ld/eagle.app.v6.new.2048.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lcirom -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lmain -lfreertos -llwip -lssl -liot_sdk -ldriver -lmirom -lmbedtls -lopenssl -ljson -lsmartconfig -lspiffs user/.output/eagle/debug/lib/libuser.a components/aliyun/platform/.output/eagle/debug/lib/libplatform.a -lcrypto -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -liot_sdk
collect2: error: ld returned 1 exit status
make: *** [/home/esp8266-aliyun/esp8266-rtos-sdk/Makefile:404：.output/eagle/debug/image/eagle.app.v6.out] 错误 1

ustccw, 2018-06-12:
@sofwu `cannot find -liot_sdk`，代表链接 libiot_sdk.a 失败，可否确认下 `libiot_sdk.a`是否生成。

ustccw, 2018-06-13:
@sofwu 我这边试了下，没有问题的。请问，你是用 README.md 中 4.1 节下载的 cygwin 吗

sofwu, 2018-06-13:
@ustccw 你能留个联系方式吗？我有几个问题要请教你

sofwu, 2018-06-13:
@ustccw 就是我用你的demo时候，模块配网后gpio的电压一直不稳定，接上led一直在闪是什么情况呢?

closed 2018-06-20 by sofwu