espressif/esp-aliyun #46: ESP-07s模块不断出现 Fatal exception

hanhuakai, 2018-06-26:
使用esp-07s模块，源代码除了wifi信息和iot参数未做其他任何改变，模块一直在重启。log如下：

ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : QIO
SPI Flash Size & Map: 16Mbit(1024KB+1024KB)
jump to run user1 @ 1000

OS SDK ver: 2.0.0(e271380) compiled @ Mar 30 2018 18:54:06
phy ver: 1055_1, pp ver: 10.7

rf cal sector: 507
tcpip_task_hdl : 40107a00, prio:10,stack:512
idle_task_hdl : 40107ab0,prio:0, stack:384
tim_task_hdl : 40107bf8, prio:2,stack:512
mode : sta(a0:20:a6:15:18:69)
add if0

******************************************
demo compile time:Jun 26 2018 22:42:01
******************************************
scandone

MQTT task started...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
pm open phy_2,type:2 0 0
cnt 

connected with hanhuakai-2.4G, channel 11
dhcp client start...
ip:192.168.0.115,mask:255.255.255.0,gw:192.168.0.1
WiFi connected
MQTT client example begin, free heap size:61184
[inf] iotx_device_info_init(40): device_info created successfully!
Fatal exception (20): 
epc1=0x00000000
epc2=0x00000000
epc3=0x40100774
epcvaddr=0x00000000
depc=0x00000000
rtn_add=0x00000000

Free Heap Size: 59104
Stack Point: 3fff6340
3fff6340: 401020f3 00000004 40101b0c 40101b4f
3fff6350: 00000000 401021d6 00000000 3fff6376
3fff6360: 0000000a 00000061 3fff637b 00000001
3fff6370: 5ffffe00 31360000 00343831 40102920
3fff6380: 00000000 00000000 00000000 40102360
3fff6390: 40102078 00000008 0000002a 00000000
3fff63a0: 4010223b a5a5a5a5 3fff16bd 401022f4
3fff63b0: 00000061 00000000 3fff1680 401003f8

hanhuakai, 2018-06-26:
使用 乐鑫官⽅开发板 ESP-LAUNCHER 也是相同的情况
从官方下载的代码，只有如下修改：
esp8266@esp8266-VirtualBox:~/work_aliyun/esp8266-aliyun_git/esp8266-aliyun$ git diff
diff --git a/include/aliyun_config.h b/include/aliyun_config.h
index 5d4290b..b094205 100644
--- a/include/aliyun_config.h
+++ b/include/aliyun_config.h
@@ -44,10 +44,10 @@ typedef enum {
 
 #define DEFAULT_TASK_MODE MQTT_TASK     // there are three modes for user as the enum TASK_MODE depicted, default: mqtt task
 
-#define WIFI_SSID       "********"       // type:string, your AP/router SSID to config your device networking
-#define WIFI_PASSWORD   "********"    // type:string, your AP/router password
+#define WIFI_SSID       "hanhuakai-2.4G"       // type:string, your AP/router SSID to config your device networking
+#define WIFI_PASSWORD   "19881218"    // type:string, your AP/router password
 
-#define START_SNTP 1    // 1: enable  SNTP and get realtime timestamp(for ssl handshake, check certificate validity)
+#define START_SNTP 0    // 1: enable  SNTP and get realtime timestamp(for ssl handshake, check certificate validity)
 // 0: disable SNTP
 
 #define RF_ADDR 0x1fc000 // more info see as gen_misc.sh step 5
@@ -57,9 +57,10 @@ typedef enum {
 // 0x3fc000 -> compile option 6: 4096KB(1024KB + 1024KB)
 
 // device name, device secret, product key which defined in aliyun cloud
-#define PRODUCT_KEY             "********"  // type:string
-#define DEVICE_NAME             "********"  // type:string
-#define DEVICE_SECRET           "********"  // type:string
+#define PRODUCT_KEY             "kfmiLOaXWrt"  // type:string
+#define DEVICE_NAME             "Auto-01"  // type:string
+#define DEVICE_SECRET           "s6oCFN6tom0bfr0REmxcLrVCWIKHFJ"  // type:string

thisdream, 2018-08-27:
您好，请问出现Fatal exception错误后，您是怎么解决的