espressif/esp-aliyun #51: 离线消息丢失，存在严重问题(附原因及解决办法)

JeremyHe-cn, 2018-07-04:
如题，使用sdk2.0时离线消息无法收到，严重影响业务的进行。
经过排查，发现以下原因：
## clean_session 未赋值
即使初始化时将 `mqtt_params.clean_session = 0`，但是实际构造的时候值依旧是默认为 1。
原因是IOT_MQTT_Construct方法中并未对其中的 cleansession 进行赋值，导致cleansession永远为 1.
这个将导致 qos1和qos2的订阅支持不完善，离线消息将无法收到。

> 我这边暂时在IOT_MQTT_Construct进行了修复，对 cleansession 进行了赋值

## 回调未调用
经过以上修复后，阿里云会对离线消息进行再次发送。从日志来看，模组也对离线消息回应了 puback。
但是用户程序却无法收到消息回调。  
经排查，原因是sdk 中订阅回调函数是在收 suback 后才进行设置的。而我调试发现，离线消息先于 suback 收到，于是导致此离线消息无法找到回调函数，用户程序在业务上则丢失了此消息，严重影响业务逻辑。
```
PUBLISH // 先收到 publish 消息

********** event payload [len:46] start addr:40108e3c **********
ff 01 c1530675535481 00 1b @ 1b E 01 1d ! 11 1b a 01 c 0d 0a 1b E 00 1d ! 00 1b a 00 0d 0a 1 1b d 03
---------- event payload End ----------
SUBACK // 后收到的 suback 消息
SUBACK
```
sdk 中具体的代码位于iotx_mc_cycle中，还望仔细排查。

>   目前解决办法有两，一是设置默认消息回调，在默认回调函数中再进行过滤处理。二是修改sdk 中的iotx_mc_handle_recv_PUBLISH实现，找不到回调时不回复 puback，等待阿里云再一次重发  

以上问题在 sdk 2.0 中出现，而 sdk2.1 已经换成了 iotkit，我不清楚是否同样存在问题，还望sdk 开发者仔细排查，并作好充分测试。毕竟一个sdk，你们发布出来就会有很多人使用，务必对自己写的代码甚至是公司的代码负责。谢谢。

hanhuakai, 2018-07-04:
@JeremyHe-cn  请问您用的哪个版本   git log 对应的 commit 号是多少？
目前 我 pull 最新的代码 直接会导致 WiFi 模组不断重启，--> ESP-07s模块不断出现 Fatal exception #46
原先用的 3月份的 sdk 代码  是没有上述不断重启问题
但是为了以后代码更新维护，还是想要用最新的代码进行开发，不知您如何认为？

JeremyHe-cn, 2018-07-06:
@chery-kai 我现在是不敢随便用新sdk，因为新的已经更换为 iotkit 了，跟之前的还有点区别，官方又没给出迁移的修改项，只能自己对比。之前迁移了，但是不敢保证没问题，所以还是继续使用旧 sdk。并且旧sdk中存在上述的问题，我已经自己解决了，所以不打算换新的。

这个 sdk 其实用的也是 paha 的 mqtt 库，我甚至考虑后期采用 mqtt 库自己封装。

附上 commit: e94ea98eca210ae9d343e84a7765bf928b9326ca

chery-kai, 2018-07-09:
@JeremyHe-cn 非常感谢您的回复，我看到您还提到了 ： https://github.com/espressif/esp8266-aliyun-demo/issues/49  //连接十分不稳定，并且会导致异常重启
因为我也打算使用 8266 模组用于一些应用中，但是在测试过程中遇到一些问题，不知道能够可以使用到实际应用中，您那边目前应用情况如何，方便告知一些情况吗？

JeremyHe-cn, 2018-07-10:
@chery-kai 不建议使用。正如这个项目名字，这就是一个 DEMO，官方人员根本没有好好测试，实际用起来一堆坑，提了 issue 他们也不关注。如果用到的只是mqtt或者ota，建议自己移植mqtt。我这边 stm32 + 2G 模组移植mqtt就没出现过程序跑飞、不断重连、qos=1消息丢失等问题。如果用到阿里云的东西比较多，那你可以考虑使用最新的sdk，上述问题是基于 sdk2.0出现的。我已经疲于给官方找bug了，还有很多需求要做呢。你体验下新版，如果不错，可以通知我，非常感谢

chery-kai, 2018-07-10:
@JeremyHe-cn 非常感谢您的建议，我会继续做一些测试和验证，如果方便，希望您能加一下我的QQ：189789985   在这一方面，我觉得以后有任何进展可以和您及时沟通和交流

closed 2019-01-09 by Eavo