espressif/esp-aliyun #52: make: *** No rule to make target 'clean'.  Stop.

semieie, 2018-07-19:
make[1]: Leaving directory '/home/semi/esp8266-aliyun-demo/components/aliyun/iotkit-embedded'
rm components/aliyun/iotkit-embedded/src/configs/config.espressif.esp8266
mv components/aliyun/iotkit-embedded/src/configs/default_settings.mk.bak components/aliyun/iotkit-embedded/src/configs/default_settings.mk
make: *** No rule to make target 'clean'.  Stop.
make: *** No rule to make target 'all', needed by 'dummy'.  Stop.

你好，
  我在用Ubuntu下 执行 gen_misc.sh 时出现以上显示的错误；
  另外，我在试图使用 git submodule update --init --recursive 这条命令将 esp8266-rtos-sdk 与阿里云的 iotkit-embedded clone到本地时，出现以下错误；
error: Server does not allow request for unadvertised object d8895578d30917696faa9800a2e468c7e41cb7be
Fetched in submodule path 'components/aliyun/iotkit-embedded', but it did not contain d8895578d30917696faa9800a2e468c7e41cb7be. Direct fetching of that commit failed.

  所以我只好对其repositorie进行单独 clone 。

盼回复，多谢！

weiecn, 2018-08-17:
我是这样做的 
在./gen_misc.sh
中添加了   （仅供参考）
export SDK_PATH=~/ESP8266-Aliyun-Iot/esp8266-rtos-sdk
export BIN_PATH=~/ESP8266-Aliyun-Iot/esp8266-rtos-sdk/bin
export PATH=~/ESP8266-Aliyun-Iot/xtensa-lx106-elf/bin:$PATH

rtos退回到了
git reset –hard 581aec0

搞定可以编译成功且能运行

CatAndCoffee, 2018-08-17:
是要将官方的RTOS_SDK退回到581aec0吗？
那个RTOS_SDK下载下来需要执行其他的什么操作吗？第一次上手ESP8266，看文档看了好多，也没太弄明白
@weiecn

closed 2019-01-09 by Eavo