espressif/esp-aliyun #54: ESP8266运行几个小时候出现SSL断连，并且一直处于阻塞状态

titawork, 2018-07-28:
日志信息：
[inf] _network_ssl_read(468): ssl recv error: code = -30848, err_str = ''
[dbg] iotx_mc_read_packet(744): mqtt read error, rc=-2
[dbg] iotx_mc_cycle(1239): readPacket error,result = -1
[inf] IOT_MQTT_Yield(2415): error occur 
[err] iotx_mc_keepalive(1871): network is disconnected!

请问哪里有 error: code = -30848 的文档描述？
这个问题怎么解决？看起来像是库的问题。

过程：
1. 启动运行，一切正常
2. 在MQTT出现几次重连后，出现上述打印
3. 此时，接收是好的，但是我的发送线程因为调用了 IOT_MQTT_Yield(pclient, 200);
而一直处于**阻塞状态**。

chery-kai, 2018-08-12:
你好   你的这个现象我也遇到    请问你解决了吗？
-30848  对应的   0x7880
//./esp8266-rtos-sdk/include/mbedtls/ssl.h
#define MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY                 -0x7880  /**< The peer notified us that the connection is going to be closed. */

yaozhenfa, 2018-11-18:
https://code.aliyun.com/edward.yangx/public-docs/wikis/user-guide/linkkit/TLS/SSL%C3%A8%C2%BF%C2%9E%C3%A6%C2%8E%C2%A5%C3%A9%C2%94%C2%99%C3%A8%C2%AF%C2%AF

titawork, 2018-11-24:
单独开一个线程调用 IOT_MQTT_Yield 来解决。 IOT_MQTT_Yield 是收消息的函数调用

closed 2018-11-24 by titawork