espressif/esp-aliyun #56: 测试一晚上 第二天来看挂掉了 下面是日志 严重怀疑这个sdk 没有测试过（超过24小时的测试）

shangzhe, 2018-08-29:
der_print_conn_info(255):       TLS PubKey : 0x40205004 ('-----BEGIN CERTI ...')
[dbg] guider_print_conn_info(258): -----------------------------------------
[inf] iotx_mc_attempt_reconnect(2049): reconnect params: MQTTVersion=4, clientID=iXTESaYy3TG.60019473763A|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom-02|, keepAliveInterval=60, username=60019473763A&iXTESaYy3TG
[inf] _ssl_client_init(189): Loading the CA root certificate ...

[09:47:04.644]收←◆[inf] _ssl_parse_crt(156): crt content:2
[inf] _ssl_client_init(197):  ok (0 skipped)
[inf] _TLSConnectNetwork(359): Connecting to /iXTESaYy3TG.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...

[09:47:04.735]收←◆[inf] _TLSConnectNetwork(372):  ok
[inf] _TLSConnectNetwork(377):   . Setting up the SSL/TLS structure...
[inf] _TLSConnectNetwork(387):  ok
[inf] _TLSConnectNetwork(422): Performing the SSL/TLS handshake...

[09:47:05.785]收←◆[inf] _TLSConnectNetwork(430):  ok
[inf] _TLSConnectNetwork(434):   . Verifying peer X.509 certificate..
[inf] _real_confirm(95): certificate verification result: 0x00

[09:47:05.845]收←◆[inf] iotx_mc_connect(2035): mqtt connect success!
[inf] iotx_mc_keepalive(1861): network is reconnected!
MQTT reconnect.

[09:48:36.308]收←◆[inf] _network_ssl_read(468): ssl recv error: code = -30848, err_str = ''
[dbg] iotx_mc_read_packet(744): mqtt read error, rc=-2
[dbg] iotx_mc_cycle(1239): readPacket error,result = -1
[inf] IOT_MQTT_Yield(2415): error occur 
[err] iotx_mc_keepalive(1871): network is disconnected!
MQTT disconnect.

[09:48:36.360]收←◆[inf] _network_ssl_disconnect(531): ssl_disconnect
[dbg] iotx_mc_cycle(1225): state = 4
[inf] IOT_MQTT_Yield(2415): error occur 
[inf] iotx_mc_handle_reconnect(2072): Waiting to reconnect...

[09:48:37.396]收←◆[inf] iotx_mc_handle_reconnect(2078): start to reconnect
[dbg] guider_print_dev_guider_info(271): ....................................................
[dbg] guider_print_dev_guider_info(272):           ProductKey : iXTESaYy3TG
[dbg] guider_print_dev_guider_info(273):           DeviceName : 60019473763A
[dbg] guider_print_dev_guider_info(274):             DeviceID : iXTESaYy3TG.60019473763A
[dbg] guider_print_dev_guider_info(276): ....................................................
[dbg] guider_
[09:48:37.465]收←◆print_dev_guider_info(277):        PartnerID Buf : ,partner_id=espressif
[dbg] guider_print_dev_guider_info(278):         ModuleID Buf : ,module_id=wroom-02
[dbg] guider_print_dev_guider_info(279):           Guider URL : 
[dbg] guider_print_dev_guider_info(281):       Guider SecMode : 2 (TLS + Direct)
[dbg] guider_print_dev_guider_info(283):     Guider Timestamp : 2524608000000
[dbg] guider_print_dev_guider_info(284): ....................................................
[dbg] guider_print_dev_gu
[09:48:37.533]收←◆ider_info(290): ....................................................
[dbg] guider_print_conn_info(248): -----------------------------------------
[dbg] guider_print_conn_info(249):             Host : iXTESaYy3TG.iot-as-mqtt.cn-shanghai.aliyuncs.com
[dbg] guider_print_conn_info(250):             Port : 1883
[dbg] guider_print_conn_info(253):         ClientID : iXTESaYy3TG.60019473763A|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom-02|
[dbg] 
[09:48:37.600]收←◆guider_print_conn_info(255):       TLS PubKey : 0x40205004 ('-----BEGIN CERTI ...')
[dbg] guider_print_conn_info(258): -----------------------------------------
[inf] iotx_mc_attempt_reconnect(2049): reconnect params: MQTTVersion=4, clientID=iXTESaYy3TG.60019473763A|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom-02|, keepAliveInterval=60, username=60019473763A&iXTESaYy3TG
[inf] _ssl_client_init(189): Loading the CA root certificate ...

[09:48:37.641]收←◆[inf] _ssl_parse_crt(156): crt content:2
[inf] _ssl_client_init(197):  ok (0 skipped)
[inf] _TLSConnectNetwork(359): Connecting to /iXTESaYy3TG.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...

[09:48:37.711]收←◆[inf] _TLSConnectNetwork(372):  ok
[inf] _TLSConnectNetwork(377):   . Setting up the SSL/TLS structure...
[inf] _TLSConnectNetwork(387):  ok
[inf] _TLSConnectNetwork(422): Performing the SSL/TLS handshake...

[09:48:38.892]收←◆[inf] _TLSConnectNetwork(430):  ok
[inf] _TLSConnectNetwork(434):   . Verifying peer X.509 certificate..
[... 88 lines cut ...]
[dbg] guider_print_conn_info(249):             Host : iXTESaYy3TG.iot-as-mqtt.cn-shanghai.aliyuncs.com
[dbg] guider_print_conn_info(250):             Port : 1883
[dbg] guider_print_conn_info(253):         ClientID : iXTESaYy3TG.60019473763A|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom-02|
[dbg] 
[09:52:18.679]收←◆guider_print_conn_info(255):       TLS PubKey : 0x40205004 ('-----BEGIN CERTI ...')
[dbg] guider_print_conn_info(258): -----------------------------------------
[inf] iotx_mc_attempt_reconnect(2049): reconnect params: MQTTVersion=4, clientID=iXTESaYy3TG.60019473763A|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom-02|, keepAliveInterval=60, username=60019473763A&iXTESaYy3TG
[inf] _ssl_client_init(189): Loading the CA root certificate ...

[09:52:18.722]收←◆[inf] _ssl_parse_crt(156): crt content:2
[inf] _ssl_client_init(197):  ok (0 skipped)
[inf] _TLSConnectNetwork(359): Connecting to /iXTESaYy3TG.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...

[09:52:19.027]收←◆[inf] _TLSConnectNetwork(372):  ok
[inf] _TLSConnectNetwork(377):   . Setting up the SSL/TLS structure...
[inf] _TLSConnectNetwork(387):  ok
[inf] _TLSConnectNetwork(422): Performing the SSL/TLS handshake...

[09:52:20.046]收←◆[inf] _TLSConnectNetwork(430):  ok
[inf] _TLSConnectNetwork(434):   . Verifying peer X.509 certificate..
[inf] _real_confirm(95): certificate verification result: 0x00

[09:52:20.110]收←◆[inf] iotx_mc_connect(2035): mqtt connect success!
[inf] iotx_mc_keepalive(1861): network is reconnected!
MQTT reconnect.

24540477zxc, 2018-08-29:
这个esp8266-aliyun-demo是不是删掉了。。。。

shangzhe, 2018-08-29:
好像是 昨天还在

27-301, 2018-08-29:
改名了，当前 esp-aliyun 支持 ESP-IDF & ESP8266_RTOS_SDK v3.0。如果需要使用基于 ESP8266_RTOS_SDK v2.x.x 的版本，checkout feature/for_esp8266_rtos_sdk_v2.x.x 。

closed 2019-01-09 by Eavo