espressif/esp-aliyun #65: compile for esp32 with error

rillhu, 2018-09-03:
When I use esp-idf-V3.0 to compile the alien demo, I got below error message:

/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -liot_sdk
collect2: error: ld returned 1 exit status

It looks the iot_sdk lib is not compiled, maybe some mk files are missed in the latest release version.

Could please espresso expert have a look?

Thanks!

ustccw, 2018-09-03:
@rillhu please run `make aliyun` firstly

rillhu, 2018-09-04:
@ustccw Thanks so much for your hint!
I tried the cmd: `make aliyun`, however I got the waring msg as below:
warning: 

> /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: /Users/rillhu/esp/esp-aliyun/components/aliyun/iotkit-embedded/.O/usr/lib/libiot_sdk.a the table of contents is empty (no object file members in the library define global symbols)

And when I run cmd :`make`, I got lots errors:
LD build/main.elf
```
/Users/rillhu/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_client+0x50): undefined reference to `IOT_SetupConnInfo'
/Users/rillhu/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_client+0x54): undefined reference to `IOT_MQTT_Construct'
/Users/rillhu/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_client+0x58): undefined reference to `IOT_MQTT_Publish'
/Users/rillhu/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_client+0x5c): undefined reference to `IOT_MQTT_Destroy'
```

It looks `make aliyun` does not work properly, could you please check the issue again?

Thank you so much in advance!

rillhu, 2018-10-15:
@gregrgr Please fine the .mk in below path:
esp-aliyun\components\aliyun\iotkit-embedded\build-rules

closed 2018-10-15 by rillhu

gregrgr, 2018-10-15:
> @gregrgr Please fine the .mk in below path:
> esp-aliyun\components\aliyun\iotkit-embedded\build-rules

thanks