espressif/esp-aliyun #69: NO matching any topic, call default handle function。整段代码都没有用到topic_data，为什么会有data主题推送的消息。

24540477zxc, 2018-09-05:
发起的订阅是get，但是串口输出显示还是订阅了data。整段代码都没有用到topic_data，为什么会有data主题推送的消息。
[inf] iotx_mc_connect(2035): mqtt connect success!
[dbg] iotx_mc_report_mid(2259): MID Report: started in MQTT
[dbg] iotx_mc_report_mid(2276): MID Report: json data = '{"id":"a17ZkMFEaL9_test0_mid","params":{"_sys_device_mid":"wroom-32","_sys_device_pid":"espressif"}}'
[dbg] iotx_mc_report_mid(2292): MID Report: topic name = '/sys/a17ZkMFEaL9/test0/thing/status/update'
[dbg] iotx_mc_report_mid(2309): MID Report: finished, IOT_MQTT_Publish() = 0
[inf] iotx_mc_subscribe(1388): mqtt subscribe success,topic = /a17ZkMFEaL9/test0/get!
[dbg] iotx_mc_cycle(1277): PUBLISH
[dbg] iotx_mc_handle_recv_PUBLISH(1091):         Packet Ident : 00055645
[dbg] iotx_mc_handle_recv_PUBLISH(1092):         Topic Length : 23
[dbg] iotx_mc_handle_recv_PUBLISH(1096):           Topic Name : /a17ZkMFEaL9/test0/data
[dbg] iotx_mc_handle_recv_PUBLISH(1099):     Payload Len/Room : 46 / 995
[dbg] iotx_mc_handle_recv_PUBLISH(1100):       Receive Buflen : 1024
[dbg] iotx_mc_handle_recv_PUBLISH(1111): delivering msg ...
[dbg] iotx_mc_deliver_message(887): NO matching any topic, call default handle function
event_handle|144 :: topic message arrived but without any related handle: topic=/a17ZkMFEaL9/test0/data, topic_msg={"attr_name":"temperature", "attr_value":"15"}
[dbg] iotx_mc_cycle(1277): PUBLISH
[dbg] iotx_mc_handle_recv_PUBLISH(1091):         Packet Ident : 00055646
[dbg] iotx_mc_handle_recv_PUBLISH(1092):         Topic Length : 23
[dbg] iotx_mc_handle_recv_PUBLISH(1096):           Topic Name : /a17ZkMFEaL9/test0/data
[dbg] iotx_mc_handle_recv_PUBLISH(1099):     Payload Len/Room : 46 / 995
[dbg] iotx_mc_handle_recv_PUBLISH(1100):       Receive Buflen : 1024
[dbg] iotx_mc_handle_recv_PUBLISH(1111): delivering msg ...
[dbg] iotx_mc_deliver_message(887): NO matching any topic, call default handle function
event_handle|144 :: topic message arrived but without any related handle: topic=/a17ZkMFEaL9/test0/data, topic_msg={"attr_name":"temperature", "attr_value":"19"}
mqtt_client|305 :: 
 publish message: 
 topic: /a17ZkMFEaL9/test0/update
 payload: hello!Lock Coming Soon!
 rc = 2
[dbg] iotx_mc_cycle(1277): PUBLISH
[dbg] iotx_mc_handle_recv_PUBLISH(1091):         Packet Ident : 00055647
[dbg] iotx_mc_handle_recv_PUBLISH(1092):         Topic Length : 23
[dbg] iotx_mc_handle_recv_PUBLISH(1096):           Topic Name : /a17ZkMFEaL9/test0/data
[dbg] iotx_mc_handle_recv_PUBLISH(1099):     Payload Len/Room : 46 / 995
[dbg] iotx_mc_handle_recv_PUBLISH(1100):       Receive Buflen : 1024
[dbg] iotx_mc_handle_recv_PUBLISH(1111): delivering msg ...
[dbg] iotx_mc_deliver_message(887): NO matching any topic, call default handle function
event_handle|144 :: topic message arrived but without any related handle: topic=/a17ZkMFEaL9/test0/data, topic_msg={"attr_name":"temperature", "attr_value":"17"}

24540477zxc, 2018-09-07:
设备会无故掉线
[inf] _network_ssl_read(456): ssl recv error: code = -30848, err_str = ''
[dbg] iotx_mc_read_packet(744): mqtt read error, rc=-2
[dbg] iotx_mc_cycle(1239): readPacket error,result = -1
[err] iotx_mc_keepalive(1871): network is disconnected!
event_handle|100 :: MQTT disconnect.
[inf] _network_ssl_disconnect(519): ssl_disconnect
[dbg] iotx_mc_cycle(1225): state = 4
[inf] IOT_MQTT_Yield(2415): error occur 
[inf] iotx_mc_handle_reconnect(2072): Waiting to reconnect...
[inf] iotx_mc_handle_reconnect(2078): start to reconnect

ustccw, 2018-09-17:
设备掉线问题在排查中

wanghui0574xs, 2018-09-26:
> 设备掉线问题在排查中

我也发生无故掉线问题了
[inf] iotx_mc_keepalive_sub(2226): send MQTT ping...
[inf] _network_ssl_read(456): ssl recv error: code = -30848, err_str = ''
[dbg] iotx_mc_read_packet(744): mqtt read error, rc=-2
[dbg] iotx_mc_cycle(1239): readPacket error,result = -1
[inf] IOT_MQTT_Yield(2415): error occur
[err] iotx_mc_keepalive(1871): network is disconnected!
mqtt_event_handle|093 :: MQTT disconnect.
[inf] _network_ssl_disconnect(519): ssl_disconnect
[dbg] iotx_mc_cycle(1225): state = 4
[inf] iotx_mc_handle_reconnect(2072): Waiting to reconnect...
[inf] iotx_mc_handle_reconnect(2078): start to reconnect

wanghui0574xs, 2018-09-26:
> 发起的订阅是get，但是串口输出显示还是订阅了data。整段代码都没有用到topic_data，为什么会有data主题推送的消息。
> [inf] iotx_mc_connect(2035): mqtt connect success!
> [dbg] iotx_mc_report_mid(2259): MID Report: started in MQTT
> [dbg] [...]

data主题 做了发布跟订阅. 用例里会发布消息, 同时也会收到自己发布的消息.

Shaco-Ma, 2018-11-27:
NO matching any topic, call default handle function。这个是阿里云c-sdk的问题，可以追溯一下sub的时候list_add和suback的时候list_get(具体名字记不住了)，所以不会对topic进行限制，你可以在上层的hp这个callback进行限制。但是阿里云最新的c-sdk已经修复了，我们都准备评估一下准备自己重新移植最新的阿里云c-sdk了。。。。
另外，如果你没有订阅会不会收到。。。。建议还是查一下代码，我记得example里面有data的默认订阅。

Eavo, 2018-12-25:
请问下，该问题是否已解决？

wanghui0574xs, 2018-12-27:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    还未解决
    
        
            
        
    
    
        
            
        
    
    
          
        
            
    
        font{
            line-height: 1.5;
        }
    
----------------------------wanghui0574xs@sina.com

closed 2019-01-25 by Eavo