espressif/esp-aliyun #70: ESP32对接阿里云

zhaozejun, 2018-09-07:
根据文档中的步骤使用ESP32对接阿里云
可以进行交叉编译，但是过程中出现：
error: implicit declaration of function 'mbedtls_cipher_auth_decrypt' [-Werror=implicit-function-declaration]
     if( ( ret = mbedtls_cipher_auth_decrypt( &key->ctx, iv, 12,
                 ^
cc1: some warnings being treated as errors
显示mbedtls的库函数没有完全添加到工程？
使用这个工程8266是能够跑通的，跑ESP32是不是没验证过？

ustccw, 2018-09-17:
@zhaozejun 跑 ESP32 之前，请删除 build 目录，sdkconfig 和 sdkconfig.old. 切换 ESP-IDF 分支到 `release/v3.0`, 再试试看

zhaozejun, 2018-09-25:
切换到release/v3.0跑通了，谢谢！
另外在使用过程中发现一个函数命名的问题
文件：esp-aliyun-master\components\aliyun\iotkit-embedded\src\utils\misc\utils_list.c中的
list_t *list_new(void)函数与
void list_remove(list_t *self, list_node_t *node)函数
和ESP32蓝牙驱动文件（路径：esp-idf-master\components\bt\bluedroid\osi\list.c）中的
list_t *list_new(list_free_cb callback)函数
bool list_remove(list_t *list, void *data)函数
函数命名是相同的 导致ESP32蓝牙移植到阿里云网关程序的时候发生函数命名相同的冲突。

closed 2019-01-09 by Eavo