espressif/esp-aliyun #72: mac 下编译链接libiot_sdk.a 时出错, 建议支持mac下开发.

wanghui0574xs, 2018-09-13:
ESP8266_RTOS_SDK支持mac下编译, esp_aliyun基本上也能在mac下编译, 只是在链接libiot_sdk.a 时出错.  mac里用虚拟机很是蛋疼, 编辑器基本上都装mac上, 虚拟机里没有开发用的编辑器. 两边代码同步也是麻烦. 强烈建议, 支持mac下开发.

27-301, 2018-09-14:
为什么不能呢？

https://github.com/espressif/ESP8266_RTOS_SDK#get-toolchain

wanghui0574xs, 2018-09-14:
[AR] libiot_cota.a              <= config_ota.o                                    
[AR] libiot_sdk.a               <=      
                                   src/cmp/iotx_cmp_api.o
                                   src/cmp/iotx_cmp_cloud_conn.o
                                   src/cmp/iotx_cmp_common.o
                                   src/cmp/iotx_cmp_log.o
                                   src/cmp/iotx_cmp_mqtt_direct.o
                                   src/cmp/iotx_cmp_ota.o
                                   src/cota/config_ota.o
                                   src/dm/cJSON.o
                                   src/dm/cmp_abstract_impl.o
                                   src/dm/cmp_message_info.o
                                   src/dm/dm_impl.o
                                   src/dm/dm_thing.o
                                   src/dm/dm_thing_manager.o
                                   src/dm/logger.o
                                   src/dm/single_list.o
                                   src/fota/service_ota.o
                                   src/http/iotx_http_api.o
                                   src/log/lite-log.o
                                   src/mqtt/MQTTConnectClient.o
                                   src/mqtt/MQTTDeserializePublish.o
                                   src/mqtt/MQTTPacket.o
                                   src/mqtt/MQTTSerializePublish.o
                                   src/mqtt/MQTTSubscribeClient.o
                                   src/mqtt/MQTTUnsubscribeClient.o
                                   src/mqtt/mqtt_client.o
                                   src/mqtt/mqtt_instance.o
                                   src/ota/ota.o
                                   src/shadow/shadow.o
                                   src/shadow/shadow_common.o
                                   src/shadow/shadow_delta.o
                                   src/shadow/shadow_update.o
                                   src/system/ca.o
                                   src/system/class_interface.o
                                   src/system/device.o
                                   src/system/guider.o
                                   src/system/id2_guider.o
                                   src/system/report.o
                                   src/system/sdk-impl.o
                                   src/utils/json_parser.o
                                   src/utils/json_token.o
                                   src/utils/lite-utils_prog.o
                                   src/utils/lite-utils_testsuites.o
                                   src/utils/mem_stats.o
                                   src/utils/string_utils.o
                                   src/utils/utils_base64.o
                                   src/utils/utils_epoch_time.o
                                   src/utils/utils_hmac.o
                                   src/utils/utils_httpc.o
                                   src/utils/utils_list.o
                                   src/utils/utils_md5.o
                                   src/utils/utils_net.o
                                   src/utils/utils_sha1.o
                                   src/utils/utils_sha256.o
                                   src/utils/utils_timer.o
_warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: /Users/wanghui/Workspace/GitProject/ESP8266/esp-aliyun/components/aliyun/iotkit-embedded/.O/usr/lib/libiot_sdk.a the table of contents is empty (no object file members in the library define global symbols)_

27-301, 2018-09-14:
ok，理解错误。

临时解决一下，回头合进去。

_rules-complib.mk 中：
```
define Finalize_CompLib
( \
	EXIST_OBJS="$$(ls $(1) 2>/dev/null)"; \
\
	if [ "$${EXIST_OBJS}" != "" ]; then \
	    $(AR) -rcs $(2)/lib$(3).a $${EXIST_OBJS}; \
	fi \
)
endef
改为下面即可：
define Finalize_CompLib
( \
	EXIST_OBJS="$$(ls $(1) 2>/dev/null)"; \
\
	if [ "$${EXIST_OBJS}" != "" ]; then \
	    xtensa-lx106-elf-ar -rcs $(2)/lib$(3).a $${EXIST_OBJS}; \
	fi \
)
endef

wanghui0574xs, 2018-09-15:
如上所改, 跟linux上差别是, mac里链接完没有打印信息,