espressif/esp-aliyun #77: 重复定义了cJSON的部分函数

wanghui0574xs, 2018-11-21:
当使用下列json函数时,会发生链接错误.
cJSON *root1 = cJSON_CreateObject();
cJSON_AddStringToObject(root1, "key", "aaaa");
cJSON_AddStringToObject(root1, "head", "bbb");

错误如下:
~/Workspace/GitProject/ESP8266/esp-aliyun/examples/mqtt/build/cjson/libcjson.a(cJSON.o): In function `cJSON_GetErrorPtr':
~/Workspace/GitProject/ESP8266/ESP8266_RTOS_SDK/components/cjson/cJSON/cJSON.c:73: multiple definition of `cJSON_GetErrorPtr'
~/Workspace/GitProject/ESP8266/esp-aliyun/components/aliyun/iotkit-embedded/output/release/lib/libiot_sdk.a(cJSON.o):cJSON.c:(.text.cJSON_GetErrorPtr+0x0): first defined here
~/Workspace/GitProject/ESP8266/esp-aliyun/examples/mqtt/build/cjson/libcjson.a(cJSON.o): In function `cJSON_Version':
cJSON.c:(.text.cJSON_Version+0x0): multiple definition of `cJSON_Version'

wanghui0574xs, 2018-11-22:
去掉aliyun模块里的 iotkit-embedded/src/dm/src/cJson.c

Eavo, 2018-12-25:
请问下，该问题是否已解决？如果解决了，麻烦 close issue，有任何问题，可随时 reopen。谢谢。