espressif/esp-aliyun #78: ESP8266 mqtt连接错误

donny681, 2018-11-21:
您好，我使用ESP8266_RTOS_SDK V3.0编译该工程，发现无法连接阿里云后台。我确定product_key和product_secret肯定是正确的，但是一直返回以下错误。

`[dbg] guider_print_conn_info(243): -----------------------------------------
[inf] iotx_mc_init(1703): MQTT init success!
[inf] _ssl_client_init(175): Loading the CA root certificate ...
[inf] _ssl_parse_crt(142): crt content:2
[inf] _ssl_client_init(183):  ok (0 skipped)
[inf] _TLSConnectNetwork(345): Connecting to /b17UAUgolD6.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...
[inf] _TLSConnectNetwork(358):  ok
[inf] _TLSConnectNetwork(363):   . Setting up the SSL/TLS structure...
[inf] _TLSConnectNetwork(373):  ok
[inf] _TLSConnectNetwork(410): Performing the SSL/TLS handshake...
[inf] _TLSConnectNetwork(414): failed  ! mbedtls_ssl_handshake returned -0x7280
[inf] _network_ssl_disconnect(519): ssl_disconnect
[inf] HAL_SSL_Destroy(535): handle is NULL
[err] iotx_mc_connect(2001): TCP or TLS Connection failed
[inf] HAL_SSL_Destroy(535): handle is NULL
[inf] iotx_mc_disconnect(2129): mqtt disconnect!
[inf] iotx_mc_release(2175): mqtt release!
mqtt_client|221 :: MQTT construct failed
[err] LITE_dump_malloc_free_stats(594): WITH_MEM_STATS = 0
I (36735) MQTT: MQTT client example end, free heap size:52976
I (36744) MQTT: MQTT client example begin, free heap size:529`

请问出现以上错误原因是什么？

Eavo, 2018-12-25:
请问问题是否解决了？

donny681, 2018-12-29:
已经解决了。是填入配置参数错误。

closed 2018-12-29 by Eavo

wollir, 2019-07-09:
> 已经解决了。是填入配置参数错误。

同样的问题，请问是哪里的配置参数错了呢