espressif/esp-aliyun #79: ESP32-IDF + ESP-ALIYUN @ UBUNTU 18.04 x64 无法编译

nickfox-taterli, 2018-11-25:
主要出错在LD阶段，已经切换ESP-IDF到V3.0分支（切换不切换都是一个错误），也重新配置所有环境，尝试修改链接规则，无果，依然出现错误。

`AR build/xtensa-debug-module/libxtensa-debug-module.a
LD build/main.elf
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_client+0x50): undefined reference to `IOT_SetupConnInfo'
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_client+0x54): undefined reference to `IOT_MQTT_Construct'
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_client+0x58): undefined reference to `IOT_MQTT_Publish'
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_client+0x5c): undefined reference to `IOT_MQTT_Destroy'
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_client+0x60): undefined reference to `IOT_MQTT_Subscribe'
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_client+0x64): undefined reference to `IOT_MQTT_Yield'
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_client+0x68): undefined reference to `IOT_MQTT_Unsubscribe'
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_task+0x18): undefined reference to `IOT_OpenLog'
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_task+0x1c): undefined reference to `IOT_SetLogLevel'
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_task+0x20): undefined reference to `IOT_DumpMemoryStats'
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o):(.literal.mqtt_task+0x24): undefined reference to `IOT_CloseLog'
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o): In function `mqtt_client':
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:186: undefined reference to `IOT_SetupConnInfo'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:204: undefined reference to `IOT_MQTT_Construct'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:228: undefined reference to `IOT_MQTT_Publish'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:229: undefined reference to `IOT_MQTT_Destroy'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:240: undefined reference to `IOT_MQTT_Subscribe'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:245: undefined reference to `IOT_MQTT_Destroy'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:251: undefined reference to `IOT_MQTT_Publish'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:259: undefined reference to `IOT_MQTT_Yield'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:267: undefined reference to `IOT_MQTT_Publish'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:276: undefined reference to `IOT_MQTT_Yield'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:284: undefined reference to `IOT_MQTT_Yield'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:285: undefined reference to `IOT_MQTT_Unsubscribe'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:286: undefined reference to `IOT_MQTT_Yield'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:286: undefined reference to `IOT_MQTT_Destroy'
/home/tater/esp/esp-aliyun/examples/mqtt/build/main/libmain.a(mqtt_example.o): In function `mqtt_task':
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:301: undefined reference to `IOT_OpenLog'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:301: undefined reference to `IOT_SetLogLevel'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:318: undefined reference to `IOT_DumpMemoryStats'
/home/tater/esp/esp-aliyun/examples/mqtt/main/./mqtt_example.c:322: undefined reference to `IOT_CloseLog'
collect2: error: ld returned 1 exit status
make: *** [/home/tater/esp/esp-aliyun/examples/mqtt/build/main.elf] Error 1
`
ESP-ALIYUN版本是c9e6ae9
ESP-IDF版本是7e0ea6b05

ustccw, 2018-11-25:
@nickfox-taterli 先 `make aliyun` 再  `make`

nickfox-taterli, 2018-11-27:
any one here?

nickfox-taterli, 2018-11-29:
wtf?

xwdreamer, 2018-12-14:
https://github.com/espressif/esp-aliyun/issues/65
看这个连接，能解决问题

closed 2019-01-09 by Eavo