espressif/esp-aliyun #81: 跑官方demo 连接阿里云时，DNS失败

erfengwelink, 2018-11-28:
[dbg] iotx_device_info_init(32): device_info already created, return!
[dbg] iotx_device_info_set(50): start to set device info!
[dbg] iotx_device_info_set(64): device_info set successfully!
[dbg] guider_print_dev_guider_info(256): ....................................................
[dbg] guider_print_dev_guider_info(257):           ProductKey : a1Ugih4wMVP
[dbg] guider_print_dev_guider_info(258):           DeviceName : device_IAFh
[dbg] guider_print_dev_guider_info(259):             DeviceID : a1Ugih4wMVP.device_IAFh
[dbg] guider_print_dev_guider_info(261): ....................................................
[dbg] guider_print_dev_guider_info(262):        PartnerID Buf : ,partner_id=espressif
[dbg] guider_print_dev_guider_info(263):         ModuleID Buf : ,module_id=wroom-32
[dbg] guider_print_dev_guider_info(264):           Guider URL :
[dbg] guider_print_dev_guider_info(266):       Guider SecMode : 2 (TLS + Direct)
[dbg] guider_print_dev_guider_info(268):     Guider Timestamp : 2524608000000
[dbg] guider_print_dev_guider_info(269): ....................................................
[dbg] guider_print_dev_guider_info(275): ....................................................
[dbg] guider_print_conn_info(233): -----------------------------------------
[dbg] guider_print_conn_info(234):             Host : a1Ugih4wMVP.iot-as-mqtt.cn-shanghai.aliyuncs.com
[dbg] guider_print_conn_info(235):             Port : 1883
[dbg] guider_print_conn_info(238):         ClientID : a1Ugih4wMVP.device_IAFh|securemode=2,timestamp=2524608000000,signm
ethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom-32|
[dbg] guider_print_conn_info(240):       TLS PubKey : 0x3f40278c ('-----BEGIN CERTI ...')
[dbg] guider_print_conn_info(243): -----------------------------------------
[inf] iotx_mc_init(1703): MQTT init success!
[inf] _ssl_client_init(175): Loading the CA root certificate ...
[inf] _ssl_parse_crt(142): crt content:2
[inf] _ssl_client_init(183):  ok (0 skipped)
[inf] _TLSConnectNetwork(345): Connecting to /a1Ugih4wMVP.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...
[inf] _TLSConnectNetwork(354):  failed ! net_connect returned -0x0052
[inf] _network_ssl_disconnect(519): ssl_disconnect
[inf] HAL_SSL_Destroy(535): handle is NULL
[err] iotx_mc_connect(2001): TCP or TLS Connection failed
[inf] HAL_SSL_Destroy(535): handle is NULL
[inf] iotx_mc_disconnect(2129): mqtt disconnect!
[inf] iotx_mc_release(2175): mqtt release!
mqtt_client|222 :: MQTT construct failed
[err] LITE_dump_malloc_free_stats(594): WITH_MEM_STATS = 0
I (15887) MQTT: MQTT client example end, free heap size:230168
I (15887) MQTT: MQTT client example begin, free heap size:230168

erfengwelink, 2018-11-28:
同一WiFi下pc端ping 却能ping通：

# ping a1Ugih4wMVP.iot-as-mqtt.cn-shanghai.aliyuncs.com

正在 Ping a1ugih4wmvp.iot-as-mqtt.cn-shanghai.aliyuncs.com [106.15.100.2] 具有 32 字节的数据:
来自 106.15.100.2 的回复: 字节=32 时间=25ms TTL=38
来自 106.15.100.2 的回复: 字节=32 时间=25ms TTL=38
来自 106.15.100.2 的回复: 字节=32 时间=25ms TTL=38
来自 106.15.100.2 的回复: 字节=32 时间=25ms TTL=38

106.15.100.2 的 Ping 统计信息:
    数据包: 已发送 = 4，已接收 = 4，丢失 = 0 (0% 丢失)，
往返行程的估计时间(以毫秒为单位):
    最短 = 25ms，最长 = 25ms，平均 = 25ms

erfengwelink, 2018-11-28:
完整日志如下：
# make -j8 flash monitor
CC build/main/mqtt_example.o
AR build/main/libmain.a
LD build/main.elf
esptool.py v2.6-beta1
Flashing binaries to serial port COM10 (app at offset 0x10000 )...
esptool.py v2.6-beta1
Serial port COM10
Connecting......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 84:0d:8e:17:be:64
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 21088 bytes to 12450...
Wrote 21088 bytes (12450 compressed) at 0x00001000 in 1.1 seconds (effective 153.1 kbit/s)...
Hash of data verified.
Compressed 773840 bytes to 471789...
Wrote 773840 bytes (471789 compressed) at 0x00010000 in 41.8 seconds (effective 148.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 1365.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on COM10 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5892
load:0x40078000,len:9032
load:0x40080000,len:6064
0x40080000: _iram_start at C:/Users/Administrator/Desktop/net-conn-test/bare-test/IDF/esp-idf-v3.1.1/components/freertos
/xtensa_vectors.S:1685

entry 0x40080330
0x40080330: _KernelExceptionVector at ??:?

I (28) boot: ESP-IDF v3.1.1-rc2-2-gd1d2ce8c2-dirty 2nd stage bootloader
I (29) boot: compile time 20:02:09
I (30) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1d858 (120920) map
I (136) esp_image: segment 1: paddr=0x0002d880 vaddr=0x3ffb0000 size=0x02790 ( 10128) load
I (140) esp_image: segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x8b468 (570472) map
0x400d0018: _stext at ??:?

I (342) esp_image: segment 3: paddr=0x000bb488 vaddr=0x3ffb2790 size=0x00e88 (  3720) load
I (344) esp_image: segment 4: paddr=0x000bc318 vaddr=0x3ffb3618 size=0x00000 (     0) load
I (349) esp_image: segment 5: paddr=0x000bc320 vaddr=0x40080000 size=0x00400 (  1024) load
[... 61 lines cut ...]
[dbg] guider_print_dev_guider_info(269): ....................................................
[dbg] guider_print_dev_guider_info(275): ....................................................
[dbg] guider_print_conn_info(233): -----------------------------------------
[dbg] guider_print_conn_info(234):             Host : a1Ugih4wMVP.iot-as-mqtt.cn-shanghai.aliyuncs.com
[dbg] guider_print_conn_info(235):             Port : 1883
[dbg] guider_print_conn_info(238):         ClientID : a1Ugih4wMVP.device_IAFh|securemode=2,timestamp=2524608000000,signm
ethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom-32|
[dbg] guider_print_conn_info(240):       TLS PubKey : 0x3f40278c ('-----BEGIN CERTI ...')
[dbg] guider_print_conn_info(243): -----------------------------------------
[inf] iotx_mc_init(1703): MQTT init success!
[inf] _ssl_client_init(175): Loading the CA root certificate ...
[inf] _ssl_parse_crt(142): crt content:2
[inf] _ssl_client_init(183):  ok (0 skipped)
[inf] _TLSConnectNetwork(345): Connecting to /a1Ugih4wMVP.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...
[inf] _TLSConnectNetwork(354):  failed ! net_connect returned -0x0052
[inf] _network_ssl_disconnect(519): ssl_disconnect
[inf] HAL_SSL_Destroy(535): handle is NULL
[err] iotx_mc_connect(2001): TCP or TLS Connection failed
[inf] HAL_SSL_Destroy(535): handle is NULL
[inf] iotx_mc_disconnect(2129): mqtt disconnect!
[inf] iotx_mc_release(2175): mqtt release!
mqtt_client|222 :: MQTT construct failed
[err] LITE_dump_malloc_free_stats(594): WITH_MEM_STATS = 0
I (4467) MQTT: MQTT client example end, free heap size:230168
I (4467) MQTT: MQTT client example begin, free heap size:230168

erfengwelink, 2018-11-29:
深入跟踪 发现是 lwip lib函数出错：

int
lwip_getaddrinfo(const char *nodename, const char *servname,
       const struct addrinfo *hints, struct addrinfo **res)
{
.....
/* LWIP_IPV4 && LWIP_IPV6 */
      err = netconn_gethostbyname_addrtype(nodename, &addr, type);
      if (err != ERR_OK) {
        printf("%s EAI_FAIL err %d \n", __func__, err);
        return EAI_FAIL;
      }
.....

erfengwelink, 2018-11-29:
失败的打印：

@@@@@@@@@@@@@@IOT_MQTT_Construct@@@@@@@@@@@@start@@@@@@@
iotx_mc_set_connect_params/1508: Input heartbeat interval(10000 ms) < Allowed minimum(60000 ms)
iotx_mc_set_connect_params/1511: Reset heartbeat interval => 60000 Millisecond
iotx_net_init,
iotx_mc_init/1637: MQTT init success --------- !
*******************  IOT_MQTT_Construct iotx_mc_connect *********************
iotx_mc_connect Establish TCP or TLS connectionHAL_SSL_Establish  port_str 1883
[inf] _ssl_client_init(168): Loading the CA root certificate ...
cert. version     : 3
serial number     : 04:00:00:00:00:01:15:4B:5A:C3:94
issuer name       : C=BE, O=GlobalSign nv-sa, OU=Root CA, CN=GlobalSign Root CA
subject name      : C=BE, O=GlobalSign nv-sa, OU=Root CA, CN=GlobalSign Root CA
issued  on        : 1998-09-01 12:00:00
expires on        : 2028-01-28 12:00:00
signed using      : RSA with SHA1
RSA key size      : 2048 bits
basic constraints : CA=true
key usage         : Key Cert Sign, CRL Sign
[inf] _ssl_parse_crt(136): crt content:451
[inf] _ssl_client_init(176):  ok (0 skipped)
[inf] TLSConnectNetwork(346): Connecting to /a1Ugih4wMVP.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...
mbedtls_net_connect start ...
lwip_getaddrinfo netconn_gethostbyname_addrtype err -6 addr -2146621232
lwip_getaddrinfo EAI_FAIL err -6
[inf] TLSConnectNetwork(349):  failed ! net_connect returned -0x0052
 pClient->ipstack->connect rc = -1
[inf] HAL_SSL_Destroy(446): handle is NULL
iotx_mc_connect/1936: TCP or TLS Connection failed
68 c a2 1 0 0 0 0 0 0 0 0 af 43
iotx_mc_release/2111: mqtt release!
@@@@@@@@@@@@@@IOT_MQTT_Construct@@@@@@@@@@@@end@@@@@@@

ustccw, 2018-12-04:
@erfengwelink 可以参考一下 https://blog.csdn.net/ustccw/article/details/84786289

ustccw, 2018-12-04:
1. 确认 errno = -6, 是否是超时或内存不足引起此问题
```
  if (ipaddr == NULL) {
    /* timeout or memory error */
    *msg->err = ERR_VAL;
  } else {
    /* address was resolved */
    *msg->err = ERR_OK;
    *msg->addr = *ipaddr;
  }
```

2. 如果是超时问题, 那有可能是 DNS server 没有回复模块
        您提到同一个路由器下的 PC 可以访问该域名: 是否有可能以前该 PC 访问过该域名, PC 将此条记录存储在 PC 的路由表中, 后续不再需要 DNS 解析该域名.

3. 可否依据博客, 设备和 PC 同步测试:
        - 同时解析一个未曾访问过的域名

4. 抓包查看 DNS 交互情况
        这通常比较直接能看出问题出在哪里
   
5. 是否有极端情况
-   比如运营商对 MAC 地址有限制, 而导致不转发其 DNS 包

ustccw, 2018-12-04:
@erfengwelink

Eavo, 2018-12-25:
请问这个问题是否已解决？因为长期未更新，问题先关闭，如果还有问题，可随时 reopen issue，谢谢。

closed 2018-12-25 by Eavo