espressif/esp-aliyun #82: _network_ssl_write(495): ssl write fail, code=-78, str=UNKNOWN ERROR CODE (004E)

wanghui0574xs, 2018-11-29:
使用的是QS1
出现多个IOT_MQTT_Publish, 
但没有如下2行的提示成功, 
[dbg] iotx_mc_cycle(1260): PUBACK
mqtt_event_handle|136 :: publish success, packet-id=378

经过很久后(2117086 - 582928) 大约26分钟后, 出现如下信息:
[inf] _network_ssl_write(495): ssl write fail, code=-78, str=UNKNOWN ERROR CODE (004E)
[err] iotx_mc_publish(1470): MQTTPublish is error, rc = -14

Eavo, 2018-12-25:
请问下这个问题是否还存在？

wanghui0574xs, 2018-12-27:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        
            
        
    
    存在, 在esp8266的平台里, 出现的概率蛮高, 一天出现10来次
    
        
            
        
    
    
          
        
            
    
        font{
            line-height: 1.5;
        }
    
----------------------------wanghui0574xs@sina.com

closed 2019-01-09 by Eavo