espressif/esp-aliyun #90: 编译esp-aliyun/examples/mqtt_example提示fatal error: iot_import.h

simonliu009, 2019-01-30:
~/Espressif/ESP8266/freeRTOS/myProjects/esp-aliyun/platform/hal/os/espressif/HAL_OS_espressif.c:34:24: fatal error: iot_import.h: No such file or directory
compilation terminated.
make[1]: *** [platform/hal/os/espressif/HAL_OS_espressif.o] Error 1
make: *** [component-esp-aliyun-build] Error 2

ustccw, 2019-01-31:
请按照 README.md 先编译 libiot_sdk.a

simonliu009, 2019-01-31:
谢谢。
当时没注意这个目录是外部链接，从这里git clone 看到目录是空的所以就跳过去了。
但是还有其他编译问题：
``...esp-aliyun/iotkit-embedded/output/release/lib/libiot_sdk.a: error adding symbols: Archive has no index; run ranlib to add one``
RTOS SDK已经切换到``5df0b1d71``
iotkit-embedded分别尝试了``V2.3.0``分支和``master``分支
查看了
[#87](https://github.com/espressif/esp-aliyun/issues/87)
[#88](https://github.com/espressif/esp-aliyun/issues/88)
似乎并没有什么帮助。

closed 2019-01-31 by simonliu009

wanghui0574xs, 2019-03-28:
> 请按照 README.md 先编译 libiot_sdk.a

编译libiot_sdk.a有没有详细说明, 以前只要make aliyun 现在都找不到怎么编译了