espressif/esp-aliyun #91: mac下编译aliyun v2.30仍然提示Archive has no index; run ranlib to add one

simonliu009, 2019-02-11:
```
/Users/simonliu/Espressif/ESP8266/freeRTOS/myProjects/esp-aliyun/iotkit-embedded/output/release/lib/libiot_sdk.a: error adding symbols: Archive has no index; run ranlib to add one
collect2: error: ld returned 1 exit status
make: *** [/Users/simonliu/Espressif/ESP8266/freeRTOS/myProjects/esp-aliyun/examples/mqtt_example/build/mqtt_example.elf] Error 1
```
``~/Espressif/ESP8266/freeRTOS/myProjects/esp-aliyun/iotkit-embedded/output/release/lib/libiot_sdk.a``已经生成了
freeRTOS已经切换到`5df0b1d`分支
iotkit-embedded分别尝试了`V2.3.0`分支和`master`分支都是相同的错误。

ustccw, 2019-02-13:
@simonliu009 暂时还不支持在 mac 上开发, 后续会支持此.

Eavo, 2019-02-21:
如果问题还未解决，可以随时 reopen。谢谢支持。

closed 2019-02-21 by Eavo