espressif/esp-aliyun #93: ESP8266平台编译问题

MinjunHuang, 2019-02-20:
使用的是ESP8266 V3.1 SDK，项目需要使用mbedTLS的库，因此通过make menuconfig-->SSL-->Choose SSL/TLS library，选择mbedTLS，在components.mk ,做如下修改：
ifdef CONFIG_TARGET_PLATFORM_ESP8266
# ESP8266 platform
# COMPONENT_SRCDIRS += platform/hal/ssl/openssl
# else
# ESP32 platform
COMPONENT_SRCDIRS += platform/hal/ssl/mbedtls
endif
mqtt_example编译不通过，提示错误：
error: implicit declaration of function 'mbedtls_platform_set_calloc_free' [-Werror=implicit-function-declaration]
     mbedtls_platform_set_calloc_free(_SSLCalloc_wrapper, _SSLFree_wrapper);

是不是ESP8266平台只能支持wolfSSL，不能使用mbedTLS？

Eavo, 2019-02-21:
您好，由于 MBEDTLS_PLATFORM_MEMORY 宏在 mbedtls 中默认是禁用的，所以无法调用 mbedtls_platform_set_calloc_free 。如果需要使用该函数，需要在 esp_config.h 文件中将该宏开启即可。

MinjunHuang, 2019-02-21:
Hi @Eavo 
好的，这个函数启用对内存的使用有不好的影响吗？

Eavo, 2019-02-22:
@MinjunHuang 没有什么影响的，只是替换了系统默认使用的内存分配函数 malloc 和 free，方便用户对内存的管理。

MinjunHuang, 2019-02-22:
@Eavo 
好的，多谢您的支持！

27-301, 2019-08-16:
请使用基于 iotkit-embedded v3.0.1 重新适配的 esp-aliyun release v2.0 测试。
相关文档已更新，请参考。建议重新 clone 一下 esp-aliyun。
如在新版本上遇到问题，请重新提交 issue。
谢谢。

closed 2019-08-16 by 27-301