espressif/esp-aliyun #94: 请问ESP8266 smart_light DEMO什么时候完善？

taksing-leung, 2019-03-06:
目前上电后会不停扫描，容易到时内存占用过多，重启
配网完成后，手机端app经常卡死在配网成功界面，点击开始使用 没反应
希望尽快完善，这么马上量产了

Eavo, 2019-03-06:
这个配网问题是当前的 known issue，我们正在修改，完成后会尽快更新的。谢谢。

Eavo, 2019-03-07:
内存优化是排在计划中的 item，会安排的。

taksing-leung, 2019-03-07:
不错，你们效率很高！要怎样才能加入你们的测试团队呢？我可以帮你们做各种测试wechat：taksing-leung

Eavo, 2019-03-08:
@taksing-leung 您好，我从企业微信上搜不到您的账户名，请问是否能提供手机号码？

taksing-leung, 2019-03-08:
13414571724

taksing-leung, 2019-03-08:
这是不是内存不足导致重启?请看LOG
`
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0x79
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x90
ho 0 tail 12 room 4
load 0x3ffe8420, len 3544, room 12 
tail 12
chksum 0xb5
csum 0xb5
[0;32mI (50) boot: ESP-IDF v3.1-14-g49a48fd 2nd stage bootloader[0m
[0;32mI (50) boot: compile time 11:31:06[0m
[0;32mI (51) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (58) boot: SPI Speed      : 40MHz[0m
[0;32mI (65) boot: SPI Mode       : QIO[0m
[0;32mI (71) boot: SPI Flash Size : 4MB[0m
[0;32mI (77) boot: Partition Table:[0m
[0;32mI (82) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (94) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (105) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (117) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (128) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (140) boot:  4 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (152) boot: End of partition table[0m
[0;32mI (158) boot: No factory image, trying OTA 0[0m
[0;32mI (166) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xac2d8 (705240) map[0m
[0;32mI (416) esp_image: segment 1: paddr=0x000bc2f0 vaddr=0x3ffe8000 size=0x007f0 (  2032) load[0m
[0;32mI (418) esp_image: segment 2: paddr=0x000bcae8 vaddr=0x3ffe87f0 size=0x0011c (   284) load[0m
[0;32mI (428) esp_image: segment 3: paddr=0x000bcc0c vaddr=0x40100000 size=0x0694c ( 26956) load[0m
[0;32mI (450) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (491) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (501) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (691) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
[0;32mI (701) app main: IDF version: v3.1-14-g49a48fd[0m
[0;32mI (701) app main: esp-aliyun verison: v1.0-99-g1d5ba38-dirty[0m
[0;32mI (711) app main: iotkit-embedded version: RELEASED_V2.30-dirty[0m
[0;32mI (721) factory: Don't Factory[0m
[0;32mI (721) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (741) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (751) pwm: --- PWM v3.0
[0m
lightbulb_set_aim r 0 g 0 b 0

mode : sta(cc:50:e3:d6:41:3c)
add if0
[0;33mW (781) awss_config: No data storage,the load data is empty[0m
[0;32mI (781) app main: IOTX_AWSS_START[0m
[crt] __awss_start

[0;32mI (801) awss_config: wifi running at monitor mode[0m
[crt] chan 1

[crt] [1] ssid:Tenda_6C06B0, mac:0840f36c06b0, chn:1, auth:shared, wep, wep, rssi:-80, adha:0

[1;35m[wrn] map_cipher_to_encry(474): unknow cipher type: 0

[0m[crt] [2] ssid:鐩涗赴, mac:74c330f35440, chn:1, auth:wpa-psk, invalid encry, aes, rssi:-68, adha:0

[crt] [3] ssid:TP-LINK_46AC74, mac:7844fd2188ec, chn:1, auth:wpa2-psk, aes, aes, rssi:-50, adha:0

[crt] [4] ssid:HBTX-VIVO-OPPO, mac:246968f51a88, chn:1, auth:wpa2-psk, aes, aes, rssi:-80, adha:0

[1;35m[wrn] map_cipher_to_encry(474): unknow cipher type: 0

[... 6222 lines cut ...]
[0m[1;33m[inf] guider_print_dev_guider_info(329):         ModuleID Buf : ,module_id=wroom

[0m[1;33m[inf] guider_print_dev_guider_info(330):           Guider URL : 

[0m[1;33m[inf] guider_print_dev_guider_info(332):       Guider SecMode : 2 (TLS + Direct)

[0m[1;33m[inf] guider_print_dev_guider_info(334):     Guider Timestamp : 2524608000000

[0m[1;33m[inf] guider_print_dev_guider_info(338): ....................................................

[0m[1;33m[inf] guider_print_conn_info(297): -----------------------------------------

[0m[1;33m[inf] guider_print_conn_info(298):             Host : a17F5W6Icug.iot-as-mqtt.cn-shanghai.aliyuncs.com

[0m[1;33m[inf] guider_print_conn_info(299):             Port : 1883

[0m[1;33m[inf] guider_print_conn_info(304):         ClientID : a17F5W6Icug.shebei01|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom|

[0m[1;33m[inf] guider_print_conn_info(306):       TLS PubKey : 0x40231003 ('-----BEGIN CERTI ...')

[0m[1;33m[inf] guider_print_conn_info(309): -----------------------------------------

[0m[1;33m[inf] iotx_mc_init(2334): MQTT init success!

[0m`

taksing-leung, 2019-03-08:
@Eavo 这是不是内存不足导致重启?请看LOG

> 
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0x79
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x90
ho 0 tail 12 room 4
load 0x3ffe8420, len 3544, room 12 
tail 12
chksum 0xb5
csum 0xb5
[0;32mI (50) boot: ESP-IDF v3.1-14-g49a48fd 2nd stage bootloader[0m
[0;32mI (50) boot: compile time 11:31:06[0m
[0;32mI (51) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (58) boot: SPI Speed      : 40MHz[0m
[0;32mI (65) boot: SPI Mode       : QIO[0m
[0;32mI (71) boot: SPI Flash Size : 4MB[0m
[0;32mI (77) boot: Partition Table:[0m
[0;32mI (82) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (94) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (105) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (117) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (128) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (140) boot:  4 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (152) boot: End of partition table[0m
[0;32mI (158) boot: No factory image, trying OTA 0[0m
[0;32mI (166) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xac2d8 (705240) map[0m
[0;32mI (416) esp_image: segment 1: paddr=0x000bc2f0 vaddr=0x3ffe8000 size=0x007f0 (  2032) load[0m
[0;32mI (418) esp_image: segment 2: paddr=0x000bcae8 vaddr=0x3ffe87f0 size=0x0011c (   284) load[0m
[0;32mI (428) esp_image: segment 3: paddr=0x000bcc0c vaddr=0x40100000 size=0x0694c ( 26956) load[0m
[0;32mI (450) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (491) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (501) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (691) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
[0;32mI (701) app main: IDF version: v3.1-14-g49a48fd[0m
[0;32mI (701) app main: esp-aliyun verison: v1.0-99-g1d5ba38-dirty[0m
[0;32mI (711) app main: iotkit-embedded version: RELEASED_V2.30-dirty[0m
[0;32mI (721) factory: Don't Factory[0m
[0;32mI (721) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (741) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (751) pwm: --- PWM v3.0
[0m
lightbulb_set_aim r 0 g 0 b 0

mode : sta(cc:50:e3:d6:41:3c)
add if0
[0;33mW (781) awss_config: No data storage,the load data is empty[0m
[0;32mI (781) app main: IOTX_AWSS_START[0m
[crt] __awss_start

[0;32mI (801) awss_config: wifi running at monitor mode[0m
[crt] chan 1

[crt] [1] ssid:Tenda_6C06B0, mac:0840f36c06b0, chn:1, auth:shared, wep, wep, rssi:-80, adha:0

[1;35m[wrn] map_cipher_to_encry(474): unknow cipher type: 0

[0m[crt] [2] ssid:鐩涗赴, mac:74c330f35440, chn:1, auth:wpa-psk, invalid encry, aes, rssi:-68, adha:0

[crt] [3] ssid:TP-LINK_46AC74, mac:7844fd2188ec, chn:1, auth:wpa2-psk, aes, aes, rssi:-50, adha:0

[crt] [4] ssid:HBTX-VIVO-OPPO, mac:246968f51a88, chn:1, auth:wpa2-psk, aes, aes, rssi:-80, adha:0

[1;35m[wrn] map_cipher_to_encry(474): unknow cipher type: 0
[... 6223 lines cut ...]
[0m[1;33m[inf] guider_print_dev_guider_info(329):         ModuleID Buf : ,module_id=wroom

[0m[1;33m[inf] guider_print_dev_guider_info(330):           Guider URL : 

[0m[1;33m[inf] guider_print_dev_guider_info(332):       Guider SecMode : 2 (TLS + Direct)

[0m[1;33m[inf] guider_print_dev_guider_info(334):     Guider Timestamp : 2524608000000

[0m[1;33m[inf] guider_print_dev_guider_info(338): ....................................................

[0m[1;33m[inf] guider_print_conn_info(297): -----------------------------------------

[0m[1;33m[inf] guider_print_conn_info(298):             Host : a17F5W6Icug.iot-as-mqtt.cn-shanghai.aliyuncs.com

[0m[1;33m[inf] guider_print_conn_info(299):             Port : 1883

[0m[1;33m[inf] guider_print_conn_info(304):         ClientID : a17F5W6Icug.shebei01|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom|

[0m[1;33m[inf] guider_print_conn_info(306):       TLS PubKey : 0x40231003 ('-----BEGIN CERTI ...')

[0m[1;33m[inf] guider_print_conn_info(309): -----------------------------------------

[0m[1;33m[inf] iotx_mc_init(2334): MQTT init success!

[0m

Eavo, 2019-03-08:
@taksing-leung 这个是 8266 的，还是 32 的？

taksing-leung, 2019-03-08:
@Eavo 8266

Eavo, 2019-03-08:
好的，我已安排复现，请稍等。

taksing-leung, 2019-03-08:
@Eavo 你微信多少?加一下

ytzhitong, 2019-06-11:
我也遇到同样的问题，求解。

27-301, 2019-08-16:
请使用基于 iotkit-embedded v3.0.1 重新适配的 esp-aliyun release v2.0 测试。
相关文档已更新，请参考。建议重新 clone 一下 esp-aliyun。
如在新版本上遇到问题，请重新提交 issue。
谢谢。

closed 2019-08-16 by 27-301