espressif/esp-aliyun #96: commit 51b3741  设备配网重启后,进入死循环,无法启动.看LOG

taksing-leung, 2019-03-08:
运行到(706) factory: Handle Factory 69[0m,后不停重启

 ets Jan  8 2013,rst cause:2, boot mode:(3,4)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0x79
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x90
ho 0 tail 12 room 4
load 0x3ffe8420, len 3544, room 12 
tail 12
chksum 0xb5
csum 0xb5
[0;32mI (50) boot: ESP-IDF v3.1-14-g49a48fd 2nd stage bootloader[0m
[0;32mI (50) boot: compile time 11:31:06[0m
[0;32mI (51) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (58) boot: SPI Speed      : 40MHz[0m
[0;32mI (65) boot: SPI Mode       : QIO[0m
[0;32mI (71) boot: SPI Flash Size : 4MB[0m
[0;32mI (77) boot: Partition Table:[0m
[0;32mI (82) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (94) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (105) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (117) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (128) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (140) boot:  4 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (152) boot: End of partition table[0m
[0;32mI (158) boot: No factory image, trying OTA 0[0m
[0;32mI (166) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xac2d8 (705240) map[0m
[0;32mI (416) esp_image: segment 1: paddr=0x000bc2f0 vaddr=0x3ffe8000 size=0x007f0 (  2032) load[0m
[0;32mI (418) esp_image: segment 2: paddr=0x000bcae8 vaddr=0x3ffe87f0 size=0x0011c (   284) load[0m
[0;32mI (428) esp_image: segment 3: paddr=0x000bcc0c vaddr=0x40100000 size=0x0694c ( 26956) load[0m
[0;32mI (450) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (486) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (486) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (676) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
[0;32mI (686) app main: IDF version: v3.1-14-g49a48fd[0m
[0;32mI (686) app main: esp-aliyun verison: v1.0-99-g1d5ba38-dirty[0m
[0;32mI (696) app main: iotkit-embedded version: RELEASED_V2.30-dirty[0m
[0;32mI (706) factory: Handle Factory 69[0m

 ets Jan  8 2013,rst cause:2, boot mode:(3,4)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0x79
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x90
ho 0 tail 12 room 4
load 0x3ffe8420, len 3544, room 12 
tail 12
chksum 0xb5
csum 0xb5
[0;32mI (972) boot: ESP-IDF v3.1-14-g49a48fd 2nd stage bootloader[0m
[0;32mI (972) boot: compile time 11:31:06[0m
[0;32mI (972) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (980) boot: SPI Speed      : 40MHz[0m
[0;32mI (986) boot: SPI Mode       : QIO[0m
[0;32mI (993) boot: SPI Flash Size : 4MB[0m
[0;32mI (999) boot: Partition Table:[0m
[0;32mI (1005) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (1016) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (1028) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (1040) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (1051) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[... 192 lines cut ...]
chksum 0x90
ho 0 tail 12 room 4
load 0x3ffe8420, len 3544, room 12 
tail 12
chksum 0xb5
csum 0xb5
[0;32mI (5719) boot: ESP-IDF v3.1-14-g49a48fd 2nd stage bootloader[0m
[0;32mI (5719) boot: compile time 11:31:06[0m
[0;32mI (5720) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (5728) boot: SPI Speed      : 40MHz[0m
[0;32mI (5735) boot: SPI Mode       : QIO[0m
[0;32mI (5741) boot: SPI Flash Size : 4MB[0m
[0;32mI (5747) boot: Partition Table:[0m
[0;32mI (5753) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (5765) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (5777) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (5788) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (5800) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (5812) boot:  4 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (5824) boot: End of partition table[0m
[0;32mI (5830) boot: No factory image, trying OTA 0[0m
[0;32mI (5838) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xac2d8 (705240) map[0m
[0;32mI (6089) esp_image: segment 1: paddr=0x000bc2f0 vaddr=0x3ffe8000 size=0x007f0 (  2032) load[0m
[0;32mI (6090) esp_image: segment 2: paddr=0x000bcae8 vaddr=0x3ffe87f0 size=0x0011c (   284) load[0m
[0;32mI (6100) esp_image: segment 3: paddr=0x000bcc0c vaddr=0x40100000 size=0x0

wjj-chn, 2019-03-08:
请问使用的是ESP32模组吗？

taksing-leung, 2019-03-08:
@wjj-chn ESP8266

wjj-chn, 2019-03-08:
根据log信息，使用的是ESP32 的IDF，建议确认一下 SDK 路径
export IDF_PATH=~/.../ESP8266_RTOS_SDK
然后重新编译iotkit库和工程。

taksing-leung, 2019-03-08:
@wjj-chn 确认了，如果用esp32是不能烧录的

taksing-leung, 2019-03-09:
printenv IDFPATH
@wjj-chn 
```
# printenv IDF_PATH
D:/msys32/home/Administrator/ESP8266_RTOS_SDK
```

wjj-chn, 2019-03-11:
请问上述现象可以复现吗？
如果可以复现，这边我也测试一下。
为了给您更好的支持，请帮忙完善以下信息，谢谢！ 

ESP 芯片 
ESP 模组名称 
ESP SDK 版本号  
问题复现步骤 
Log

taksing-leung, 2019-03-11:
https://github.com/espressif/esp-aliyun/blob/master/examples/solutions/smart_light/README.md
根据上面的链接制作
芯片：ESP 8266
模组：ESP 12-f
ESP-IDF: release/v3.2
ESP8266_RTOS_SDK: release/v3.1

wjj-chn, 2019-03-12:
你好，我使用了同样的芯片和SDK版本，无法复现你的问题。
不知道你的问题是在使用32模组，切换到8266上，出现的问题吗？
推荐你重新编译 libiot_sdk.a, 请先执行: rm output -rf && make clean && make distclean 
然后 smart light编译中，请先确认导入 8266 RTOS SDK路径  ，并删除sdkconfig。
 cd examples/solutions/smart_light
 export IDF_PATH=~/路径/ESP8266_RTOS_SDK
 rm sdkconfig
然后重新编译，擦除芯片后，进行烧录调试。
以上步骤后，可以正常使用。

我们的esp-aliyun 近日已经更新，可以同步到本地了。

wjj-chn, 2019-03-12:
你好，我使用了同样的芯片和SDK版本，无法复现你的问题。
不知道你的问题是在使用32模组，切换到8266上，出现的问题吗？
推荐你重新编译 libiot_sdk.a, 请先执行: rm output -rf && make clean && make distclean
然后 smart light编译中，请先确认导入 8266 RTOS SDK路径 ，并删除sdkconfig。
然后重新编译，擦除芯片后，进行烧录调试。
以上步骤后，可以正常使用。
命令参考
cd iotkit-embedded
rm output -rf && make clean && make distclean
make reconfig
make

cd examples/solutions/smart_light
export IDF_PATH=~/路径/ESP8266_RTOS_SDK
rm sdkconfig
make 
make erase_flash
make flash monitor

我们的esp-aliyun 近日已经更新，可以同步到本地了。

taksing-leung, 2019-03-12:
@wjj-chn 好的,谢谢解答.我去试试,有问题再反映

taksing-leung, 2019-03-14:
@wjj-chn 
rm sdkconfig
运行这个不是不能烧录了?烧录端口在里边
按你说的来做,现在app卡在配网成功那里,点击 开始使用.提示绑定设备失败

27-301, 2019-08-16:
请使用基于 iotkit-embedded v3.0.1 重新适配的 esp-aliyun release v2.0 测试。
相关文档已更新，请参考。建议重新 clone 一下 esp-aliyun。
如在新版本上遇到问题，请重新提交 issue。
谢谢。

closed 2019-08-16 by 27-301