espressif/esp-aliyun #97: esp8266编译问题

XXXXzzzz000, 2019-03-12:
```c
/** @brief Received packet radio metadata header, this is the common header at the beginning of all promiscuous mode RX callback buffers */
typedef struct {
    signed rssi: 8;           /**< signal intensity of packet */
    unsigned rate: 4;         /**< data rate */
    unsigned is_group: 1;
    unsigned : 1;             /**< reserve */
    unsigned sig_mode: 2;     /**< 0:is not 11n packet; 1:is 11n packet */
    unsigned legacy_length: 12;
    unsigned damatch0: 1;
    unsigned damatch1: 1;
    unsigned bssidmatch0: 1;
    unsigned bssidmatch1: 1;
    unsigned mcs: 7;          /**< if is 11n packet, shows the modulation(range from 0 to 76) */
    unsigned cwb: 1;          /**< if is 11n packet, shows if is HT40 packet or not */
    unsigned HT_length: 16;            /**< reserve */
    unsigned smoothing: 1;    /**< reserve */
    unsigned not_sounding: 1; /**< reserve */
    unsigned : 1;             /**< reserve */
    unsigned aggregation: 1;  /**< Aggregation */
    unsigned stbc: 2;         /**< STBC */
    unsigned fec_coding: 1;   /**< Flag is set for 11n packets which are LDPC */
    unsigned sgi: 1;          /**< SGI */
    unsigned rxend_state: 8;
    unsigned ampdu_cnt: 8;    /**< ampdu cnt */
    unsigned channel: 4;      /**< which channel this packet in */
    unsigned : 4;             /**< reserve */
    signed noise_floor: 8;
} wifi_pkt_rx_ctrl_t;

/** @brief Payload passed to 'buf' parameter of promiscuous mode RX callback.
 */
typedef struct {
    wifi_pkt_rx_ctrl_t rx_ctrl; /**< metadata header */
    uint8_t payload[0];       /**< Data or management payload. Length of payload is described by rx_ctrl.sig_len. Type of content determined by packet type argument of callback. */
} wifi_promiscuous_pkt_t;
```
wifi_pkt_rx_ctrl_t rx_ctrl缺少sig_len

27-301, 2019-03-15:
这个注释有问题，8266 没有 sig_len，这是 32 的。需要修改一下注释。

XXXXzzzz000, 2019-03-20:
除了这个问题还有报错.这是编译例程mqtt_example时发生的
前置工作:
配置工具链,idfpath
编译iotkit-embedded 为8266固件
```
esp-aliyun/platform/hal/os/espressif/HAL_AWSS_espressif.c:23:24: fatal error: iot_import.h: No such file or directory
```

XXXXzzzz000, 2019-03-20:
此处编译会发生编译错误
错误信息
```
esp-aliyun/platform/hal/os/espressif/HAL_AWSS_espressif.c: In function 'wifi_sniffer_cb':
esp-aliyun/platform/hal/os/espressif/HAL_AWSS_espressif.c:279:27: error: 'wifi_pkt_rx_ctrl_t {aka struct <anonymous>}' has no member named 'sig_len'
         len = pkt->rx_ctrl.sig_len;
                           ^
CC build/esp-aliyun/platform/hal/os/espressif/HAL_Crypt_espressif.o
CC build/espos/platform/freertos/espos_semaphore.o
CC build/esp_http_client/lib/http_header.o
make[1]: *** [Esp-idf/ESP8266_RTOS_SDK/make/component_wrapper.mk:286：platform/hal/os/espressif/HAL_AWSS_espressif.o] 错误 1
CC build/freertos/freertos/ringbuf.o
CC build/esp_http_client/lib/http_auth.o
esp-aliyun/platform/hal/os/espressif/HAL_Crypt_espressif.c:10:25: fatal error: mbedtls/aes.h: No such file or directory
compilation terminated.
AR build/esp-tls/libesp-tls.a
```

netbolo, 2019-07-28:
检查ESP8266 SDK是否v3.1版本，如果不是v3.1会出现以上错误。

closed 2019-07-29 by XXXXzzzz000