espressif/esp-aliyun #98: 连接几分钟后会掉线附LOG

taksing-leung, 2019-03-14:
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0x79
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x90
ho 0 tail 12 room 4
load 0x3ffe8420, len 3544, room 12 
tail 12
chksum 0xb0
csum 0xb0
[0;32mI (50) boot: ESP-IDF v3.1-14-g49a48fd 2nd stage bootloader[0m
[0;32mI (50) boot: compile time 09:49:05[0m
[0;32mI (51) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (58) boot: SPI Speed      : 40MHz[0m
[0;32mI (65) boot: SPI Mode       : QIO[0m
[0;32mI (71) boot: SPI Flash Size : 4MB[0m
[0;32mI (77) boot: Partition Table:[0m
[0;32mI (82) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (94) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (105) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (117) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (128) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (140) boot:  4 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (152) boot: End of partition table[0m
[0;32mI (158) boot: No factory image, trying OTA 0[0m
[0;32mI (166) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xac398 (705432) map[0m
[0;32mI (416) esp_image: segment 1: paddr=0x000bc3b0 vaddr=0x3ffe8000 size=0x007f0 (  2032) load[0m
[0;32mI (418) esp_image: segment 2: paddr=0x000bcba8 vaddr=0x3ffe87f0 size=0x0011c (   284) load[0m
[0;32mI (428) esp_image: segment 3: paddr=0x000bcccc vaddr=0x40100000 size=0x068e8 ( 26856) load[0m
[0;32mI (450) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (492) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (502) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (712) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
[0;32mI (712) app main: IDF version: v3.1-14-g49a48fd[0m
[0;32mI (712) app main: esp-aliyun verison: v1.0-107-g4cd3a62-dirty[0m
[0;32mI (722) app main: iotkit-embedded version: RELEASED_V2.30-dirty[0m
[0;32mI (742) factory: Don't Factory[0m
[0;32mI (742) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (752) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (772) pwm: --- PWM v3.0
[0m
lightbulb_set_aim r 0 g 0 b 0

mode : sta(cc:50:e3:d6:41:3c)
add if0
[0;33mW (792) awss_config: No data storage,the load data is empty[0m
[0;32mI (792) app main: IOTX_AWSS_START[0m
[crt] __awss_start

[0;32mI (822) awss_config: wifi running at monitor mode[0m
[crt] chan 1

[1;35m[wrn] map_cipher_to_encry(474): unknow cipher type: 0

[0m[crt] [1] ssid:ChinaNet-SD2S, mac:a0086f023514, chn:2, auth:wpa-psk, invalid encry, tkip, rssi:-58, adha:0

[1;35m[wrn] map_cipher_to_encry(474): unknow cipher type: 0

[0m[crt] [2] ssid:鐩涗赴, mac:74c330f35440, chn:1, auth:wpa-psk, invalid encry, aes, rssi:-81, adha:0

[crt] [3] ssid:TP-LINK_46AC74, mac:7844fd2188ec, chn:1, auth:wpa2-psk, aes, aes, rssi:-49, adha:0

[1;35m[wrn] map_cipher_to_encry(474): unknow cipher type: 0

[0m[1;35m[wrn] map_cipher_to_encry(474): unknow cipher type: 0

[... 2411 lines cut ...]
3ffe82e0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000001 00000000 00000000 00000000 00000000 00000000 00000000 00000000 ffffffff 

3ffe8320  00000004 3ffede40 3fff1010 40268b54 00000000 00000000 00000000 4026865e 00000000 00000000 00000000 00000000 00000000 00000000 00000000 4022657c 

3ffe8360  40226594 3ffede40 00000001 4026878c 00000000 00000000 00000000 40252858 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000001 

3ffe83a0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 ffffffff 00000004 3ffede40 3fff1010 40268b54 00000000 00000000 00000000 4026865e 

3ffe83e0  00000000 00000000 3fff0f3b 40292fa7 00000000 00000000 00000000 4022657c 40226594 00000080 00000001 4026878c 00002073 3fff0f34 00000000 40252858 

3ffe8420  3ffe8460 3ffe8450 00000008 40100455 00000004 00000000 3ffef254 00000001 00000000 00000000 00000004 40252572 40224438 3fff0f34 40105328 00000001 

3ffe8460  00000080 00000080 3ffef254 40268b54 3ffe97da 00000001 3ffe97d8 40252022 00000400 00000400 3ffe84e8 40250dfc 00000008 00000000 00000001 40105570 

3ffe84a0  40251ec7 00000001 00000001 40251eb0 40251fd2 4023f19c 3ffef26c 40251fe5 3fff0248 00000002 40222010 3fff09c4 fffffffd 3fff0aa0 401052c8 40105311 

      PC: 0x40268783       PS: 0x00000031       A0: 0x3ff7fff0       A1: 0x3ff7fff0 

      A2: 0x3ffede40       A3: 0x40226594       A4: 0x4022657c       A5: 0x00000001 

      A6: 0x00000000       A7: 0x00000000       A8: 0x00000003       A9: 0x3fff0b90 

     A10: 0x3fff0984      A11: 0x4022f774      A12: 0x00000001      A13: 0x3ffede40 

     A14: 0x40226594      A15: 0x4022657c      SAR: 0x00000019 EXCCAUSE: 0x00000003

taksing-leung, 2019-03-14:
```

乭ta+??".'蝦讋◎~P+vK-Ph
?/t5譊N?*.北	┠??	爹*薄乶(A瑝8x簐(?
?(嚺h?5kAn 晟*璽?盄j|单▇鵫増r駂
麻v(?塇~惫蛜An 晟*R?*?┠P?.	爹*?乶zA瑡辱蒼A,p壢m~)夰P慌 ?ˉ?$`á≧Hj(?瓏-r?胙晟jp鵓?(%但?(Xm@轾?z~(?(z?惫儽?%珌錚籱?(X裩z龟血∧▅?+!穐|P?$(?(PV+P9醻但jA?塲7v?(〉憠徨"?(%(Po@轾?塲壯*　戝?"?(%但jC碌塲嵣-Ae?*?錚9醻但jA?F媧嚝Afp夙~﹣?"慌 	?:?C?*??　&r?(　v晢t壞vx鵓?(%(ho@轾?杯r?　　咺(rr?　眮,北北北鵓9醻但jAh?胙-
/嘓nP　?(r　”?北?”?$晧j ?"籱?(Hl?%+X,間?(P　?#(r?　?

杯薄?睍夊嵩PHm淡r"?　? 　?薄北北?痹﹣錚籱?(闭(??/嘡　　3　　???北鼻,冰P9C~但jAh硡▇?-#*/嘗-zz+1.|WP9醻但jAh?%+X-╮妦跴jv%(mz勹? 鵯9嵩PH/﹥gv砶?+P当Dt?,P??盩??xm,r簝l???"Ⅻ?玧A(保?$pV%kX7積|P当Dt?,,?:R姩r盧穭北祬j盤角(@??┥陗﹣錚慌 ?々媙jvu鑦И穴?1Nt?,P:p,絅t?璽e?,祬v,?*P　薾@?(J塲 ?"礫j	誅?$pV鑦mv?+P?(r盤睊閞紻猑9??∮m^9?#B々申?+
|祦jA(筟胙-壣jt*?tn?+玶??北鵓Ⅻ[0;32mI (494) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (504) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (714) reset_reason: RTC reset 4 wakeup 0 store 0, reason is 7[0m
[0;32mI (714) app main: IDF version: v3.1-14-g49a48fd[0m
[0;32mI (714) app main: esp-aliyun verison: v1.0-107-g4cd3a62-dirty[0m
[0;32mI (724) app main: iotkit-embedded version: RELEASED_V2.30-dirty[0m
[0;33mW (734) awss_config: No data storage,the load data is empty[0m
[0;32mI (744) factory: Don't Factory[0m
[0;32mI (754) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (764) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (784) pwm: --- PWM v3.0
[0m
lightbulb_set_aim r 0 g 0 b 0

mode : sta(cc:50:e3:d6:41:3c)
add if0
del if0
usl
sul 0 0
[0;32mI (804) awss_config: ap ssid: abc, password: asdfghjkl[0m
mode : sta(cc:50:e3:d6:41:3c)
add if0
[0;32mI (814) awss_config: SYSTEM_EVENT_STA_START[0m
[0;32mI (824) app main: SYSTEM_EVENT_STA_START[0m
[0;32mI (824) awss_config: SYSTEM_EVENT_STA_START[0m
scandone
[0;32mI (834) app main: SYSTEM_EVENT_STA_START[0m
scandone
[0;33mW (3264) awss_config: SYSTEM_EVENT_STA_DISCONNECTED[0m
[0;33mW (3274) app main: SYSTEM_EVENT_STA_DISCONNECTED[0m
scandone
[0;33mW (5704) awss_config: SYSTEM_EVENT_STA_DISCONNECTED[0m
[0;33mW (5704) app main: SYSTEM_EVENT_STA_DISCONNECTED[0m
[0;32mI (6734) factory: Timeout, must be clear[0m
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
pm open phy_2,type:2 0 0
cnt 

connected with abc, channel 11
[0;32mI (10084) awss_config: SYSTEM_EVENT_STA_CONNECTED[0m
[0;32mI (11084) event: sta ip: 192.168.2.107, mask: 255.255.255.0, gw: 192.168.2.1[0m
[0;32mI (11084) awss_config: SYSTEM_EVENT_STA_GOT_IP[0m
[0;32mI (11094) app main: SYSTEM_EVENT_STA_GOT_IP[0m
[0;32mI (11094) app main: Network is Ready![0m
[0;32mI (11104) app main: entry linkkit main...[0m
[prt] log level set as: [ 5 ]

[1;36m[dbg] _dm_mgr_search_dev_by_devid(54): Device Not Found, devid: 0

[0m[1;33m[inf] dm_client_open(37): CM Fd: 0

[0m[1;33m[inf] guider_print_dev_guider_info(320): ....................................................

[0m[1;33m[inf] guider_print_dev_guider_info(321):           ProductKey : a17F5W6Icug

[0m[1;33m[inf] guider_print_dev_guider_info(322):           DeviceName : shebei01
[... 5482 lines cut ...]
[0m[1;36m[dbg] iotx_mc_cycle(1749): state = 4

[0m[1;31m[err] IOT_MQTT_Yield(3213): error occur rc=-27

[0m[1;33m[inf] iotx_mc_handle_reconnect(2765): Waiting to reconnect...

[0m[1;36m[dbg] iotx_mc_keepalive(2584): reconnect network fail, rc = -1

[0m[1;36m[dbg] iotx_mc_cycle(1749): state = 4

[0m[1;33m[inf] iotx_mc_handle_reconnect(2765): Waiting to reconnect...

[0m[1;36m[dbg] iotx_mc_keepalive(2584): reconnect network fail, rc = -1

[0m[1;36m[dbg] iotx_mc_cycle(1749): state = 4

[0m[1;31m[err] IOT_MQTT_Yield(3213): error occur rc=-27

[0m[1;33m[inf] iotx_mc_handle_reconnect(2765): Waiting to reconnect...

[0m[1;36m[dbg] iotx_mc_keepalive(2584): reconnect network fail, rc = -1

[0m[1;36m[dbg] iotx_mc_cycle(1749): state = 4

[0m

27-301, 2019-03-15:
[0m***ERROR*** A stack overflow in task Tmr Svc has been detected.

可以先 Component config - > FreeRTOS -> Timer stack size 改大为 4096 做下测试。

后面一个待查。

qiufeng81888, 2019-07-17:
同样问题 有结果吗？

27-301, 2019-08-16:
请使用基于 iotkit-embedded v3.0.1 重新适配的 esp-aliyun release v2.0 测试。
相关文档已更新，请参考。建议重新 clone 一下 esp-aliyun。
如在新版本上遇到问题，请重新提交 issue。
谢谢。

closed 2019-08-16 by 27-301