espressif/esp-aliyun #100: 如何编译

hellosunvi, 2019-03-20:
patch目录下的文件没有Makefile，执行复制指令到iotkit-embedded，怎么可以编译呢？

27-301, 2019-03-21:
只是把 patch 下的文件拷贝到 iotkit-embedded，编译的话需要到 iotkit-embedded 目录下执行：
```
make reconfig  #选择1 或 2
make
```

然后到 examples/solutions/smart_light
```
make defconfig
make menuconfig
```

hellosunvi, 2019-03-23:
1. 直接git clone https://github.com/espressif/esp-aliyun
在iotkit-embedded目录是没有任何东西，
2.执行cp -r patch/* iotkit-embedded/后，iotkit-embedded这个目录下面也没有makefile
3. 到iotkit-embedded目录执行 make defconfig ，因为没有Makefile所以提示make: *** No rule to make target 'reconfig'。

wanghui0574xs, 2019-03-28:
> 1. 直接git clone https://github.com/espressif/esp-aliyun
>    在iotkit-embedded目录是没有任何东西，
>    2.执行cp -r patch/* iotkit-embedded/后，iotkit-embedded这个目录下面也没有makefile
> 2. 到iotkit-embedded目录执行 make defconfig ，因为没有Makefile所以提示make: *** No rule to make target 'reconfig'。

我遇到跟你一样的情况

wjj-chn, 2019-03-29:
建议切换到esp-aliyun目录下，使用 git submodule update --init --recursive  更新 iotkit-embedded 子模
块。

gengyuchao, 2019-05-06:
> > 1. 直接git clone https://github.com/espressif/esp-aliyun
> >    在iotkit-embedded目录是没有任何东西，
> >    2.执行cp -r patch/* iotkit-embedded/后，iotkit-embedded这个目录下面也没有makefile
> > 2. 到iotkit-embedded目录执行 make defconfig ，因为没有Makefile所以提示make: *** No rule to make target 'reconfig'。
> 
> 我遇到跟你一样的情况

我遇到了相同的情况，并且发现了是因为执行git clone的时候没有克隆其子项目iotkit-embedded。因此 遇到同样的情况建议使用 git clone --recursive https://github.com/espressif/esp-aliyun.git  命令循环克隆git子项目。

yconnor, 2019-05-15:
大佬们都是用的什么样的软件编辑esp-aliyun下的examples等的。我这里用eclipse（用Espressif官网中的教程详细配置）在其它例程中编译成功后可以很方便的用F3来找到函数或定义的原型，但在本例程中编译成功后却按F3不能到达。非常不便于理解学习例程。哪位大神给指点一下？

yconnor, 2019-05-15:
> 大佬们都是用的什么样的软件编辑esp-aliyun下的examples等的。我这里用eclipse（用Espressif官网中的教程详细配置）在其它例程中编译成功后可以很方便的用F3来找到函数或定义的原型，但在本例程中编译成功后却按F3不能到达。非常不便于理解学习例程。哪位大神给指点一下？

OK，我知道了。在C/C++ Build 里的环境变量中加入 "IE_PATH" 里加入类似于“**/**/esp-aliyun\iotkit-embedded” （你具体的 iotkit-embedded 文件位置就可以了）希望这里能给大家带来帮助

27-301, 2019-08-16:
请使用基于 iotkit-embedded v3.0.1 重新适配的 esp-aliyun release v2.0 测试。
相关文档已更新，请参考。建议重新 clone 一下 esp-aliyun。
如在新版本上遇到问题，请重新提交 issue。
谢谢。

closed 2019-08-16 by 27-301