espressif/esp-aliyun #109: ***ERROR*** A stack overflow in task Tmr Svc has been detected.

yqj2012, 2019-05-10:
请问这个怎么解决啊？日志如下：
r l湗r??鈔???傡p寍巶??掃菕茕?p騨n??騨剴茕?b?	rrp騨鄡?鄠嚴l€€€b?n忏n冷幟?b€?~騨?牧寧 l`鼝?俷€巤nr帓?劺?p騨嗝?鄏?廛幚pp挏<勀?岒nn?娜r??俷??nr帓?劺傡rp騨鄡?鄏?廛幠pp捾<勄l庁?岒nn?呢r??俷??nr帓?睦?宭8rr咣抧€郻?掖?跉抝%丅J浇验*5?姅$?)S塖S闁S?N$&??W琕$??V柈?L[C叄+L?R悢44) boot: compile time 18:16:01[0m
[0;32mI (44) boot: SPI Speed      : 40MHz[0m
[0;32mI (47) boot: SPI Mode       : DOUT[0m
[0;32mI (52) boot: SPI Flash Size : 4MB[0m
[0;32mI (56) boot: Partition Table:[0m
[0;32mI (59) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (89) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (96) boot:  4 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (104) boot: End of partition table[0m
[0;31mE (108) boot: ota data partition invalid and no factory, will try all partitions[0m
[0;32mI (116) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xacae8 (707304) map[0m
[0;32mI (435) esp_image: segment 1: paddr=0x000bcb00 vaddr=0x3ffe8000 size=0x008dc (  2268) load[0m
[0;32mI (437) esp_image: segment 2: paddr=0x000bd3e4 vaddr=0x3ffe88dc size=0x0011c (   284) load[0m
[0;32mI (443) esp_image: segment 3: paddr=0x000bd508 vaddr=0x40100000 size=0x07790 ( 30608) load[0m
[0;32mI (464) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (509) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (519) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (729) reset_reason: RTC reset 2 wakeup 0 store 1817276786, reason is 2[0m
[0;32mI (739) app main: IDF version: v3.1-45-g217298f[0m
[0;32mI (749) app main: esp-aliyun verison: v1.0-111-gf6797e3-dirty[0m
[0;32mI (749) app main: iotkit-embedded version: RELEASED_V2.30-dirty[0m
[0;33mW (759) awss_config: No data storage,the load data is empty[0m
[0;32mI (769) factory: Don't Factory[0m
[0;32mI (779) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (789) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (799) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (809) pwm: --- PWM v3.0
[0m
lightbulb_set_aim r 0 g 0 b 0

[0;32mI (829) app main: Setting WiFi configuration SSID XIAOSHA...[0m
mode : sta(3c:71:bf:30:01:e0)
add if0
del if0
usl
sul 0 0
mode : sta(3c:71:bf:30:01:e0)
add if0
[0;32mI (859) awss_config: SYSTEM_EVENT_STA_START[0m
[0;32mI (869) app main: SYSTEM_EVENT_STA_START[0m
[0;32mI (879) awss_config: SYSTEM_EVENT_STA_START[0m
scandone
[0;32mI (879) app main: SYSTEM_EVENT_STA_START[0m
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
pm open phy_2,type:2 0 0
cnt 
[0;32mI (6989) factory: Timeout, must be clear[0m

connected with XIAOSHA, channel 1
[0;32mI (7269) awss_config: SYSTEM_EVENT_STA_CONNECTED[0m
[0;32mI (8269) event: sta ip: 192.168.0.117, mask: 255.255.255.0, gw: 192.168.0.1[0m
[0;32mI (8279) awss_config: SYSTEM_EVENT_STA_GOT_IP[0m
[0;32mI (8289) app main: SYSTEM_EVENT_STA_GOT_IP[0m
[0;32mI (8289) app main: Network is Ready![0m
[0;32mI (8299) app main: entry linkkit main...[0m
[prt] log level set as: [ 5 ]

[1;36m[dbg] _dm_mgr_search_dev_by_devid(54): Device Not Found, devid: 0

[0m[1;33m[inf] dm_client_open(37): CM Fd: 0

[... 803 lines cut ...]
3ffe82e0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000001 00000000 00000000 00000000 00000000 00000000 00000000 00000000 ffffffff 

3ffe8320  00000004 3ffedf34 3fff1108 40268018 00000000 00000000 00000000 40267dfe 00000000 00000000 00000000 00000000 00000000 00000000 00000000 4022655c 

3ffe8360  40226574 3ffedf34 00000001 40267f2c 00000000 00000000 00000000 40251e30 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000001 

3ffe83a0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 ffffffff 00000004 3ffedf34 3fff1108 40268018 00000000 00000000 00000000 40267dfe 

3ffe83e0  00000000 00000000 3fff1033 4029248f 00000000 00000000 00000000 4022655c 40226574 00000080 00000001 40267f2c 00002073 3fff102c 00000000 40251e30 

3ffe8420  3ffe8460 3ffe8450 00000008 3ffee76c 00000004 00000000 3ffef34c 00000001 00000000 00000000 00000004 40107518 4022430c 3fff102c a5a5a5a5 00000001 

3ffe8460  00000080 00000080 3ffef34c 40268018 3ffe98c2 00000001 3ffe98c0 40251d7a 00000400 00000400 3ffe85c8 401008c0 00000008 00000000 00000001 40105ae4 

3ffe84a0  40100a8a 00000001 00000001 40100a74 40251d2a 4023f254 3ffef364 40251d3d 3fff0340 00000002 3ffe84e4 00000000 fffffffe 3fff0a7c 40105e50 40105e99 

      PC: 0x40267f23       PS: 0x00000031       A0: 0x3ff7fff0       A1: 0x3ff7fff0 

      A2: 0x3ffedf34       A3: 0x40226574       A4: 0x4022655c       A5: 0x00000001 

      A6: 0x00000000       A7: 0x00000000       A8: 0x00000003       A9: 0x3fff0a73 

     A10: 0x00000001      A11: 0x4022a8ed      A12: 0x00000001      A13: 0x3ffedf34 

     A14: 0x40226574      A15: 0x4022655c      SAR: 0x00000019 EXCCAUSE: 0x00000003

yqj2012, 2019-05-10:
已经解决了， 主要是代码默认时钟任务栈分配小了。
解决办法：Component config  --->FreeRTOS  ---> (2560) Timer stack size
                默认是2048,  2M已经不小了， 看来这个 代码开销好大啊

closed 2019-05-10 by yqj2012

ytzhitong, 2019-06-06:
请问，Timer stack size设置成多少能正常工作？

yqj2012, 2019-06-19:
> 请问，Timer stack size设置成多少能正常工作？
这个要根据具体情况决定啊， 你可以逐步增加你的这个值看看