espressif/esp-aliyun #116: esp-aliyun 部分AP无法连接到ALIYUN IOT服务器

linweifu, 2019-06-27:
我用esp-aliyun连接部分手机热点，或者老毛子路由器固件的，AP模式。
无法连接阿里云 iot平台。老毛子路由器，切换到路由模式，就可以连接。
怀疑过DNS服务器不对。
使用tcpip_adapter_set_dns_info 修改了DNS为阿里云dns也是无法使用。
AliOS-Things项目上有看到同样的问题，说是已经修复了。
[alibaba/AliOS-Things issues 350](https://github.com/alibaba/AliOS-Things/issues/350)

下面是日志
> I (787477) app main: entry linkkit main...
[prt] log level set as: [ 5 ]
[dbg] _dm_mgr_search_dev_by_devid(54): Device Not Found, devid: 0
[inf] dm_client_open(37): CM Fd: 0
[inf] guider_print_dev_guider_info(320): ....................................................
[inf] guider_print_dev_guider_info(321):           ProductKey : a1k*****bu
[inf] guider_print_dev_guider_info(322):           DeviceName : lw*****02
[inf] guider_print_dev_guider_info(323):             DeviceID : a1*****bu.lwf-test-002
[inf] guider_print_dev_guider_info(327): ....................................................
[inf] guider_print_dev_guider_info(328):        PartnerID Buf : ,partner_id=espressif
[inf] guider_print_dev_guider_info(329):         ModuleID Buf : ,module_id=wroom
[inf] guider_print_dev_guider_info(330):           Guider URL : 
[inf] guider_print_dev_guider_info(332):       Guider SecMode : 2 (TLS + Direct)
[inf] guider_print_dev_guider_info(334):     Guider Timestamp : 2524608000000
[inf] guider_print_dev_guider_info(338): ....................................................
[inf] guider_print_conn_info(297): -----------------------------------------
[inf] guider_print_conn_info(298):             Host : a1*****bu.iot-as-mqtt.cn-shanghai.aliyuncs.com
[inf] guider_print_conn_info(299):             Port : 1883
[inf] guider_print_conn_info(304):         ClientID : a1*****bu.lwf-test-002|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom|
[inf] guider_print_conn_info(306):       TLS PubKey : 0x3f435354 ('-----BEGIN CERTI ...')
[inf] guider_print_conn_info(309): -----------------------------------------
[inf] iotx_mc_init(2334): MQTT init success!
I (787657) in: Loading the CA root certificate ...
I (787657) in:  ok (0 skipped)
I (787657) in: Connecting to /a1*****bu.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...
E (801257) in:  failed ! net_connect returned -0x0052
I (801257) in: ssl_disconnect
E (801257) in: handle is NULL
[err] iotx_mc_connect(2690): TCP or TLS Connection failed
E (801357) in: handle is NULL
[inf] iotx_mc_disconnect(2843): mqtt disconnect!
[inf] iotx_mc_release(2936): mqtt release!
[err] _mqtt_connect(310): mqtt connect failed
[err] _iotx_linkkit_master_connect(881): DM Start Failed
linkkit_main.950: IOT_Linkkit_Connect Failed

[err] IOT_MQTT_Destroy(3181): Invalid argument, client = 0x0
E (801467) awss_config: wifi_event_group not init
I (801477) app main: Network is Ready!

linweifu, 2019-06-27:
错误代码0x0052的宏定义是MBEDTLS_ERR_NET_UNKNOWN_HOST

27-301, 2019-08-16:
请使用基于 iotkit-embedded v3.0.1 重新适配的 esp-aliyun release v2.0 测试。
相关文档已更新，请参考。建议重新 clone 一下 esp-aliyun。
如在新版本上遇到问题，请重新提交 issue。
谢谢。

closed 2019-08-16 by 27-301