espressif/esp-aliyun #120: 烧录后 打印W (5777) factory: Erase restart count 就没反应了。

liubaiwu, 2019-06-29:
执行 make flash monitor 后 一直停留在 W (5777) factory: Erase restart count 。具体信息如下，请问这个是什么原因。芯片是esp8266 
I (44) boot: ESP-IDF  2nd stage bootloader
I (44) boot: compile time 00:47:53
I (44) qio_mode: Enabling default flash chip QIO
I (50) boot: SPI Speed      : 40MHz
I (56) boot: SPI Mode       : QIO
I (62) boot: SPI Flash Size : 4MB
I (69) boot: Partition Table:
I (74) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (97) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (108) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (120) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (132) boot:  4 ota_1            OTA app          00 11 00110000 000f0000
I (143) boot: End of partition table
I (150) boot: No factory image, trying OTA 0
I (157) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xaebc0 (71
5712) map
I (412) esp_image: segment 1: paddr=0x000bebd8 vaddr=0x3ffe8000 size=0x00838 (
2104) load
I (413) esp_image: segment 2: paddr=0x000bf418 vaddr=0x3ffe8838 size=0x0019c (
 412) load
I (423) esp_image: segment 3: paddr=0x000bf5bc vaddr=0x40100000 size=0x06794 ( 2
6516) load
0x40100000: _stext at ??:?

I (445) boot: Loaded app from partition at offset 0x10000
I (467) system_api: Base MAC address is not set, read default base MAC address f
rom BLK0 of EFUSE
I (477) system_api: Base MAC address is not set, read default base MAC address f
rom BLK0 of EFUSE
W (487) phy_init: failed to load RF calibration data (0x1102), falling back to f
ull calibration
I (667) phy_init: phy ver: 1055_12
W (667) phy_init: saving new calibration data because of checksum failure, mode(
2)
I (677) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
I (687) app main: IDF version:
I (687) app main: IDF ------------------ hello look.
I (697) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldow
n: 0| Intr:0
I (707) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldow
n: 0| Intr:0
I (717) pwm: --- PWM v3.2

lightbulb_set_aim r 0 g 0 b 0
[prt] log level set as: [ 4 ]
I (777) app main: SYSTEM_EVENT_STA_START
I (787) app main: Network is Ready!
W (5777) factory: Erase restart count

wjj-chn, 2019-07-01:
你好，可以更详细的描述以下信息吗？
ESP 芯片 
ESP 模组名称 
ESP SDK 版本号  
所使用的例程
您参考的文档链接  
问题复现步骤 
Log

liubaiwu, 2019-07-01:
这个我要具体在查下。 查了后我在重新提交上来。
esp芯片 和模组 我要具体去查下在来提交，好像esp8266 12f  倒是在具体确认下。
esp sdk 的版本 使用的是feature/update_3.0.1_for_mesh 分支，因为我使用master 主分支 ，make编译会报错。
所使用的例程是 ：examples/solutions/smart_light/
参考文档链接 https://github.com/espressif/esp-aliyun/tree/feature/update_3.0.1_for_mesh
问题复现的步骤，就是安文档正常步骤执行，在最后执行 make flash monitor 的完成后，开始输出日志信息，输出部分信息后，就在输出 W (5777) factory: Erase restart count地方停止了。

另外有两个问题质询下。
1、请问nonos sdk  是否有支持阿里云 云智能app 配网 和 微信配网的demo 。
2、rtos sdk 是否有中文版api，接口说明文档。

ljy770, 2019-07-25:
1、请问nonos sdk 是否有支持阿里云 云智能app 配网 和 微信配网的demo 。
nonos sdk 只有支持 微信配网的demo：ESP8266_NONOS_SDK/examples/smart_config
2、rtos sdk 是否有中文版api，接口说明文档。
https://www.espressif.com/zh-hans/support/download/documents
ESP8266 RTOS SDK API 参考 (仅针对旧版 SDK V2.X 及之前版本)

liubaiwu, 2019-08-07:
你好，请问esp8266 12-f 模组 在使用esp-aliyun 项目中 

Smart Light 解决方案  进行配网的时候 
打印W (5777) factory: Erase restart count 就没反应了。已经在https://github.com/espressif/esp-aliyun/issues 这里提了相应的问题。这个应该如何解决？

 https://www.espressif.com/zh-hans/support/download/documents

27-301, 2019-08-16:
请使用基于 iotkit-embedded v3.0.1 重新适配的 esp-aliyun release v2.0 测试。
相关文档已更新，请参考。建议重新 clone 一下 esp-aliyun。
如在新版本上遇到问题，请重新提交 issue。
谢谢。

closed 2019-08-16 by 27-301