espressif/esp-aliyun #121: 断网重连后，内存泄漏问题

wzbhbb, 2019-07-13:
请看日志:

第一部分，正常启动:
I (66) boot: ESP-IDF v3.1-43-gd7da7bd 2nd stage bootloader
I (67) boot: compile time 22:38:18
I (75) qio_mode: Enabling default flash chip QIO
I (76) boot: SPI Speed      : 40MHz
I (81) boot: SPI Mode       : QIO
I (87) boot: SPI Flash Size : 2MB
I (93) boot: Partition Table:
I (99) boot: ## Label            Usage          Type ST Offset   Length
I (110) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (121) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (133) boot:  2 factory          factory app      00 00 00010000 000f0000
I (145) boot: End of partition table
I (151) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x82908 (534792) map
I (344) esp_image: segment 1: paddr=0x00092920 vaddr=0x3ffe8000 size=0x00840 (  2112) load
I (345) esp_image: segment 2: paddr=0x00093168 vaddr=0x3ffe8840 size=0x0011c (   284) load
I (356) esp_image: segment 3: paddr=0x0009328c vaddr=0x40100000 size=0x075a4 ( 30116) load
0x40100000: _stext at ??:?

I (379) boot: Loaded app from partition at offset 0x10000
I (423) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (433) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (653) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
I (653) app main: IDF version: 100.100.100_3
I (653) app main: esp-aliyun verison: 100.100.100_3
I (663) app main: iotkit-embedded version: 100.100.100_3
I (683) app main: Setting WiFi configuration SSID My-502...
mode : sta(5c:cf:7f:69:83:70)
add if0
I (693) app main: wait wifi connect...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
pm open phy_2,type:2 0 0
cnt 

connected with My-502, channel 8
I (6033) event: sta ip: 192.168.1.7, mask: 255.255.255.0, gw: 192.168.1.1
[prt] log level set as: [ 5 ]
[inf] guider_print_dev_guider_info(320): ....................................................
[inf] guider_print_dev_guider_info(321):           ProductKey : 2Ry2FSorRbU
[inf] guider_print_dev_guider_info(322):           DeviceName : RLd4yrYfhH9KewPlu3HA
[inf] guider_print_dev_guider_info(323):             DeviceID : 2Ry2FSorRbU.RLd4yrYfhH9KewPlu3HA
[inf] guider_print_dev_guider_info(327): ....................................................
[inf] guider_print_dev_guider_info(328):        PartnerID Buf : ,partner_id=espressif
[inf] guider_print_dev_guider_info(329):         ModuleID Buf : ,module_id=wroom
[inf] guider_print_dev_guider_info(330):           Guider URL : 
[inf] guider_print_dev_guider_info(332):       Guider SecMode : 2 (TLS + Direct)
[inf] guider_print_dev_guider_info(334):     Guider Timestamp : 2524608000000
[inf] guider_print_dev_guider_info(338): ....................................................
[inf] guider_print_conn_info(297): -----------------------------------------
[inf] guider_print_conn_info(298):             Host : 2Ry2FSorRbU.iot-as-mqtt.cn-shanghai.aliyuncs.com
[inf] guider_print_conn_info(299):             Port : 1883
[inf] guider_print_conn_info(304):         ClientID : 2Ry2FSorRbU.RLd4yrYfhH9KewPlu3HA|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,ext=0,partner_id=espressif,module_id=wroom|
[inf] guider_print_conn_info(306):       TLS PubKey : 0x4022a5b8 ('-----BEGIN CERTI ...')
0x4022a5b8: chip_v6_unset_chanfreq at ??:?

[inf] guider_print_conn_info(309): -----------------------------------------
[inf] iotx_mc_init(2334): MQTT init success!
I (6793) in: Creat socket ...
I (6793) in: OK
I (6793) in: bind socket ......
I (6793) in: OK
I (6793) in: setsockopt SO_REUSEADDR......
I (6803) in: OK
[... 107 lines cut ...]
I (116503) in: establish tcp connection with server(host=2Ry2FSorRbU.iot-as-mqtt.cn-shanghai.aliyuncs.com port=1883)
I (116563) in: OK
[inf] iotx_mc_connect(2726): mqtt connect success!
[inf] iotx_mc_keepalive(2588): network is reconnected!
event_handle|060 :: MQTT reconnect.
I (118333) MQTT: min:26168 heap:41056
mqtt_client|246 :: packet-id=15, publish topic msg={"attr_name":"temperature","attr_value":"9"}
[dbg] iotx_mc_cycle(1791): PUBACK
event_handle|088 :: publish success, packet-id=15
I (128353) MQTT: min:26168 heap:41060
mqtt_client|246 :: packet-id=16, publish topic msg={"attr_name":"temperature","attr_value":"10"}
[dbg] iotx_mc_cycle(1791): PUBACK
event_handle|088 :: publish success, packet-id=16
I (138363) MQTT: min:26168 heap:41060
mqtt_client|246 :: packet-id=17, publish topic msg={"attr_name":"temperature","attr_value":"11"}
[dbg] iotx_mc_cycle(1791): PUBACK
event_handle|088 :: publish success, packet-id=17

总结:
第一次的内存信息为
I (63413) MQTT: min:26236 heap:41248
第二次重连后变成
I (128353) MQTT: min:26168 heap:41060

这个是example工程，在我实际的工程里，会由于这种情况导致内存不足而程序卡死状态，请问这个该如何解决？？？

ljy770, 2019-07-25:
ESP-IDF: 请切换到 release/v3.2 分支： git checkout release/v3.2
ESP8266_RTOS_SDK: 请切换到 release/v3.1 分支： git checkout release/v3.1
看 您的 log ，是 ESP-IDF v3.1，切换到 release/v3.2 分支呢？

wzbhbb, 2019-07-25:
你好，我用的就是 ESP8266_RTOS_SDK， release3.1的分支。

ljy770, 2019-07-26:
@wzbhbb 请问您在哪里设置的打印信息，我这边同步复现一下。谢谢。

wzbhbb, 2019-07-29:
@ljy770 就是demo中的代码

27-301, 2019-08-16:
请使用基于 iotkit-embedded v3.0.1 重新适配的 esp-aliyun release v2.0 测试。
相关文档已更新，请参考。建议重新 clone 一下 esp-aliyun。
如在新版本上遇到问题，请重新提交 issue。
谢谢。

closed 2019-08-16 by 27-301