espressif/esp-aliyun #124: 请问下这里面如何控制gpio

liubaiwu, 2019-07-24:
好像不能使用rtos sdk 的 gpio_set_level 来控制gpio 。请问下这里的版本是用的什么版本。感觉跟rtos sdk 目录结构等等都完全不一样。

ljy770, 2019-07-25:
请问 gpio 有进行正确的配置吗？请参考 ESP8266_RTOS_SDK/examples/peripherals/gpio/README.md
esp-aliyun 是子项目，并不是完成完整的 sdk, 需要依赖于 rtos sdk 。使用中需要先下载 ESP8266_RTOS_SDK (for esp8266) 或者 esp-idf (for esp32).

liubaiwu, 2019-07-25:
我使用的是 feature/update_3.0.1_for_mesh 分支版本。可以正常编译通过，能正常烧录并且运行。配网也能成功。 我也没有像你说的要先下载 ESP8266_RTOS_SDK ，看你们的说明里面好像也没有地方提到要下载，或者是要在esp-aliyun项目里面需要集成 ESP8266_RTOS_SDK 。如果想要在esp-aliyun项目里面使用完整的 ESP8266_RTOS_SDK 里面的api。需要如何做集成？能做吗？另外如何知道esp-aliyun项目目前能够使用哪些rtos-sdk 的方法函数？

lqq96, 2019-07-26:
目前  feature/update_3.0.1_for_mesh 当前在 esp8266 中测试使用了 master 分支.
使用的方式为  export IDF_PATH= "文件路径"/ESP8266_RTOS_SDK.

27-301, 2019-08-16:
请使用基于 iotkit-embedded v3.0.1 重新适配的 esp-aliyun release v2.0 测试。
相关文档已更新，请参考。建议重新 clone 一下 esp-aliyun。
如在新版本上遇到问题，请重新提交 issue。
谢谢。

closed 2019-08-16 by 27-301