espressif/esp-aliyun #125: 使用阿里云demo app配网失败

yqj2012, 2019-07-25:
系统上电串口打印处于wifi监控模式
W (707) awss_config: No data storage,the load data is empty
I (717) app main: IOTX_AWSS_START
[crt] __awss_start

I (737) awss_config: wifi running at monitor mode
之后就一直打印wifi信道信息
[crt] [10] ssid:ChinaNet-CvaX, mac:40f42059339a, chn:11, auth:wpa2-psk, aes, tkip, rssi:-83, adha:0

[crt] chan 1

[crt] chan 2

使用阿里demo app，点击添加设备，选择插座wifi--点击连接--输入wifi密码，下一步---确认在闪烁--之后从日志里看说是密码错误， 可是我确认密码正确啊。
一下是完整日志：

 (121) boot: ota data partition invalid and no factory, will try all partitions
I (129) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xac698 (706200) map
I (375) esp_image: segment 1: paddr=0x000bc6b0 vaddr=0x3ffe8000 size=0x008bc (  2236) load
I (377) esp_image: segment 2: paddr=0x000bcf74 vaddr=0x3ffe88bc size=0x0011c (   284) load
I (383) esp_image: segment 3: paddr=0x000bd098 vaddr=0x40100000 size=0x07660 ( 30304) load
I (401) boot: Loaded app from partition at offset 0x10000
I (437) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (437) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (637) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
I (637) app main: IDF version: v3.1-51-g913a06a9
I (637) app main: esp-aliyun verison: v1.0-111-gf6797e3-dirty
I (647) app main: iotkit-embedded version: RELEASED_V2.30
W (647) awss_config: No data storage,the load data is empty
I (657) factory: Don't Factory
mode : sta(5c:cf:7f:34:a2:43)
add if0
W (727) awss_config: No data storage,the load data is empty
I (727) app main: IOTX_AWSS_START
[crt] __awss_start

I (747) awss_config: wifi running at monitor mode
[crt] chan 1

[crt] [1] ssid:DIRECT-15-HP M277 LaserJet, mac:fada0c7ec115, chn:1, auth:wpa2-psk, aes, aes, rssi:-72, adha:0

[crt] [2] ssid:SasmartGuest, mac:9ad9b32c713c, chn:1, auth:wpa2-psk, aes, aes, rssi:-63, adha:0

[crt] [3] ssid:sasmart, mac:94d9b32c713c, chn:1, auth:wpa2-psk, aes, aes, rssi:-62, adha:0

[crt] chan 6

[crt] [4] ssid:ChinaNet-Wgl4, mac:ca50e9731df5, chn:6, auth:wpa2-psk, aes, tkip, rssi:-94, adha:0

[crt] [5] ssid:DIRECT-78-HP M130 LaserJet, mac:dea266017c78, chn:6, auth:wpa2-psk, aes, aes, rssi:-90, adha:0

[crt] [6] ssid:iPhone, mac:52fa841ee65b, chn:6, auth:wpa2-psk, aes, aes, rssi:-75, adha:0

[crt] chan 11

[crt] [7] ssid:XIAOSHA, mac:d076e7b38436, chn:11, auth:wpa2-psk, aes, aes, rssi:-54, adha:0

[crt] [8] ssid:保利1210, mac:0c4b541d642f, chn:11, auth:wpa2-psk, aes, aes, rssi:-85, adha:0

[crt] [9] ssid:ChinaNet-CvaX, mac:40f42059339a, chn:11, auth:wpa2-psk, aes, tkip, rssi:-83, adha:0

[crt] chan 1

[crt] chan 2

[crt] chan 3

[crt] chan 4

[crt] chan 5
[... 2314 lines cut ...]
[wrn] get_ssid_passwd_from_w(210): p2p decrypt passwd content err
E (30707) app main: IOTX_AWSS_PASSWD_ERR
[wrn] get_ssid_passwd_from_w(210): p2p decrypt passwd content err
E (30707) app main: IOTX_AWSS_PASSWD_ERR
E (30967) in: lock mutex failed: - 'Invalid argument' (22)
[crt] chan 7

E (30967) in: unlock mutex failed - 'Invalid argument' (22)
E (31247) in: lock mutex failed: - 'Invalid argument' (22)
[crt] chan 8

E (31247) in: unlock mutex failed - 'Invalid argument' (22)
E (31527) in: lock mutex failed: - 'Invalid argument' (22)
[crt] chan 9

E (31527) in: unlock mutex failed - 'Invalid argument' (22)
E (31807) in: lock mutex failed: - 'Invalid argument' (22)
[crt] chan 10

E (31807) in: unlock mutex failed - 'Invalid argument' (22)
E (32087) in: lock mutex failed: - 'Invalid argument' (22)
[crt] chan 11

E (32087) in: unlock mutex failed - 'Invalid argument' (22)
E (32367) in: lock mutex failed: - 'Invalid argument' (22)

ljy770, 2019-07-25:
您好，请您按照 esp-aliyun/examples/solutions/smart_light/README.md 操作，配网成功需要注意几点：
1 在 智能生活开放平台 创建产品，会有相应的产品配网二维码，需用下方说明的 APP 扫描这个二维码配网。
2 正确配置四元组。
3 手机从阿里巴巴官网 https://living.aliyun.com/doc#muti-app.html下载智能家居公版 APP, 使用开发版本，公版APP 扫描二维码会显示二维码不支持。。

yqj2012, 2019-07-26:
@ljy770 谢谢解答，确实是需要扫描二维码才能正常配网

closed 2019-07-26 by yqj2012