espressif/esp-aliyun #126: examples/solutions/smart_light多次配网失败

yqj2012, 2019-07-26:
问题描述：
   我做了个按键，每按一次按键都调用esp_info_erase("wifi_config");去擦除wifi配置信息，然后复位，重新扫码配网，大概重复几次就会出现以下bug，请帮忙查看是什么问题，我也查下。

[dbg] _iotx_linkkit_event_callback(287): Current Devid: 0
user_initialized.400: Device Initialized, Devid: 0
[dbg] CoAPServerPath_2_option(47): The uri is /sys/awss/event/connectap/notify
[dbg] __awss_suc_notify(456): resp:0
[inf] _iotx_linkkit_event_callback(223): Receive Message Type: 0
user_connected_event_handler.87: Cloud Connected
[dbg] CoAPServerPath_2_option(47): The uri is /sys/awss/event/connectap/notify
[dbg] __awss_suc_notify(456): resp:0
[dbg] CoAPRequestMessage_handle(730): Request path is /sys/awss/device/connectap/info/get
[dbg] CoAPServerPath_2_option(47): The uri is /sys/awss/event/connectap/notify
I (42047) app main: IOTX_AWSS_BIND_NOTIFY
E (42057) in: lock mutex failed: - 'Invalid argument' (22)
I (42057) app main: IOTX_AWSS_BIND_NOTIFY
E (42067) in: lock mutex failed: - 'Invalid argument' (22)
I (42067) app main: IOTX_AWSS_BIND_NOTIFY
E (42067) in: lock mutex failed: - 'Invalid argument' (22)
I (42077) app main: IOTX_AWSS_BIND_NOTIFY
E (42077) in: lock mutex failed: - 'Invalid argument' (22)
I (42087) app main: IOTX_AWSS_BIND_NOTIFY
E (42087) in: lock mutex failed: - 'Invalid argument' (22)
I (42097) app main: IOTX_AWSS_BIND_NOTIFY
E (42107) in: lock mutex failed: - 'Invalid argument' (22)
I (42107) app main: IOTX_AWSS_BIND_NOTIFY
E (42117) in: lock mutex failed: - 'Invalid argument' (22)
I (42117) app main: IOTX_AWSS_BIND_NOTIFY
E (42127) in: lock mutex failed: - 'Invalid argument' (22)
I (42127) app main: IOTX_AWSS_BIND_NOTIFY
E (42137) in: lock mutex failed: - 'Invalid argument' (22)
I (42137) app main: IOTX_AWSS_BIND_NOTIFY
E (42147) in: lock mutex failed: - 'Invalid argument' (22)
I (42157) app main: IOTX_AWSS_BIND_NOTIFY
E (42157) in: lock mutex failed: - 'Invalid argument' (22)
I (42167) app main: IOTX_AWSS_BIND_NOTIFY
E (42167) in: lock mutex failed: - 'Invalid argument' (22)
I (42177) app main: IOTX_AWSS_BIND_NOTIFY
E (42177) in: lock mutex failed: - 'Invalid argument' (22)
I (42187) app main: IOTX_AWSS_BIND_NOTIFY
E (42187) in: lock mutex failed: - 'Invalid argument' (22)
I (42297) app main: IOTX_AWSS_BIND_NOTIFY
E (42297) in: lock mutex failed: - 'Invalid argument' (22)
I (42597) app main: IOTX_AWSS_BIND_NOTIFY
E (42597) in: lock mutex failed: - 'Invalid argument' (22)
I (42897) app main: IOTX_AWSS_BIND_NOTIFY
E (42897) in: lock mutex failed: - 'Invalid argument' (22)
I (43197) app main: IOTX_AWSS_BIND_NOTIFY
E (43197) in: lock mutex failed: - 'Invalid argument' (22)
I (43497) app main: IOTX_AWSS_BIND_NOTIFY
E (43497) in: lock mutex failed: - 'Invalid argument' (22)
I (43797) app main: IOTX_AWSS_BIND_NOTIFY
E (43797) in: lock mutex failed: - 'Invalid argument' (22)
I (44097) app main: IOTX_AWSS_BIND_NOTIFY
E (44097) in: lock mutex failed: - 'Invalid argument' (22)
I (44397) app main: IOTX_AWSS_BIND_NOTIFY
E (44397) in: lock mutex failed: - 'Invalid argument' (22)
I (44697) app main: IOTX_AWSS_BIND_NOTIFY
E (44697) in: lock mutex failed: - 'Invalid argument' (22)
[dbg] CoAPRequestMessage_handle(730): Request path is /sys/awss/device/connectap/info/get

27-301, 2019-08-16:
请使用基于 iotkit-embedded v3.0.1 重新适配的 release v2.0 测试，谢谢。

27-301, 2019-08-16:
相关文档已更新，请参考。建议重新 clone 一下 esp-aliyun。
如在新版本上遇到问题，请重新提交 issue。
谢谢。

closed 2019-08-16 by 27-301