espressif/esp-aliyun #128: 烧录后 打印W (5777) factory: Erase restart count 就没反应了。

liubaiwu, 2019-07-30:
执行 make flash monitor 后 一直停留在 W (5777) factory: Erase restart count 。具体信息如下，请问这个是什么原因。芯片是esp8266
I (44) boot: ESP-IDF 2nd stage bootloader
I (44) boot: compile time 00:47:53
I (44) qio_mode: Enabling default flash chip QIO
I (50) boot: SPI Speed : 40MHz
I (56) boot: SPI Mode : QIO
I (62) boot: SPI Flash Size : 4MB
I (69) boot: Partition Table:
I (74) boot: ## Label Usage Type ST Offset Length
I (85) boot: 0 nvs WiFi data 01 02 00009000 00004000
I (97) boot: 1 otadata OTA data 01 00 0000d000 00002000
I (108) boot: 2 phy_init RF data 01 01 0000f000 00001000
I (120) boot: 3 ota_0 OTA app 00 10 00010000 000f0000
I (132) boot: 4 ota_1 OTA app 00 11 00110000 000f0000
I (143) boot: End of partition table
I (150) boot: No factory image, trying OTA 0
I (157) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xaebc0 (71
5712) map
I (412) esp_image: segment 1: paddr=0x000bebd8 vaddr=0x3ffe8000 size=0x00838 (
2104) load
I (413) esp_image: segment 2: paddr=0x000bf418 vaddr=0x3ffe8838 size=0x0019c (
412) load
I (423) esp_image: segment 3: paddr=0x000bf5bc vaddr=0x40100000 size=0x06794 ( 2
6516) load
0x40100000: _stext at ??:?

I (445) boot: Loaded app from partition at offset 0x10000
I (467) system_api: Base MAC address is not set, read default base MAC address f
rom BLK0 of EFUSE
I (477) system_api: Base MAC address is not set, read default base MAC address f
rom BLK0 of EFUSE
W (487) phy_init: failed to load RF calibration data (0x1102), falling back to f
ull calibration
I (667) phy_init: phy ver: 1055_12
W (667) phy_init: saving new calibration data because of checksum failure, mode(
2)
I (677) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
I (687) app main: IDF version:
I (687) app main: IDF ------------------ hello look.
I (697) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldow
n: 0| Intr:0
I (707) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldow
n: 0| Intr:0
I (717) pwm: --- PWM v3.2

lightbulb_set_aim r 0 g 0 b 0
[prt] log level set as: [ 4 ]
I (777) app main: SYSTEM_EVENT_STA_START
I (787) app main: Network is Ready!
W (5777) factory: Erase restart count

我这边使用的相关硬件和资料。
esp8266 12f 。
esp sdk 的版本 是 ESP8266_RTOS_SDK的master主分支。 esp-aliyun使用的是feature/update_3.0.1_for_mesh 分支 。
所使用的例程是 ：https://github.com/espressif/esp-aliyun/tree/master/examples/solutions/smart_light
参考文档链接 https://github.com/espressif/esp-aliyun/tree/feature/update_3.0.1_for_mesh
问题复现的步骤，就是安文档正常步骤执行，在最后执行 make flash monitor 的完成后，开始输出日志信息，输出部分信息后，就在输出 W (5777) factory: Erase restart count地方停止了。

检查了下代码应该是在smart_light 目录下的app_main.c文件的 app_main方法里面
下面的代码处停止的。
这个代码处
if (restart_count_get() >= LINKKIT_RESTART_COUNT_RESET) {

liubaiwu, 2019-08-01:
使用master分支 执行命令make flash monitor时， 报如下错误，跟上面同样的环境。
$ make flash monitor
fatal: 不是一个 git 仓库（或者任何父目录）：.git
Toolchain path: /opt/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: crosstool-ng-1.22.0-92-g8facf4c0
Compiler version: 5.2.0
fatal: 不是一个 git 仓库（或者任何父目录）：.git
CC build/esp-aliyun/platform/hal/os/espressif/HAL_OS_espressif.o
I:/esp8266/idf/msys/msys32/home/look/esp-aliyun/platform/hal/os/espressif/HAL_OS_espressif.c: In function 'HAL_UptimeMs':
I:/esp8266/idf/msys/msys32/home/look/esp-aliyun/platform/hal/os/espressif/HAL_OS_espressif.c:165:5: error: implicit declaration of function 'clock_gettime' [-Werror=implicit-function-declaration]
     clock_gettime(CLOCK_MONOTONIC, &ts);
     ^
cc1.exe: some warnings being treated as errors
make[1]: *** [/home/look/ESP8266_RTOS_SDK/make/component_wrapper.mk:286：platform/hal/os/espressif/HAL_OS_espressif.o] 错误 1
make: *** [I:/esp8266/idf/msys/msys32/home/look/ESP8266_RTOS_SDK/make/project.mk:517：component-esp-aliyun-build] 错误 2

ljy770, 2019-08-07:
esp sdk 的版本 是 ESP8266_RTOS_SDK的master主分支。
==>您好，这里需要 切换到 release/v3.1 分支： git checkout release/v3.1

liubaiwu, 2019-08-08:
好的 ，我试试。

27-301, 2019-08-16:
请使用基于 iotkit-embedded v3.0.1 重新适配的 esp-aliyun release v2.0 测试。
相关文档已更新，请参考。建议重新 clone 一下 esp-aliyun。
如在新版本上遇到问题，请重新提交 issue。
谢谢。

closed 2019-08-16 by 27-301