espressif/esp-aliyun #130: 编译错误

shibaoyi, 2019-09-06:
错误如下：The recommended way to install a packages is via "pacman". Please run "pacman -Ss <package_name>" for searching the package database and if found then "pacman -S mingw-w64-i686-python2-<package_name>" for installing it.
NOTE: You may need to run "pacman -Syu" if your package database is older and run twice if the previous run updated "pacman" itself.
Please read https://github.com/msys2/msys2/wiki/Using-packages for further information about using "pacman"
WARNING: The cryptography package have dependencies on system packages so please make sure you run "pacman -Syu" followed by "pacman -S mingw-w64-i686-python2-cryptography".
Alternatively, you can run "D:/msys32/mingw32/bin/python.exe -m pip install --user -r D:/msys32/home/root/esp/esp-idf/requirements.txt" for resolving the issue.
make: *** [D:/msys32/home/root/esp/esp-idf/make/project.mk:479：check_python_dependencies] 错误 1
执行：D:/msys32/mingw32/bin/python.exe -m pip install --user -r D:/msys32/home/root/esp/esp-idf/requirements.txt 仍然报错， esp32-idf-v3.2-sdk 重新安装依赖包也不行 哪位大神帮帮忙

27-301, 2019-09-06:
你先按照 idf 的手册，https://docs.espressif.com/projects/esp-idf/en/v3.2/get-started/index.html，把 idf examples 里的 hello_world 编译过了。

shibaoyi, 2019-09-09:
问题已经找到，idf软件包版本太低导致，运行：pacman -Syu 更新就好。

make -j8  又报错，我太难了，谁遇到这个问题的，说一下，谢谢

make：python：命令未找到
make: *** [D:/msys32/home/Administrator/esp32/esp-idf-v3.2/make/project.mk:479：check_python_dependencies] 错误 127

/bin/sh: python: 未找到命令
make[1]: *** [/home/Administrator/esp32/esp-idf-v3.2/components/esptool_py/Makefile.projbuild:59：/home/Administrator/esp32/esp-idf-v3.2/esp-aliyun-master/examples/mqtt/mqtt_example/build/bootloader/bootloader.bin] 错误 127
make: *** [/home/Administrator/esp32/esp-idf-v3.2/components/bootloader/Makefile.projbuild:41：/home/Administrator/esp32/esp-idf-v3.2/esp-aliyun-master/examples/mqtt/mqtt_example/build/bootloader/bootloader.bin] 错误 2

27-301, 2019-09-10:
你的环境里没有 python，需要安装。

closed 2019-11-06 by Eavo