espressif/esp-aliyun #131: 编译时的错误，该如何解决？

hantown, 2019-09-10:
## 1. 开发环境

- 模组或开发板：
- esp-aliyun 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP8266_RTOS_SDK 版本（``git rev-parse --short HEAD`` 来获取 commit id）：
- ESP-IDF 版本（``git rev-parse --short HEAD`` 来获取 commit id）：

## 2. 问题描述
编译时出现错误：
CC build/wpa_supplicant/src/crypto/sha1.o
CC build/wpa_supplicant/src/crypto/aes-internal.o
CC build/wpa_supplicant/src/crypto/md5-internal.o
CC build/wpa_supplicant/src/crypto/sha256-internal.o
CC build/wpa_supplicant/src/crypto/aes-internal-enc.o
CC build/wpa_supplicant/port/os_xtensa.o
AR build/wpa_supplicant/libwpa_supplicant.a
Generating esp8266.project.ld
LD build/smart_light.elf
E:/My_Documents/ESP8266/ESP_8266_IDF/ESP8266_RTOS_SDK/esp-aliyun-master/examples/solutions/smart_light/build/esp-aliyun-master\libesp-aliyun-master.a(wrapper_system.o):(.literal.HAL_UptimeMs+0x0): undefined reference to `esp_log_timestamp'
E:/My_Documents/ESP8266/ESP_8266_IDF/ESP8266_RTOS_SDK/esp-aliyun-master/examples/solutions/smart_light/build/esp-aliyun-master\libesp-aliyun-master.a(wrapper_system.o): In function `HAL_UptimeMs':
E:/My_Documents/ESP8266/ESP_8266_IDF/ESP8266_RTOS_SDK/esp-aliyun-master/wrappers/wrapper_system.c:188: undefined reference to `esp_log_timestamp'
collect2.exe: error: ld returned 1 exit status
make: *** [E:\My_Documents\ESP8266\ESP_8266_IDF\ESP8266_RTOS_SDK/make/project.mk:510：/e/My_Documents/ESP8266/ESP_8266_IDF/ESP8266_RTOS_SDK/esp-aliyun-master/examples/solutions/smart_light/build/smart_light.elf] 错误 1

### 2.1 复现步骤

make 编译的时候

### 2.2 复现问题的代码

无代码

## 3. 调试 Logs

无
## 4. 其他项
无

27-301, 2019-09-10:
其实是希望你能提供详细的版本信息，这样比较容易排查。
esp-aliyun 版本（git rev-parse --short HEAD 来获取 commit id）：
ESP8266_RTOS_SDK 版本（git rev-parse --short HEAD 来获取 commit id）：

当前的问题，应该是你使用了 master 版本的 RTOS SDK，请使用 release v3.2 的版本。可以在 RTOS SDK 下 `git checkout release/v3.2`切到 release/v3.2 分支。
建议你使用 git 来管理代码，不要直接下载。

hantown, 2019-09-10:
我SDK版本是release v3.2的，是git下来的。esp-aliyun是下载的master版本。

27-301, 2019-09-10:
aliyun master + v3.2 没有问题。请再确认 rtos sdk 的版本，请在 rtos sdk 目录下 git rev-parse --short HEAD 看下具体 commit

hantown, 2019-09-11:
直接注释掉return esp_log_timestamp(); 这行，换成return 0；编译通过了！

27-301, 2019-09-11:
这个不是解决问题的最优方式。

ljy770, 2019-09-11:
请确认一下 ESP8266_RTOS_SDK/components/log/log.c 中是包含 esp_log_timestamp() 这个函数并且有效,否则可能用的是其他版本,或者也可以按下面修改函数, 头文件添加 
#include "esp_timer.h"
 uint64_t HAL_UptimeMs(void)
{
    return (esp_timer_get_time()/1000);
}

hantown, 2019-09-19:
log.c文件里面没有esp_log_timestamp()这个函数。
在source insight里面搜索整个文件夹也没有发现这个函数的原型。

hantown, 2019-09-19:
> aliyun master + v3.2 没有问题。请再确认 rtos sdk 的版本，请在 rtos sdk 目录下 git rev-parse --short HEAD 看下具体 commit

root@user-virtual-machine:/mnt/hgfs/ESP8266/ESP_8266_IDF/ESP8266_RTOS_SDK# git rev-parse --short HEAD
d7960baa

27-301, 2019-09-19:
这个 commit 是 master 分支上的。你执行 git checkout release/v3.2，后面会在 esp-aliyun 中 fix 这个问题。

27-301, 2019-09-20:
已 fix。

closed 2019-09-20 by 27-301